lvgl/lvgl #3386: SDL Renderer Issue When Zoom Set

mariotaku, 2022-05-28:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

After commit https://github.com/lvgl/lvgl/commit/318146a2c25362eabf258470be263a4cfeaefe87, 
SDL renderer will have issue drawing objects has zoom transform. Object is not properly zoomed, and when drawing it invades a part of objects in the front.

[image]

Sorry that I didn't post more details. I'll be investigating this issue and make patches.

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2022-05-28:
Hi,

Note that, before that commit only images could be scaled  and not any widget. 

Regarding the SDL support see this [comment](https://github.com/lvgl/lvgl/issues/3269#issuecomment-1109934136) for more context on it.

mariotaku, 2022-05-28:
Thanks for the info. Maybe I'll need to abstract the part that it creates snapshot buffer (as SDL uses texture instead of bitmap), then support off-screen drawing to the snapshot.

kisvegabor, 2022-05-28:
I think the `draw_ctx` should have a callback like `draw_ctx->set_buf(draw_ctx, buf)` (probably with a better name). It can create a new texture, which uses this buffer. 
IIRC correctly now `disp_drv->user_data` is used to store SDL specific stuff. Would it be possible to store them in `draw_ctx->user_data`?

kisvegabor, 2022-05-29:
I've just realized probably there is no such a thing as a "buffer" as you use the Texture directly. Se the callback should be `draw_ctx->buf_alloc(draw_ctx, hor_res, ver_res)`

I still don't know how to blend the snapshot Texture as an image once it's rendered.

mariotaku, 2022-05-29:
For an object, rendering on a texture is as simple as rendering on screen. Once it's rendered, it is an image (but can't be modified pixel by pixel easily). I'll explore this part for a bit and see if I can make some changes.

kisvegabor, 2022-05-30:
I mean it's directly passed to `lv_draw_img` [here](https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c#L997).

mariotaku, 2022-05-30:
Oh! That's also fine. Underlying I have some logic to render SDL_Texture by passing to lv_draw_img.

kisvegabor, 2022-06-14:
Let's continue in the PR: https://github.com/lvgl/lvgl/pull/3403

closed 2022-06-14 by kisvegabor