lvgl/lvgl #3392: lv_arc drawing is not properly working in case of negative angle values

rmantovani68, 2022-05-31:
### Describe the bug
lv_arc drawing is not properly working in case of negative angle values

### To Reproduce
Draw an lv_arc setting a negative value in start or end angles

### Expected behavior
I expect to see the lv_arc well painted

### Screenshots or video

As you can see the arc is closing before  the correct point.
I've use SquareLine to reproduce the issue, but the same problem happens in real life appplication.

[image]
[image]

kisvegabor, 2022-06-02:
`lv_arct_set_angles` accepts `uint16_t` parameters. See 
https://github.com/lvgl/lvgl/blob/9ac8ce69f67234563d4254e29e1903a638bb8f4e/src/widgets/lv_arc.h#L106

So it's rather related to SquareLine as it shouldn't allow negative values values here. I added this problem to SquareLine bug tracker.

rmantovani68, 2022-06-02:
Ok

All clear

Thank you

closed 2022-06-19 by kisvegabor