lvgl/lvgl #3396: More efficient text rendering when utilizing 2DGPU

AfflatusX, 2022-06-01:
### Introduce the problem
With sufficient buffer, draw character does not seem to be very efficient. We have a UI with about 150 characters on it, and we're aiming to render the frame within 16ms. After benchmarking, I noticed about 4ms was spent on this particular routine:

https://github.com/lvgl/lvgl/blob/master/src/draw/sw/lv_draw_sw_letter.c#L288

We're using a 2DGPU and have sufficient RAM to hold the mask of the glyph, is there a way to simply use the glyph's mask (pre-generated using font-tool) and blend the whole character in one run?

### Examples and cases
1. create a couple of labels and arbitrary text, totaling about 150 characters in one screen.
2. count time spent for above routine

### Suggested solution
The font tool should be able to simply generate the mask buffer for a given character, then we can directly blend it using GPU if available. The mask buffer can be stored in flash so it won't cost extra memory during runtime.

kisvegabor, 2022-06-03:
As I'm thinking about it more, probably as it works now it's an unnecessary optimization. Rendering a 30x40 sized letter is only a 1.2kB buffer. So we could say that the `get_bitmap` callback should return a 8bit alpha map which can be immediately blended.

However, as it's an API break, we can do it only in v9. (v8.3 will be released in a few weeks and after that we start to work on v9).

I added a note about it in https://github.com/lvgl/lvgl/issues/3298

GorgonMeducer, 2022-06-11:
@kisvegabor do you remember that we have hacked the renderer and enabled 2DGPU to acceleration 8bit-Alpha-only.
As @AfflatusX mentioned, they have sufficient memory to hold alpha-mask of glyph, so why not using 8bit-alpha-only for this scenario?

kisvegabor, 2022-06-14:
The font converter can create A8 font bitmaps too but LVGL still tried to optimize the rendering (which probably makes the performance worse to saves a few bytes of RAM).

What I suggested is that if e.g. an A4 bitmap is used the font engine should convert it to A8, so LVGL will always work with A8 blending. If we consider letters more like images we can pass them through the image drawing API, so all image effects can be applied (e.g. rotation, zoom).

kisvegabor, 2023-08-16:
Added in `master` (v9-dev). 
In v9 the font engines should always return A8 bitmaps.

closed 2023-08-16 by kisvegabor