lvgl/lvgl #3400: png images stored as LV_IMG_SRC_VARIABLE are not rendered properly

moinmoin-sh, 2022-06-03:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

.png image created as .c file with the online image converter is not rendered properly.
The example `lv_example_png_1.c` shows this issue.

original image descriptor 
 ```c
const lv_img_dsc_t img_wink_png = {
    .header.always_zero = 0,
    .header.w = 50,
    .header.h = 50,
    .data_size = 5158,
    .header.cf = LV_IMG_CF_RAW_ALPHA,
    .data = img_wink_png_map,
};
```

if its is manually changed to 
```c
const lv_img_dsc_t img_wink_png = {
    .header.always_zero = 0,
    .header.w = 50,
    .header.h = 50,
    .data_size = 5158,
    .header.cf = LV_IMG_CF_TRUE_COLOR_ALPHA,
    .data = img_wink_png_map,
};
```
The image is displayed correctly.

Note: the original result from the online image converter sets `header.w` and `header.h` to `0` which results in nothing displayed. With .w and .h set properly, the corrupted images as shown below is displayed.
Only if .h , .w and .cf are set as above the image is correctly displayed.

### To Reproduce

relevant code portion is the first part that deals with the image stored as variable.

```c
#include "../../lv_examples.h"
#if LV_USE_PNG && LV_USE_IMG && LV_BUILD_EXAMPLES

/**
 * Open a PNG image from a file and a variable
 */
void lv_example_png_1(void)
{
    LV_IMG_DECLARE(img_wink_png);
    lv_obj_t * img;

    img = lv_img_create(lv_scr_act());
    lv_img_set_src(img, &img_wink_png);
    lv_obj_align(img, LV_ALIGN_LEFT_MID, 20, 0);

    img = lv_img_create(lv_scr_act());
    /* Assuming a File system is attached to letter 'A'
     * E.g. set LV_USE_FS_STDIO 'A' in lv_conf.h */
    lv_img_set_src(img, "A:lvgl/examples/libs/png/wink.png");
    lv_obj_align(img, LV_ALIGN_RIGHT_MID, -20, 0);
}

#endif
```

### Expected behavior

### Screenshots or video

[image]

moinmoin-sh, 2022-06-07:
#define `LV_COLOR_DEPTH `16

that's strange ??
Please note also, that the online image converter sets `header.w` and `header.h` to `0`

embeddedt, 2022-06-07:
The image converter will always set `w` and `h` to 0 if you convert an image using a RAW format. This is because the file contents are just stored within the array and not decoded and converted at all with that format.

moinmoin-sh, 2022-06-07:
Hm, but your data above  shows `h` and `w` set to 50. In my case nothing is displayed with `h` and `w` set to 0?

moinmoin-sh, 2022-06-07:
I have to apologize ... I got confused for some reason and double checked.

Actual status : 
`.header.cf = LV_IMG_CF_RAW_ALPHA,` **works**,  image is displayed 

BUT  with
```
   .header.w = 0,
   .header.h = 0,
```
 **Nothing** is displayed.

Sorry for the confusion.

kisvegabor, 2022-06-08:
I can confirm that it works like this.

@embeddedt would it be possible to set width/height too if the image can be opened by the converter?

kisvegabor, 2022-06-09:
It'm not sure it's consistent, but when LVGL sees an `lv_img_dsc_t` variable it assumes that the width and height are set there. It will work better with @X-Ryl669 [decoder updated](https://github.com/lvgl/lvgl/pull/3294) (planned for v8.3) so I wouldn't change LVGL for this short time. I think 99% of the problems (even in the earlier version) can be solved if the converter tries to extract the dimensions of the image.

X-Ryl669, 2022-06-09:
Yes, that's one of the issue I've come upon and I think, I've solved. The online converter tool shouldn't be necessary anymore for upcoming version, since the actual image content is used to extract metadata (like size & color format) and it does not rely on the online tool to build a functional header. Ideally, the raw image decoder will become optional and can be skipped from building since it's not very efficient to store a raw image in a binary.

kisvegabor, 2022-06-11:
> Ideally, the raw image decoder will become optional and can be skipped from building since it's not very efficient to store a raw image in a binary.

IMO the biggest advantage of the raw format is that it needs no RAM and still supports transformations.

moinmoin-sh, 2022-06-22:
May I ask what the path forward will be? How will the handling with .PNG images in the "upcoming version" be?

X-Ryl669, 2022-06-22:
In LVGL v9, LVGL will become smarter, so it'll better deal with encoded image file efficiently. You won't need to use the online converter, and your binary will be smaller since you'll only embed the encoded picture. The downside is that the image will take a bit more time to load at first and it'll use a bit of memory.

The online converter will only be useful if you need to perform transformation (rotation & zoom) on some encoded picture (like GIF or PNG), or if you are very limited in RAM (but not in flash) and you can have mappable flash space as RAM (not all system allow this).

moinmoin-sh, 2022-06-22:
That sounds very promissing! Would make my life easier. Hope v9 is available soon.

moinmoin-sh, 2022-06-24:
I'm really interested in this new feature, Is v9 available for testing?

kisvegabor, 2022-06-25:
We will release v8.3 in a few weeks and start to work on v9 after that. Although, the feature @X-Ryl669 mentioned is available in this PR: https://github.com/lvgl/lvgl/pull/3294

You can try it out with:
```git
git checkout -b X-Ryl669-imgdecoder arch/img-decode-rework
git pull https://github.com/X-Ryl669/lvgl.git imgdecoder
```

moinmoin-sh, 2022-06-25:
unfortunately that didn't work for me

```
git checkout -b X-Ryl669-imgdecoder arch/img-decode-rework
>>
fatal: 'arch/img-decode-rework' is not a commit and a branch 'X-Ryl669-imgdecoder' cannot be created from it
```

kisvegabor, 2022-06-28:
I see the same issue. @X-Ryl669 could you take a look at it?

embeddedt, 2022-06-28:
This should work?

```
git remote add imgdecoder https://github.com/X-Ryl669/lvgl.git
git fetch imgdecoder
git checkout -b X-Ryl669-imgdecoder imgdecoder/imgdecoder
```