lvgl/lvgl #3401: Segmentation fault on applying style (how to debug?)

elgerg, 2022-06-03:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Recently I added lv_led to my project (running very stable until then) and today multiple of my screens crashed and re-started. Luckily 
I had the simulator running in the VSCode environment and captured:

[image]

How and what can I do to debug this issue?

### To Reproduce
I'm not 100% sure why this didnt work but my code around this issue is as follows:
```c
			else if(strcmp(roomItems[i].ObjectType, "LED") == 0)
			{
				struct json_object *bool_struct;
				json_object_object_get_ex(parsed_json, "state", &bool_struct);
				bool contact = json_object_get_boolean(bool_struct);
				if(contact)
				{
					//true (closed)
					static lv_style_t style_led;
					lv_style_init(&style_led);
					lv_style_set_bg_color(&style_led, lv_palette_main(LV_PALETTE_GREEN));
					lv_obj_add_style(roomItems[i].LVObject, &style_led, LV_PART_MAIN); 
					lv_led_set_color(roomItems[i].LVObject, lv_palette_main(LV_PALETTE_GREEN));

				}
				else
				{
					//false (open)
					static lv_style_t style_led;
					lv_style_init(&style_led);
					lv_style_set_bg_color(&style_led, lv_palette_main(LV_PALETTE_RED));
					lv_obj_add_style(roomItems[i].LVObject, &style_led, LV_PART_MAIN); 
					lv_led_set_color(roomItems[i].LVObject, lv_palette_main(LV_PALETTE_RED));
				}
				json_object_put(bool_struct);
			}
```

### Expected behavior
Style to be applied and app not to crash

### Screenshots or video
As above

Thanks for the help!

embeddedt, 2022-06-03:
A few things:

* Do you ever remove the `style_led` style from the object? It looks like you could be applying it multiple times.
* You are initializing `style_led` over and over from what I can tell, but I don't see a call to `lv_style_reset`.

I think one of those is probably leading to corruption.

For your purposes here, I don't think it's worth using an actual `lv_style_t`; why not just use `lv_obj_set_style_local_bg_color`?

elgerg, 2022-06-03:
@embeddedt you are correct in that I am not removing the style.
For anyone else in this situation that finds this the correct call would be to lv_obj_set_style_bg_color.

Thank you for the pointer (no pun intended) :)

Thanks!

embeddedt, 2022-06-03:
Sorry, I still had the v7 name memorized. :laughing: Did that solve your problem?

elgerg, 2022-06-03:
Dont worry about it, I appreciate your help!
I think it might but only time will tell.
It crashed within 12 hours so I should know by tomorrow morning if it's still an issue..

Thanks again!

elgerg, 2022-06-04:
23 hours and no crashes. I'd say this is a win!
Thanks @embeddedt !!

closed 2022-06-04 by elgerg