lvgl/lvgl #3402: lru issues

kctan0805, 2022-06-05:
1. calling lv_lru_set() with smaller value_length than old one, it will for-loop inifinitely at 
https://github.com/lvgl/lvgl/blob/c9fa4ef0fcc9d226ca2a83a7c630d366d13a1ca7/src/misc/lv_lru.c#L191

cause: the type of required variable is size_t (unsigned type), it always passed the condition even it turn to negative value:
https://github.com/lvgl/lvgl/blob/c9fa4ef0fcc9d226ca2a83a7c630d366d13a1ca7/src/misc/lv_lru.c#L190

I workaround it by changing all size_t type of lv_lru.* to int, but not sure whether it is a good idea.

2. lv_lru_remove_lru_item() is decalared as static function:
https://github.com/lvgl/lvgl/blob/c9fa4ef0fcc9d226ca2a83a7c630d366d13a1ca7/src/misc/lv_lru.c#L309
but I need to call it outside when I allocate resources fail.

mariotaku, 2022-06-07:
@kisvegabor Thanks for letting me know. Are you OK with including this patch in #3403 too?

kctan0805, 2022-06-09:
It's work, thank you everyone ;)

closed 2022-06-09 by kctan0805