lvgl/lvgl #3410: Preemptive vs Cooperative Systems

ashrafkamel5, 2022-06-12:
### Description
if lvgl still requires no RTOS, then this is a **bug**.
if it is not, this is a discussion.

when using DMA in the `disp_flush` you don't call the `disp_flush_ready()` at the end of the flush function.
so lvgl will **block** on the flush flag! which is bad.
and counters the use of a DMA.

LVGL is offering a solution, `wait` call back function.
which you can only fill with a `yield()` to allow another task to work meanwhile.
and this is the problem.
LVGL here assumes you are running on a preemptive Kernal,
which it should not.

###solution
Lvgl task handler should be redesigned to be a state machine with no blocking.
it is a simple rule.
every blocking stage should be a new state, 
where it just checks for the blocking condition and returns or jumps to a different state.

embeddedt, 2022-06-12:
> LVGL here assumes you are running on a preemptive Kernal, which it should not.

Hmm, unless I misunderstood something, I don't see why `wait_cb` can't work for both cooperative and preemptive systems, as long as there is a way of yielding.

kisvegabor, 2022-06-14:
> unlike preemptive, cooperative systems cannot make context switching.
every task should be designed with the "run to completion" rule, which is "no blocking or waiting".

AFAIK it's no true.

See [here](https://www.geeksforgeeks.org/preemptive-and-non-preemptive-scheduling/):
> Non-preemptive Scheduling is used when a process terminates, or a process switches from running to the waiting state. In this scheduling, once the resources (CPU cycles) are allocated to a process, the process holds the CPU till it gets terminated or reaches a **waiting state.** 

Where "waiting state"  means yield, wait ticks/events/semaphores.

`lv_timer_handler` can do now what you described as "cooperative operating system" but I wouldn't call it an OS. It just calls functions after each other. 

Regardless to the terminology, I don't think it's a good idea let other `lv_timer`s runs while we are waiting for DMA. Let's say you need to wait 2 ms but the task you start takes 6ms. So you can lose a lot of performance. 

> so lvgl will block on the flush flag! which is bad.
and counters the use of a DMA.

The goal of using DMA is to let LVGL render while the previous rendered block is sent to the display (via DMA). It can happen that rendering takes e.g. 12 ms and sending takes 16 ms and you have to wait 4 ms  to sync them.

ashrafkamel5, 2022-06-14:
Let's keep the discussion about the OS type out of this😄.
And just address the issue of needing the content switching.
In my opinion, it is an easy task to achieve the no blocking mechanism, i think you designed the core of Lvgl the same way.
And also give the option to block if the user chooses to "to address the case Gabor mentioned"

And Could you @embeddedt give a hint or a source that would help with the timing measure?

closed 2022-07-07 by lvgl-bot

ashrafkamel5, 2022-07-17:
Hi Gabor,
sorry for the late response.
I intended to make the changes myself and continue the discussion with the pullrequest, where it would be much more apparent.

but, the task needed more work than I thought.
I thought I could just modify the functions that block with the flushing flag.
And the way of "fixing the blocking stage"
to make it a state machine.
ex:
This is the intended fix for `lv_refr_area_part`
```c
static void lv_refr_area_part(lv_draw_ctx_t * draw_ctx)
{
    typedef enum{
        STATE_READY,
        STATE_WAIT,
    }state_t;
    static state_t state = STATE_READY;
    switch (state)
    {
    case STATE_READY:
        lv_disp_draw_buf_t * draw_buf = lv_disp_get_draw_buf(disp_refr);

    /* Below the `area_p` area will be redrawn into the draw buffer.
     * In single buffered mode wait here until the buffer is freed.*/
    if(draw_buf->buf1 && !draw_buf->buf2) {
        if(draw_buf->flushing) 
        {
            state = STATE_WAIT;
            return;
        }
    }

    lv_obj_t * top_act_scr = NULL;
    lv_obj_t * top_prev_scr = NULL;

    /*Get the most top object which is not covered by others*/
    top_act_scr = lv_refr_get_top_obj(draw_ctx->buf_area, lv_disp_get_scr_act(disp_refr));
    if(disp_refr->prev_scr) {
        top_prev_scr = lv_refr_get_top_obj(draw_ctx->buf_area, disp_refr->prev_scr);
    }

    /*Draw a display background if there is no top object*/
    if(top_act_scr == NULL && top_prev_scr == NULL) {
        if(draw_ctx->draw_bg) {
            lv_draw_rect_dsc_t dsc;
            lv_draw_rect_dsc_init(&dsc);
            dsc.bg_img_src = disp_refr->bg_img;
            dsc.bg_img_opa = disp_refr->bg_opa;
            dsc.bg_color = disp_refr->bg_color;
            dsc.bg_opa = disp_refr->bg_opa;
            draw_ctx->draw_bg(draw_ctx, &dsc, draw_ctx->buf_area);
        }
        else if(disp_refr->bg_img) {
            lv_img_header_t header;
            lv_res_t res;
            res = lv_img_decoder_get_info(disp_refr->bg_img, &header);
            if(res == LV_RES_OK) {
                lv_area_t a;
                lv_area_set(&a, 0, 0, header.w - 1, header.h - 1);
                lv_draw_img_dsc_t dsc;
                lv_draw_img_dsc_init(&dsc);
                dsc.opa = disp_refr->bg_opa;
                lv_draw_img(draw_ctx, &dsc, &a, disp_refr->bg_img);
            }
            else {
                LV_LOG_WARN("Can't draw the background image");
            }
        }
        else {
            lv_draw_rect_dsc_t dsc;
            lv_draw_rect_dsc_init(&dsc);
            dsc.bg_color = disp_refr->bg_color;
            dsc.bg_opa = disp_refr->bg_opa;
            lv_draw_rect(draw_ctx, &dsc, draw_ctx->buf_area);
        }
    }
    /*Refresh the previous screen if any*/
    if(disp_refr->prev_scr) {
        if(top_prev_scr == NULL) top_prev_scr = disp_refr->prev_scr;
        lv_refr_obj_and_children(draw_ctx, top_prev_scr);
    }

    if(top_act_scr == NULL) top_act_scr = disp_refr->act_scr;
    lv_refr_obj_and_children(draw_ctx, top_act_scr);

    /*Also refresh top and sys layer unconditionally*/
    lv_refr_obj_and_children(draw_ctx, lv_disp_get_layer_top(disp_refr));
    lv_refr_obj_and_children(draw_ctx, lv_disp_get_layer_sys(disp_refr));

    /*In true double buffered mode flush only once when all areas were rendered.
     *In normal mode flush after every area*/
    if(disp_refr->driver->full_refresh == false) {
        draw_buf_flush(disp_refr);
    }
    break;
    case STATE_WAIT:
        if(draw_buf->flushing) {
            if(disp_refr->driver->wait_cb) disp_refr->driver->wait_cb(disp_refr->driver);
        }
        else
        {
            state = STATE_READY;
        }
    break;
    }
}
```
the changes
[image]
[image]

why it doesn't work, is because the rest of the lvgl SRC would assume that the function finished normally.
so it should be a bigger fix, where the whole lvgl system doesn't block and is designed as a state machine rather than a sequential code.

IMHO, it should be redesigned like this, not just for plain c systems without an RTOS, but also for a failer cases with the flush.
what if the flush never returned.
instead of looping forever. there should be a failure mechanism.

I hope my point is clearer now than before,
and I want to hear your thoughts.

kisvegabor, 2022-07-18:
Sorry for saying this, but I myself can't see as much added value as much work it'd be to rework it. It seems it's not an issue for others either as it wasn't reported yet. :slightly_frowning_face:

ashrafkamel5, 2022-07-20:
that is unfortunate to hear.
I hope you make a place for it one day.

if there is a way that would make it easier for me to understand the lvgl core,
I would participate in such significant changes.