lvgl/lvgl #3414: Using sw_rotate with SSD1306 shield causes memory fault

richesonk, 2022-06-15:
### Perform all steps below and tick them with [x]
- [x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)
Reproducing the issue in the simulator is non-trivial. Please see below for information on the issue and steps to reproduce the issue on an MCU development kit.

### Describe the bug
Using the new `sw_rotate` feature from LVGL with the Zephyr RTOS LVGL sample code and an ssd1306_128x64 shield causes a memory fault.

The issue seems to be related to the draw_buf_rotate...() functions not accounting for the possibility that a custom set_px_cb (such as lvgl_set_px_cb_mono() function from Zephyr maybe packing the data as bits instead of full bytes. This results in out of bounds array accesses in the rotation functions and causes data corruption in other (likely display related) areas of memory.

### To Reproduce
I found/can reproduce this issue using the Zephyr RTOS based nRF Connect SDK. Unfortunatley, porting this to a simulator based example is non-trivial. The essence of the issue though is using a function like the Zephyr provided `lvgl_set_px_cb_mono()` to pack the pixels into bits, then using SW rotation which (I think) assumes the pixels are packed as bytes.

To reproduce the issue using the nRF Connect SDK and an nRF5340 Dev Kit follow the following steps.
1. Edit modules/lib/gui/lvgl/zephyr/lvgl.c adding the following lines immediately after the call to `lv_disp_drv_init(&disp_drv);` in `lvgl_init()`:
```c
	disp_drv.sw_rotate = 1;
	disp_drv.rotated = LV_DISP_ROT_90;
```
2. Create samples/subsys/display/lvgl/boards/nrf5340dk_nrf5340_cpuapp.overlay file with the following snippet (only needed for the sample to work with the nRF5340 dev kit).
```
&arduino_i2c {
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;
	zephyr,concat-buf-size = <1025>;
};
```
3. Build the LVGL display sample for a supported board that also supports the `ssd1306_128x64` shield. I am using a Nordic development kit and the Zephyr based nRF Connect SDK, so my build command is:
```
west build zephyr/samples/subsys/display/lvgl -b nrf5340dk_nrf5340_cpuapp  -- -DSHIELD=ssd1306_128x64
```
4. Flash the built code to the board (with and SSD1306 shield attached)
5. Observer the failure (memfault shown on the console output, or obvious rebooting causing display to flash)

### Expected behavior
Normal display of the LVGL display sample rotated on the display by 90 degrees (a.k.a. portrait mode instead of landscape)

### Screenshots or video
N/A

richesonk, 2022-06-15:
Originally opened in the Zephyr project (https://github.com/zephyrproject-rtos/zephyr/issues/46446) but discussion there requested this be tracked here in LVGL.

kisvegabor, 2022-06-19:
> The issue seems to be related to the draw_buf_rotate...() functions not accounting for the possibility that a custom set_px_cb (such as lvgl_set_px_cb_mono() function from Zephyr maybe packing the data as bits instead of full bytes. 

That's true, it's really not handled properly. We can either add an option for a custom display rotate function or figure out something more generic. 

With a custom rotate function I'm afraid we will end up having a lot of callback for special cases. In an other issue (from [here](https://github.com/lvgl/lvgl/issues/3379#issuecomment-1147954592)) we were talking about the concept of `buffer_post_process_cb`. This way we made LVGL work only with standard (A)RGB formats. So `set_px` were removed and LVGL rendered everything at smallest in RGB232. This format could be converted and rotated into other formats before flushing. 

This way a screen sized buffer for  128x64 display needed 8kB buffer. However you can set smaller buffers, e.g. 128x16 -> 2kB, so I think the memory overhead is not an issue. The performance will be even better, as it's much faster to render in standard formats and convert the result only once.

As these are API breaking changes, all these can be added only in v9. (v8.3 will be released in a few weeks an after it we start on work on v9)

For now, you can do something similar to what I described above:
- Set `LV_COLOR_DEPTH   8`
- Let LVGL rotate the rendered image
- Convert he buffer to monochrome in your `flush_cb`

cc @GorgonMeducer

brgl, 2022-06-20:
> > For now, you can do something similar to what I described above:
> > ```
> > * Set `LV_COLOR_DEPTH   8`
> > * Let LVGL rotate the rendered image
> > * Convert he buffer to monochrome in your `flush_cb`
> > ```
> 
> @brgl is this something that could be done in the Zephyr specific code in support of the SSD1306 shields, or will Zephyr likely wait for the v9 changes and integrate then?

I would prefer to avoid extending LVGL locally in zephyr with any new interfaces. Adding this as a fix specific to SSD1306 sounds like a better idea.

richesonk, 2022-06-21:
> I would prefer to avoid extending LVGL locally in zephyr with any new interfaces. Adding this as a fix specific to SSD1306 sounds like a better idea.

@brgl any thoughts on if/how this will get integrated into Zephyr? I don't have any preference, I just need to understand how much work I need to plan for my project vs. how much support will be updated/added to Zephyr.

richesonk, 2022-06-22:
@kisvegabor and @brgl not sure if this is useful to you or not, but here is a patch where I incorporated the idea mentioned above into my project. I tried to add appropriate configurability, but not sure if I got it all in line with what you're thinking.
```
diff --git a/zephyr/Kconfig.graphical b/zephyr/Kconfig.graphical
index 2f08481b..af31a826 100644
--- a/zephyr/Kconfig.graphical
+++ b/zephyr/Kconfig.graphical
@@ -30,4 +30,33 @@ config LV_Z_VER_RES_MAX
 	help
 	  Vertical screen resolution in pixels
 
+config LV_DISP_SW_ROTATE
+	bool "Use SW rotation from the LVGL library"
+	default false
+	help
+	  Enable software rotation from the display driver.
+
+choice LV_DISP_ROT
+	prompt "Display rotation"
+	default LV_DISP_ROT_NONE
+	help
+	  Rotation applied to the display orientation
+
+	config LV_DISP_ROT_NONE
+		bool "0: Display is not rotated"
+	config LV_DISP_ROT_90
+		bool "1: Display rotated 90 degrees"
+	config LV_DISP_ROT_180
+		bool "2: Display rotated 180 degrees"
+	config LV_DISP_ROT_270
+		bool "3: Display rotated 270 degrees"
+endchoice
+
+config LV_DISP_ROT
+	int
+	default 0 if LV_DISP_ROT_NONE
+	default 1 if LV_DISP_ROT_90
+	default 2 if LV_DISP_ROT_180
+	default 3 if LV_DISP_ROT_270
+
 endmenu
diff --git a/zephyr/lvgl.c b/zephyr/lvgl.c
index 8853f9f0..c20e54f7 100644
--- a/zephyr/lvgl.c
+++ b/zephyr/lvgl.c
@@ -331,6 +331,8 @@ static int lvgl_init(const struct device *dev)
 #endif
 
 	lv_disp_drv_init(&disp_drv);
+	disp_drv.sw_rotate = CONFIG_LV_DISP_SW_ROTATE ? 1 : 0;
+	disp_drv.rotated = CONFIG_LV_DISP_ROT;
 	disp_drv.user_data = (void *) display_dev;
 
 	err = lvgl_allocate_rendering_buffers(&disp_drv);
diff --git a/zephyr/lvgl_display.c b/zephyr/lvgl_display.c
index 3581f651..fac7188a 100644
--- a/zephyr/lvgl_display.c
+++ b/zephyr/lvgl_display.c
@@ -42,8 +42,13 @@ int set_lvgl_rendering_cb(lv_disp_drv_t *disp_drv)
 	case PIXEL_FORMAT_MONO01:
 	case PIXEL_FORMAT_MONO10:
 		disp_drv->flush_cb = lvgl_flush_cb_mono;
+#ifdef CONFIG_LV_COLOR_DEPTH_1
 		disp_drv->rounder_cb = lvgl_rounder_cb_mono;
 		disp_drv->set_px_cb = lvgl_set_px_cb_mono;
+#else
+		disp_drv->rounder_cb = NULL;
+		disp_drv->set_px_cb = NULL;
+#endif
 		break;
 	default:
 		disp_drv->flush_cb = NULL;
diff --git a/zephyr/lvgl_display_mono.c b/zephyr/lvgl_display_mono.c
index ebdc6f7a..2142d739 100644
--- a/zephyr/lvgl_display_mono.c
+++ b/zephyr/lvgl_display_mono.c
@@ -13,6 +13,32 @@ void lvgl_flush_cb_mono(lv_disp_drv_t *disp_drv,
 {
 	uint16_t w = area->x2 - area->x1 + 1;
 	uint16_t h = area->y2 - area->y1 + 1;
+	void *buf_p = (void *)color_p;
+
+#ifndef CONFIG_LV_COLOR_DEPTH_1
+	uint8_t buf[CONFIG_LV_Z_HOR_RES_MAX * CONFIG_LV_Z_VER_RES_MAX / 8];
+	buf_p = (void *)buf;
+
+	uint16_t y_idx = 0;
+	while (y_idx < h)
+	{
+		for (uint16_t x = 0; x < w; x++)
+		{
+			uint16_t buf_idx = x + ((y_idx/8) * w);
+			uint8_t  buf_data = 0;
+			for (uint16_t y = 0; y < MIN(8, (h - y_idx)); y++)
+			{
+				lv_color_t *pPixel = color_p + x;
+				pPixel += ((y_idx + y) * w);
+				if (pPixel->full == 0xff)
+					buf_data |= (1 << y);
+			}
+			buf[buf_idx] = buf_data;
+		}
+		y_idx += 8;
+	}
+#endif
+
 	const struct device *display_dev = (const struct device *)disp_drv->user_data;
 	struct display_capabilities cap;
 	struct display_buffer_descriptor desc;
@@ -23,10 +49,10 @@ void lvgl_flush_cb_mono(lv_disp_drv_t *disp_drv,
 	desc.width = w;
 	desc.pitch = w;
 	desc.height = h;
-	display_write(display_dev, area->x1, area->y1, &desc, (void *) color_p);
+	display_write(display_dev, area->x1, area->y1, &desc, buf_p);
 	if (cap.screen_info & SCREEN_INFO_DOUBLE_BUFFER) {
 		display_write(display_dev, area->x1, area->y1, &desc,
-				(void *) color_p);
+				buf_p);
 	}
 
 	lv_disp_flush_ready(disp_drv);
```
I'll also add a patch to the linked issue over in Zephyr that updates the SSD1306 config to enable SW rotation for that shield by default and provide an idea of how to configure a project to use the rotation.

richesonk, 2022-06-23:
Sorry for the long posts but, here's a minor update to the patch to properly handle monochrome rounding when rotation is enabled:
```
diff --git a/zephyr/Kconfig.graphical b/zephyr/Kconfig.graphical
index 2f08481b..af31a826 100644
--- a/zephyr/Kconfig.graphical
+++ b/zephyr/Kconfig.graphical
@@ -30,4 +30,33 @@ config LV_Z_VER_RES_MAX
 	help
 	  Vertical screen resolution in pixels
 
+config LV_DISP_SW_ROTATE
+	bool "Use SW rotation from the LVGL library"
+	default false
+	help
+	  Enable software rotation from the display driver.
+
+choice LV_DISP_ROT
+	prompt "Display rotation"
+	default LV_DISP_ROT_NONE
+	help
+	  Rotation applied to the display orientation
+
+	config LV_DISP_ROT_NONE
+		bool "0: Display is not rotated"
+	config LV_DISP_ROT_90
+		bool "1: Display rotated 90 degrees"
+	config LV_DISP_ROT_180
+		bool "2: Display rotated 180 degrees"
+	config LV_DISP_ROT_270
+		bool "3: Display rotated 270 degrees"
+endchoice
+
+config LV_DISP_ROT
+	int
+	default 0 if LV_DISP_ROT_NONE
+	default 1 if LV_DISP_ROT_90
+	default 2 if LV_DISP_ROT_180
+	default 3 if LV_DISP_ROT_270
+
 endmenu
diff --git a/zephyr/lvgl.c b/zephyr/lvgl.c
index 8853f9f0..c20e54f7 100644
--- a/zephyr/lvgl.c
+++ b/zephyr/lvgl.c
@@ -331,6 +331,8 @@ static int lvgl_init(const struct device *dev)
 #endif
 
 	lv_disp_drv_init(&disp_drv);
+	disp_drv.sw_rotate = CONFIG_LV_DISP_SW_ROTATE ? 1 : 0;
+	disp_drv.rotated = CONFIG_LV_DISP_ROT;
 	disp_drv.user_data = (void *) display_dev;
 
 	err = lvgl_allocate_rendering_buffers(&disp_drv);
diff --git a/zephyr/lvgl_display.c b/zephyr/lvgl_display.c
index 3581f651..568f3570 100644
--- a/zephyr/lvgl_display.c
+++ b/zephyr/lvgl_display.c
@@ -43,7 +43,11 @@ int set_lvgl_rendering_cb(lv_disp_drv_t *disp_drv)
 	case PIXEL_FORMAT_MONO10:
 		disp_drv->flush_cb = lvgl_flush_cb_mono;
 		disp_drv->rounder_cb = lvgl_rounder_cb_mono;
+#ifdef CONFIG_LV_COLOR_DEPTH_1
 		disp_drv->set_px_cb = lvgl_set_px_cb_mono;
+#else
+		disp_drv->set_px_cb = NULL;
+#endif
 		break;
 	default:
 		disp_drv->flush_cb = NULL;
diff --git a/zephyr/lvgl_display_mono.c b/zephyr/lvgl_display_mono.c
index ebdc6f7a..c14b2770 100644
--- a/zephyr/lvgl_display_mono.c
+++ b/zephyr/lvgl_display_mono.c
@@ -13,6 +13,32 @@ void lvgl_flush_cb_mono(lv_disp_drv_t *disp_drv,
 {
 	uint16_t w = area->x2 - area->x1 + 1;
 	uint16_t h = area->y2 - area->y1 + 1;
+	void *buf_p = (void *)color_p;
+
+#ifndef CONFIG_LV_COLOR_DEPTH_1
+	uint8_t buf[CONFIG_LV_Z_HOR_RES_MAX * CONFIG_LV_Z_VER_RES_MAX / 8];
+	buf_p = (void *)buf;
+
+	uint16_t y_idx = 0;
+	while (y_idx < h)
+	{
+		for (uint16_t x = 0; x < w; x++)
+		{
+			uint16_t buf_idx = x + ((y_idx/8) * w);
+			uint8_t  buf_data = 0;
+			for (uint16_t y = 0; y < MIN(8, (h - y_idx)); y++)
+			{
+				lv_color_t *pPixel = color_p + x;
+				pPixel += ((y_idx + y) * w);
+				if (pPixel->full == 0xff)
+					buf_data |= (1 << y);
+			}
+			buf[buf_idx] = buf_data;
+		}
+		y_idx += 8;
+	}
+#endif
+
 	const struct device *display_dev = (const struct device *)disp_drv->user_data;
 	struct display_capabilities cap;
 	struct display_buffer_descriptor desc;
@@ -23,10 +49,10 @@ void lvgl_flush_cb_mono(lv_disp_drv_t *disp_drv,
 	desc.width = w;
 	desc.pitch = w;
 	desc.height = h;
-	display_write(display_dev, area->x1, area->y1, &desc, (void *) color_p);
+	display_write(display_dev, area->x1, area->y1, &desc, buf_p);
 	if (cap.screen_info & SCREEN_INFO_DOUBLE_BUFFER) {
 		display_write(display_dev, area->x1, area->y1, &desc,
-				(void *) color_p);
+				buf_p);
 	}
 
 	lv_disp_flush_ready(disp_drv);
@@ -82,6 +108,9 @@ void lvgl_rounder_cb_mono(lv_disp_drv_t *disp_drv,
 {
 	const struct device *display_dev = (const struct device *)disp_drv->user_data;
 	struct display_capabilities cap;
+	bool is_vtiled  = (cap.screen_info & SCREEN_INFO_MONO_VTILED);
+	bool is_rotated = ((disp_drv->rotated == LV_DISP_ROT_90) ||
+	                   (disp_drv->rotated == LV_DISP_ROT_270));
 
 	display_get_capabilities(display_dev, &cap);
 
@@ -89,7 +118,7 @@ void lvgl_rounder_cb_mono(lv_disp_drv_t *disp_drv,
 		area->x1 = 0;
 		area->x2 = cap.x_resolution - 1;
 	} else {
-		if (cap.screen_info & SCREEN_INFO_MONO_VTILED) {
+		if (is_vtiled != is_rotated) {
 			area->y1 &= ~0x7;
 			area->y2 |= 0x7;
 		} else {
```

kisvegabor, 2022-07-09:
No problem. :slightly_smiling_face: 

As this should be solved in Zephyr's driver side we can't do much here. I close this issue for now. If anything comes up, of course we can reopen it.

closed 2022-07-09 by kisvegabor