lvgl/lvgl #3415: Improved image cache mechanism for large memory devices

FASTSHIFT, 2022-06-16:
### Introduce the problem
Require:
* 1.The frame rate is stable, and it is not allowed to temporarily load pictures during the sliding process to cause frame drops.
* 2.Use as much memory space as possible to buffer images to reduce image loading and decoding.
* 3.When memory allocation fails due to memory fragmentation, part of the cache is automatically cleared.

### Examples and cases
Hardware condition:
* 1.Have 8~32MB of memory, and the memory bandwidth is high enough.
* 2.It can support buffering complete pictures, and the full picture buffering of a single active page will not exceed the upper limit of available memory.

### Suggested solution
* 1.Use a queue to manage the image cache, replacing the old aging algorithm. Every time the page is loaded, `_lv_img_cache_open()` is called to load all images(such as PNG) of the page. When drawing an image, put the cache corresponding to the image to the head of the queue to ensure that the latest referenced images are in the front position.
* 2.The image cache uses an independent heap. When the cache size exceeds the set upper limit or a malloc failure occurs during image decoding, the cache clears the cache from the end of the queue, and retries to decode the image until the decoding is successfully added to the head of the cache queue.
* 3.When the image decoding fails, the return value of `lv_img_decoder_open()` needs to explain more detailed failure reasons, such as image resource error or insufficient memory caused by `malloc()` failure, and notify the cache manager to adjust the release strategy.

FASTSHIFT, 2022-06-17:
@kisvegabor @X-Ryl669

X-Ryl669, 2022-06-17:
Solutions:
1. Ok, right now it's based on effort to load (how much time it took to load&decode the picture), not last access time.
2. Agree. I've multiple ideas here to improve on this scheme (see next post)
3. Might be possible, but not always. For example, ffmpeg can fail while decoding the n-th frame due to lack of memory, but that doesn't mean you should release the previous decoding, since it contains the motion vectors and context required for the n-th frame. I don't know if in this case, it wouldn't be better to let the decoder callback the heap function to fix itself its error instead of relying on the cache. 

Just a remark: for animated format (rlottie, ffmpeg, gif), the decoder's context is captured and managed by the image (lv_img), not the cache, to avoid paying the cost of searching the cache for each frame to decode and recreating a context. It's not the same as the rendering area/buffer that might or might not be managed by the cache.

There was a discussion about the heap management in LVGL, and I think we didn't conclude on what to do here, but it's intrinsically related to the cache behavior.

For example, for optimal (best) experience on ESP32 and rlottie, I've measured this behavior:
1. Parsing the lottie's json file should happen on ESP32 internal memory (SRAM). If using PSRAM (SPI RAM) then it's 30x slower due to random access. Once the file is parsed, it's possible to copy the structure to PSRAM for storage and back to SRAM when rendering.
2. Rendering should happen with the context in SRAM, but the output rendering buffer can be in PSRAM (only 25% slower than SRAM since it's accessed mainly as a stream, so cache works well).

So, based on the above pattern, we can consider to have 2 heaps (slow & big & storage, fast & small & limited), and manage to move context from/to both heap depending on the usage scenario.
In that case, the image cache could manage both heap (based on time to decode for the fast heap and last access time for the slow heap).

Now, we had also talked about GPU constraint, and I'll let the main developer correct me here, but the image cache could also have to deal with the GPU memory (typically by managing texture id, creation and destruction).

X-Ryl669, 2022-06-17:
I don't know where such policy should be written, either in the decoder (like it's rlottie specific?) or the cache (it's the same behavior for all decoder ?). We also have to converge to how to define capabilities for heaps:
1. Should LVGL manage the heap (this implies memory lost to TLSF's allocator headers)
2. Let the implementation be platform specific ?
3. What are the flags required ? 
4. What are the possible optimization ?
5. Should heap communicate or be completely isolated ? (so moving means copying instead of just passing pointer to structures)
6. Is an API (with callback) required (for example, to cudaMemcpy from host to GPU) and if yes, should it be synchronous or asynchronous ?
7. Who is responsible for heap cleaning upon crash ? If it's internal heap, or external heap, this is important to have some fallback to recover in that case
8. etc...

FASTSHIFT, 2022-06-17:
@X-Ryl669 
I think that only static images need to be considered for the cache manager, and the life cycle of dynamic images should be managed directly by the page, which is the easiest and less prone to problems.

FASTSHIFT, 2022-06-17:
How to separate cache management from draw and decoder? Different hardware platforms have different cache algorithms.

kisvegabor, 2022-06-19:
The conclusion in the [previous cache related discussion](https://github.com/lvgl/lvgl/issues/3116) was that there should be a global cache as it can be more flexible and there are many things to cache:
- images
- letters
- gradients
- primitives (circle, shadow calculation)
- file system (e.g. fro `lv_fs_read`)

We were talking about allowing dynamically limiting the max and min size of each pool and on `alloc_from_cache` flags can be passed to tell what kind of memory do we need (large but slow, fast but small, DMA-able, temporary etc). It was also mentioned that there should be a `free_cb` for each cache entry. The callback would be called if that entry is freed to let e.g. the decoder close that image, close the file, free other memories, etc. 

But it's only one topic (i.e. in which memory the cache should be stored). Finding suitable a cache-clear policy is an other topic. I.e.  if 5 image is already cached, which one(s) to free to have space for a new one. 

As I see these 2 topics can be independent. For example if each entry has a data like `life` the cache manager can compare the `life`s to find out which entry to free. Figuring out how to set `life` can be discussed separately. 

I'd focus on the alloc from cache first as the `life` can be built on top of it. We should let the users to create their own memory managers with any number of cache types (image, file, letter, circle, gradient, etc) and we should provide a simple built-in solution (which can ignore flags, just malloc the memory from a pool).

FASTSHIFT, 2022-06-21:
> The conclusion in the [previous cache related discussion](https://github.com/lvgl/lvgl/issues/3116) was that there should be a global cache as it can be more flexible and there are many things to cache:
> 
> * images
> * letters
> * gradients
> * primitives (circle, shadow calculation)
> * file system (e.g. fro `lv_fs_read`)
> 
> We were talking about allowing dynamically limiting the max and min size of each pool and on flags can be passed to tell what kind of memory do we need (large but slow, fast but small, DMA-able, temporary etc). It was also mentioned that there should be a for each cache entry. The callback would be called if that entry is freed to let e.g. the decoder close that image, close the file, free other memories, etc.`alloc_from_cache``free_cb`
> 
> But it's only one topic (i.e. in which memory the cache should be stored). Finding suitable a cache-clear policy is an other topic. I.e. if 5 image is already cached, which one(s) to free to have space for a new one.
> 
> As I see these 2 topics can be independent. For example if each entry has a data like the cache manager can compare the s to find out which entry to free. Figuring out how to set can be discussed separately.`life``life``life`
> 
> I'd focus on the alloc from cache first as the can be built on top of it. We should let the users to create their own memory managers with any number of cache types (image, file, letter, circle, gradient, etc) and we should provide a simple built-in solution (which can ignore flags, just malloc the memory from a pool).`life`

Can we start with the image cache? Start by separating the cache manager from draw and decoder.

kisvegabor, 2022-06-23:
> Can we start with the image cache? Start by separating the cache manager from draw and decoder.

Sounds good.  But the API should be prepared to handle different data as well.

kisvegabor, 2022-07-14:
I added this topic to v9 [ROADMAP](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md).

FASTSHIFT, 2023-08-08:
We have implemented our own set of img cache management mechanism on the product. After actual verification, it can effectively solve the problem of memory fragmentation. The following is the implementation principle:

1. Combining img cache management and heap management, img cache is divided into an independent heap for management, and the following APIs are provided:

```c

/**
 * Initialize image cache heap.
 *
 * @param entry_size Maximum number of image caches.
 * @param mem_size The total size of the cache memory in bytes.
 */
void lv_img_cache_lru_init(uint16_t entry_size, uint32_t mem_size);

/* Memory allocation related functions */
void *lv_img_cache_malloc(size_t size);
void *lv_img_cache_aligned_alloc(size_t align, size_t size);
void *lv_img_cache_realloc(void *data_p, size_t new_size);
void lv_img_cache_free(void *ptr);
```

2. Added a set of APIs for applying for image memory in `lv_mem`:
```c
void * lv_mem_alloc_draw_buf(size_t size)
{
    return LV_MEM_CUSTOM_ALLOC_DRAW_BUF(size);
}

void lv_mem_free_draw_buf(void * buf)
{
    LV_MEM_CUSTOM_FREE_DRAW_BUF(buf);
}

void * lv_mem_realloc_draw_buf(void * buf, size_t new_size)
{
    return LV_MEM_CUSTOM_REALLOC_DRAW_BUF(buf, new_size);
}

void * lv_mem_aligned_alloc_draw_buf(size_t align, size_t size)
{
    return LV_MEM_CUSTOM_ALIGNED_ALLOC_DRAW_BUF(align, size);
}
```

3. Use define in Makefile to map to img cache dedicated memory allocation interface:
```makefile
ifneq ($(LV_IMG_CACHE_DEF_SIZE),0)
CSRCS += lv_porting/lv_img_cache_lru.c
ifneq ($(CONFIG_LV_MEM_DRAW_BUF_CUSTOM),)
CFLAGS += "-DLV_MEM_CUSTOM_ALLOC_DRAW_BUF=lv_img_cache_malloc"
CFLAGS += "-DLV_MEM_CUSTOM_FREE_DRAW_BUF=lv_img_cache_free"
CFLAGS += "-DLV_MEM_CUSTOM_REALLOC_DRAW_BUF=lv_img_cache_realloc"
CFLAGS += "-DLV_MEM_CUSTOM_ALIGNED_ALLOC_DRAW_BUF=lv_img_cache_aligned_alloc"
endif
endif
```

4. Modify the decoder memory allocation function, take PNG as an example:
```c
#ifdef LODEPNG_COMPILE_ALLOCATORS
static void* lodepng_malloc(size_t size) {
#ifdef LODEPNG_MAX_ALLOC
  if(size > LODEPNG_MAX_ALLOC) return 0;
#endif
  return lv_mem_aligned_alloc_draw_buf(LV_ATTRIBUTE_MEM_ALIGN_SIZE, size);
}

/* NOTE: when realloc returns NULL, it leaves the original memory untouched */
static void* lodepng_realloc(void* ptr, size_t new_size) {
#ifdef LODEPNG_MAX_ALLOC
  if(new_size > LODEPNG_MAX_ALLOC) return 0;
#endif
  return lv_mem_realloc_draw_buf(ptr, new_size);
}

static void lodepng_free(void* ptr) {
  lv_mem_free_draw_buf(ptr);
}
```

This is the core logic of img cache elimination. When the memory application fails, the cache manager will try to close a decoder that is least commonly used to free up memory. If the application still fails, it will continue to close the decoder until the application is successful or all decoders are closed.

Judging whether malloc is successful to release cache can have these benefits:
- Can make full use of the memory space, can almost fill the img cache heap, and cache the maximum number of pictures.
- It can avoid the problem of large buffer allocation failure caused by memory fragmentation.
- The decoder can not estimate the memory usage, and is very friendly to the compression format.

```c
/****************************************************************************
 * Name: img_cache_alloc
 ****************************************************************************/

static FAR void *img_cache_alloc(size_t align, size_t size, bool re, FAR
                                 void *ptr)
{
  LV_LOG_INFO("img_cache_alloc (align: %zu, size: %zu, re: %d, ptr: %p)",
              align, size, (int)re, ptr);

  if (size > cache_manager.heap_size)
    {
      LV_LOG_WARN("Image cache memory is not enough for size %zu",
                  size);
      return NULL;
    }

  while (1)
    {
      FAR void *mem;

      if (re)
        {
          mem = mm_realloc(cache_manager.heap, ptr, size);
        }
      else
        {
          if (align != IMG_CACHE_DONT_ALIGN)
            {
              mem = mm_memalign(cache_manager.heap, align, size);
            }
          else
            {
              mem = mm_malloc(cache_manager.heap, size);
            }
        }

      if (mem)
        {
          LV_LOG_INFO("Image malloc addr: %p success", mem);
          return mem;
        }

      LV_LOG_INFO(
        "Image malloc size: %zu failed, "
        "try to free the memory of the tail node.",
        size);

      /* If the memory is not enough, try to free the memory of the
       * tail node.
       * */

      if (!remove_cache_tail_entry())
        {
          break;
        }
    }

  LV_LOG_WARN("Image cache memory is not enough for size %zu",
              size);
  return NULL;
}
```

This is the complete code file:
https://gist.github.com/FASTSHIFT/915b92eb86ed0af97b10110a2eb6f671

kisvegabor, 2023-08-09:
Hi,

Thank you for sharing! I think it'd be great to use the `lv_draw_buf_...` API instead of  `lv_img_cache_malloc/free/realloc`. It would immediately solve the stride and mem align issues too.

What do you think?

FASTSHIFT, 2023-08-10:
> Hi,
> 
> Thank you for sharing! I think it'd be great to use the `lv_draw_buf_...` API instead of `lv_img_cache_malloc/free/realloc`. It would immediately solve the stride and mem align issues too.
> 
> What do you think?

I think it may be necessary to divide the API into two layers. `lv_draw_buf_xxx` is responsible for solving the problem of color format and stride calculation, and `lv_img_cache_malloc` is responsible for applying for large memory and solving the problem of memory fragmentation. `lv_draw_buf_xxx` can switch `lv_malloc` or `lv_img_cache_malloc` through a layer of wrapper. Use `lv draw_buf_xxx` for buffer allocation that requires color and stride information, and use `lv_img_cache_malloc` for ordinary large buffer allocation. All objects applied for using `lv_img_cache_malloc` need to be controlled by the decoder, so you can release the memory by closing the decoder at any time.
On embedded devices with limited memory, large memory management is a difficult problem, and we have to consider various situations.

kisvegabor, 2023-08-10:
My idea was that a custom `lv_draw_buf` implementation internally can handle large allocation request differently.  Or we can add some flags to the `lv_draw_buf_t`. 

I feel like `lv_img_cache_malloc` and `lv_draw_buf` are very similar and it would be great to unify them somehow.

FASTSHIFT, 2023-08-14:
> My idea was that a custom implementation internally can handle large allocation request differently. E.g.`lv_draw_buf`
> 
> ```c
> if(size > 1024 * 256) large_malloc();
> else normal_malloc();
> ```
> 
> Or we can add some flags to the to explictly tell what to do (I would prefer not having flags though).`lv_draw_buf_t`
> 
> I feel like and are very similar and it would be great to unify them somehow.`lv_img_cache_malloc``lv_draw_buf`

I think adding a threshold to select the memory allocation method should be a good choice, but how can we ensure that the memory can be freed to the corresponding heap when the memory is free?

kisvegabor, 2023-08-15:
We can handle it via some flags. I'll add some update the `draw_buf` this week.

kisvegabor, 2023-10-26:
Closing as it should be implemented already.