lvgl/lvgl #3425: Can't compile LVGL on ARM Cortex-M0+

gmarull, 2022-06-23:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version (using Zephyr)
- [X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

LVGL fails to compile on ARM Cortex-M0+ (Zephyr SDK, GCC 10.3.0)

```
...
modules/lvgl/CMakeFiles/..__modules__lib__gui__lvgl__zephyr.dir/__w/zephyr/modules/lib/gui/lvgl/src/core/lv_event.c.obj -c /__w/zephyr/modules/lib/gui/lvgl/src/core/lv_event.c
/tmp/ccc5rHmt.s: Assembler messages:
/tmp/ccc5rHmt.s: Error: unaligned opcodes detected in executable segmen
```

It doesn't happen if I apply this patch:

```diff
diff --git a/src/core/lv_event.c b/src/core/lv_event.c
index 143ff34b8..3e6d3bb46 100644
--- a/src/core/lv_event.c
+++ b/src/core/lv_event.c
@@ -519,7 +519,7 @@ static bool event_is_bubbled(lv_event_t * e)
         case LV_EVENT_CHILD_CHANGED:
         case LV_EVENT_SIZE_CHANGED:
         case LV_EVENT_STYLE_CHANGED:
-        case LV_EVENT_GET_SELF_SIZE:
+        /*case LV_EVENT_GET_SELF_SIZE:*/
             return false;
         default:
             return true;
```

### To Reproduce

Pull latest Zephyr `main` and run, for example:

```
west build -b nucleo_g071rb samples/subsys/display/lvgl -p -- -DSHIELD=waveshare_epaper_gdeh0213b1
```

### Expected behavior

Compilation doesn't fail

### Screenshots or video

N/A

kisvegabor, 2022-06-24:
I also saw a similar issue on a not that popular device. (I don't remember the exact case).

Lowering the optimization level solved the issue for me.

gmarull, 2022-06-27:
I can confirm that using another optimization level, optimize for speed, doesn't produce the error.

gmarull, 2022-06-27:
Note: it doesn't happen with GCC 12.1 (from dev Zephyr SDK, https://github.com/zephyrproject-rtos/sdk-ng/actions/runs/2563830735)

closed 2022-07-13 by kisvegabor