lvgl/lvgl #3426: How does lv_label implement LV_LABEL_LONG_DOT that does not match word

FASTSHIFT, 2022-06-24:
### Introduce the problem
The current `lv_label` will match the word for the replacement of `...`,as the picture shows:
[image]

I hope that without word matching, the characters at the end are`...`, such as `hello wor...`.
How should I do it?

### Examples and cases
```c
void label_test(void)
{
  lv_obj_t* label = lv_label_create(lv_scr_act());
  lv_obj_set_style_border_color(label, lv_color_black(), 0);
  lv_obj_set_style_border_width(label, 1, 0);
  lv_obj_set_size(label, 80, 20);
  lv_obj_center(label);
  lv_label_set_long_mode(label, LV_LABEL_LONG_DOT);
  lv_label_set_text(label, "hello world!");
}
```

### Suggested solution

kisvegabor, 2022-06-28:
The issue here is that "word" doesn't fit into the set width, so LVGL wraps it into next line. Therefore LVGL thinks that last character in the first line is "o" from "hello" and writes three dots after it. 

Something similar to your method could work, but we DOTS should work with multiline texts (with `\n` or word wrapping too). 

It's clear that there are a few things to reconsider with "long modes" (starting with the name "long mode" :smile: )
E.g. #3420

I think we should have a clear specification about what we want to achieve to redesign this part in v9. What I see now:

**What to do horizontally?**
- width is set to `LV_SIZE_CONTENT`
    - nothing, as the label will fit to the text size, it cannot  be scrolled, wrapped, dotted, etc
- width is set to `px`
    - wrap
    - scroll animation 
    - make it scrollable with scrollbars 
    - crop  
    - dots? but what if there is a multi line text? always keep only the first line? 

**What to do vertically?**
- height is set to `LV_SIZE_CONTENT`
    - nothing, as the label will fit to the text size, it cannot  be scrolled, wrapped, dotted, etc
- height is set to `px` 
   - scroll animation 
   - make it scrollable with scrollbars
   - crop
   - dots (at beginning, center, or end)
 
Both width and height should handle `min/max-width/height`

I can image different modes horizontally and vertically. E.g. dots if there are to many lines but make the text auto scroll horizontally. 

But maybe it's overkill. What do you think? 

cc @Akery

W-Mai, 2022-07-10:
The `Text` component in iOS has such a method - lineLimit, which can limit the number of displayed lines of the current text component. If the limited number of lines is reached, the automatic line wrapping mechanism of the last line of text will have no effect.

If you limit the line to 2. like
> apple pen
> hello world

will display like

> apple pen
> hello wor...

but lvgl now will display like

> apple pen
> hello...

https://developer.apple.com/documentation/swiftui/text/

kisvegabor, 2022-07-14:
Yeah, we should do something similar. I'm considering to introduce a new unit type like `em` on the web. (1 em means 1 row of text size).

This we could write: `lv_obj_set_height(label, LV_EM(3))`

closed 2022-08-08 by FASTSHIFT

pprasa, 2024-07-25:
> ```c
> lv_label_set_text_ex
> ```

Thank you so much for this. I had lots of trouble with using dots to clip the long text and was not working for me. I could use this function to do the job. Thank you again.