lvgl/lvgl #3434: The default LV_MEM_SIZE is too small to finish running benchmark

GorgonMeducer, 2022-06-29:
The default size of `LV_MEM_SIZE` (which is 64KByte) is too small to finish running the benchmark.

70K - Failed
128K - Failed
256K - Pass

I tested it on a 320 * 240 RGB565 env.

### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

It is blocked in an assert. 
[image]

### To Reproduce

### Expected behavior

### Screenshots or video

Viatorus, 2022-06-30:
I get the same issue under Linux and ARM.

The problem seems to be:

This isn't called:
https://github.com/lvgl/lvgl/blob/cbff8e83e50fecc2b4b43d661deb91d8d81d6696/src/core/lv_refr.c#L964

Because this returns before:
https://github.com/lvgl/lvgl/blob/cbff8e83e50fecc2b4b43d661deb91d8d81d6696/src/core/lv_refr.c#L918-L922

Why does it return? Because of this:
https://github.com/lvgl/lvgl/blob/cbff8e83e50fecc2b4b43d661deb91d8d81d6696/src/draw/sw/lv_draw_sw_layer.c#L46-L49

So setting `LV_COLOR_SCREEN_TRANSP` to 1 should do the trick.

But the screen doesn't support `LV_COLOR_SCREEN_TRANSP`.... hmm..

GorgonMeducer, 2022-06-30:
Fixed

kisvegabor, 2022-07-01:
> Comparing the benchmark results between LVGL 8.2 and LVGL 8.3 nearly all SubStr. benchmark are worse than before.
E.g. Substr. rectangle has 41 fps less than in LVGL 8.2.

Is it better is you increase `LV_LAYER_SIMPLE_BUF_SIZE`?

Viatorus, 2022-07-04:
> > Comparing the benchmark results between LVGL 8.2 and LVGL 8.3 nearly all SubStr. benchmark are worse than before.
> > E.g. Substr. rectangle has 41 fps less than in LVGL 8.2.
> 
> Is it better is you increase `LV_LAYER_SIMPLE_BUF_SIZE`?

Absolute no difference. I stepped through and the bigger buffer is allocated but no difference in FPS.

kisvegabor, 2022-07-04:
> Substr. rectangle has 41 fps less than in LVGL 8.2.

What was the original FPS?

kisvegabor, 2022-07-05:
Hmmm, in summary:
- Border got slower
- Image transformation got faster (I really improved it, the double FPS makes sense)
- Subcontracted blend mode got slower. 

I'll check it out too, but I see not reason why Border became slower.

kisvegabor, 2022-07-05:
I found the issue. In v8.2 when a color was blended LVGL cached the previous destination and source pixel and didn't do the expensive additive/subtractive blending on the same data again. In v8.3 non-normal blended objects area rendered into a layer first (which is an image) and that image is blended. But on images it was assumed that there are not many similar pixels so this caching wasn't implemented.  

However in the benchmark , e.g. in "Substr. rectangle" a lot of plain rectangles are blended and we could make a good use of this cache.  I'm trying to add caching here too.

kisvegabor, 2022-07-05:
I pushed a fix to speed it up (5a06fce47).

It's still not the same as before, but much better. Could you rerun the benchmark on your end too?

Viatorus, 2022-07-06:
| Name                               | LVGL 8.2 | 8.3 (old) | 8.3 (new) | Diff 8.3 old -> new |
|------------------------------------|----------|-----------|-----------|---------------------|
| Rectangle                          | 105      | 100       | 91        | -9                  |
| Rectangle+OPA                      | 81       | 87        | 87        | 0                   |
| Rectangle rounded                  | 84       | 84        | 88        | 4                   |
| Rectangle rounded+OPA              | 82       | 75        | 75        | 0                   |
| Circle                             | 81       | 79        | 78        | -1                  |
| Circle+OPA                         | 79       | 67        | 69        | 2                   |
| Border                             | 120      | 96        | 96        | 0                   |
| Border+OPA                         | 114      | 92        | 91        | -1                  |
| Border rounded                     | 87       | 87        | 84        | -3                  |
| Border rounded+OPA                 | 85       | 80        | 82        | 2                   |
| Circle border                      | 81       | 67        | 72        | 5                   |
| Circle border+OPA                  | 82       | 67        | 69        | 2                   |
| Border top                         | 89       | 93        | 82        | -11                 |
| Border top+OPA                     | 88       | 83        | 83        | 0                   |
| Border left                        | 91       | 83        | 80        | -3                  |
| Border left+OPA                    | 91       | 83        | 86        | 3                   |
| Border top + left                  | 86       | 80        | 80        | 0                   |
| Border top + left+OPA              | 85       | 87        | 84        | -3                  |
| Border left + right                | 85       | 81        | 84        | 3                   |
| Border left + right+OPA            | 85       | 84        | 83        | -1                  |
| Border top + bottom                | 85       | 86        | 82        | -4                  |
| Border top + bottom+OPA            | 84       | 84        | 82        | -2                  |
| Shadow small                       | 54       | 49        | 48        | -1                  |
| Shadow small+OPA                   | 53       | 52        | 50        | -2                  |
| Shadow small offset                | 50       | 53        | 54        | 1                   |
| Shadow small offset+OPA            | 50       | 54        | 49        | -5                  |
| Shadow large                       | 31       | 32        | 32        | 0                   |
| Shadow large+OPA                   | 31       | 34        | 32        | -2                  |
| Shadow large offset                | 39       | 42        | 39        | -3                  |
| Shadow large offset+OPA            | 43       | 40        | 40        | 0                   |
| Image RGB                          | 105      | 101       | 111       | 10                  |
| Image RGB+OPA                      | 102      | 95        | 95        | 0                   |
| Image ARGB                         | 100      | 95        | 92        | -3                  |
| Image ARGB+OPA                     | 98       | 96        | 93        | -3                  |
| Image chorma keyed                 | 96       | 96        | 97        | 1                   |
| Image chorma keyed+OPA             | 96       | 87        | 90        | 3                   |
| Image indexed                      | 87       | 75        | 71        | -4                  |
| Image indexed+OPA                  | 74       | 72        | 63        | -9                  |
| Image alpha only                   | 77       | 75        | 68        | -7                  |
| Image alpha only+OPA               | 72       | 68        | 58        | -10                 |
| Image RGB recolor                  | 86       | 90        | 91        | 1                   |
| Image RGB recolor+OPA              | 77       | 80        | 78        | -2                  |
| Image ARGB recolor                 | 88       | 88        | 84        | -4                  |
| Image ARGB recolor+OPA             | 90       | 74        | 82        | 8                   |
| Image chorma keyed recolor         | 86       | 89        | 87        | -2                  |
| Image chorma keyed recolor+OPA     | 89       | 77        | 85        | 8                   |
| Image indexed recolor              | 68       | 65        | 59        | -6                  |
| Image indexed recolor+OPA          | 57       | 56        | 54        | -2                  |
| Image RGB rotate                   | 35       | 80        | 80        | 0                   |
| Image RGB rotate+OPA               | 35       | 61        | 62        | 1                   |
| Image RGB rotate anti aliased      | 17       | 35        | 35        | 0                   |
| Image RGB rotate anti aliased+OPA  | 17       | 35        | 35        | 0                   |
| Image ARGB rotate                  | 35       | 72        | 76        | 4                   |
| Image ARGB rotate+OPA              | 35       | 63        | 62        | -1                  |
| Image ARGB rotate anti aliased     | 17       | 35        | 35        | 0                   |
[... cut ...]
| Substr. arc                        | 82       | 81        | 80        | -1                  |
| Substr. arc+OPA                    | 82       | 76        | 81        | 5                   |
| Substr. text                       | 36       | 27        | 25        | -2                  |
| Substr. text+OPA                   | 36       | 25        | 27        | 2                   |

Mayeb a little bit better. But some test really noise I think. Also it depends on the refresh rate of the display (15 ms in my case).

kisvegabor, 2022-07-06:
I saw double FPS on my end. You make more precise measurements if you disable refreshing the display. Just have only a `lv_disp_flush_ready()` in your `flush_cb`. this way we can test only the rendering performance.

closed 2022-07-28 by lvgl-bot