lvgl/lvgl #3436: SCROLL_END is constantly fired in LVGL 8.3

Viatorus, 2022-06-30:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version

### Describe the bug

During scrolling, SCROLL_END is fired even through, the scroll isn't finished yet.
Happens in LVGL 8.3, not in 8.2.

### To Reproduce
https://sim.lvgl.io/v8.3/micropython/ports/javascript/index.html?script_direct=c7ba43b57e2643a7afc3aa77698ca45aed5c960d

### Expected behavior
https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_direct=a1304200d0c4f01279ecd51b01087d084d70504a

Viatorus, 2022-07-01:
Looking at the source code I see on simple option:

Reverse the change of https://github.com/lvgl/lvgl/commit/777fe1ea706f38b82ab8ee180548ecb85334a469 and https://github.com/lvgl/lvgl/commit/22854ff3fba236f50893221805c9cc4d378baaca.

I see also another interesting behavior which I didn't see/know before with scrolling:

* Animated scrolling in X and Y direction at the same time fires also BEGIN and END for each direction.
* Snapping and elastic scrolling fires BEGIN and END again after the user scrolled (because it gives the user the animation object to control it which is okay). 

So first indev_scrolls fires BEGIN, then END and afterwards, the animated scrolling fires BEGIN and END again.
I understand now why this is happening in this way so I am okay if this.

-----

After writing all this and reading the code, there is maybe another fix for the issue, to still solve https://github.com/lvgl/lvgl/issues/3272:

Make this:
https://github.com/lvgl/lvgl/blob/f5759351806d8b14537c0488842e3c7132c72fc9/src/core/lv_indev_scroll.c#L88

Calling this directly (somehow, since it is a private function):
https://github.com/lvgl/lvgl/blob/c2c44c68ee69cdee16fce7833cbf6d6dc0d551ab/src/core/lv_obj_scroll.c#L655

kisvegabor, 2022-07-01:
Hi,

I was aware of this problem but didn't know a good solution. Using `scroll_by_raw` directly is a very good idea. I've pushed a fix which uses it.

Thank you!

commit 25ce6e3a (2022-07-01): fix(scroll): do not fire scroll begin/end event on every scroll step

fixes: #3436
closed 2022-07-01 by kisvegabor