lvgl/lvgl #3440: Random no update when label update, log report err : detected modifying dirty areas in render

LusterWong, 2022-07-02:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I use ESP32 witch arduino v2.0.3, lvgl is **26/6/2022 SHA-1: cb9a297bc516c7b8d264e7fc0c518ca2874a5a5c**.
And use the SPI LCD, 320 x 480, control ic ST7796, SPI lib is LovyanGFX.
I created a tabview and add some tab. And created some label or textarea used to display IP address or SSID information.
But will random no update when label or textarea update. Sometimes not displayed, or sometimes not cleared.  If the cursor passes over the area, the graph is refreshed correctly.
Once an exception occurs, Log reports:
```
[Error]	(38.433, +76)	 _lv_inv_area: detected modifying dirty areas in render 	(in lv_refr.c line #212)
[Error]	(38.443, +10)	 _lv_inv_area: detected modifying dirty areas in render 	(in lv_refr.c line #212)
[Error]	(38.453, +10)	 _lv_inv_area: detected modifying dirty areas in render 	(in lv_refr.c line #212)
[Error]	(38.462, +9)	 _lv_inv_area: detected modifying dirty areas in render 	(in lv_refr.c line #212)
[Error]	(38.472, +10)	 _lv_inv_area: detected modifying dirty areas in render 	(in lv_refr.c line #212)
[Error]	(38.481, +9)	 _lv_inv_area: detected modifying dirty areas in render 	(in lv_refr.c line #212)
[Error]	(38.490, +9)	 _lv_inv_area: detected modifying dirty areas in render 	(in lv_refr.c line #212)
[Error]	(38.500, +10)	 _lv_inv_area: detected modifying dirty areas in render 	(in lv_refr.c line #212)
[Error]	(38.509, +9)	 _lv_inv_area: detected modifying dirty areas in render 	(in lv_refr.c line #212)
[Error]	(38.518, +9)	 _lv_inv_area: detected modifying dirty areas in render 	(in lv_refr.c line #212)
[Error]	(38.528, +10)	 _lv_inv_area: detected modifying dirty areas in render 	(in lv_refr.c line #212)
[Error]	(38.537, +9)	 _lv_inv_area: detected modifying dirty areas in render 	(in lv_refr.c line #212)
```
But nothing happened with the PC emulator.

### To Reproduce

This is the initialization two buffers:
```
        static lv_disp_draw_buf_t DrawBuf;
	static lv_color_t* buf1;
	static lv_color_t* buf2;
	static lv_disp_drv_t dispDrv;
	GUICb = Cb;

	buf1 = (lv_color_t*)heap_caps_malloc(480 * 40 * sizeof(lv_color_t), MALLOC_CAP_DMA);
	buf2 = (lv_color_t*)heap_caps_malloc(480 * 40  * sizeof(lv_color_t), MALLOC_CAP_DMA);

	lv_init();
	lv_disp_draw_buf_init(&DrawBuf, buf1, buf2, 480 * 40);

	/*Initialize the display*/
	lv_disp_drv_init( &dispDrv );
	/*Change the following line to your display resolution*/
	dispDrv.hor_res = TFT_WIDTH;
	dispDrv.ver_res = TFT_HEIGHT;
	dispDrv.flush_cb = DispFlush;
	dispDrv.draw_buf = &DrawBuf;
	lv_disp_drv_register(&dispDrv);
```

This is update the label function:
```
void UpdateWiFiInfo(char* SSID, char* Local, char* GWay)
{
	lv_label_set_text(SettingsObjs.LabSSID, SSID);
	lv_label_set_text(SettingsObjs.LabLocal, Local);
	lv_label_set_text(SettingsObjs.LabGeteway, GWay);
}
```

### Expected behavior

### Screenshots or video

https://user-images.githubusercontent.com/56709609/176995609-5432e986-1091-4304-ae94-5f4782608d19.mp4

https://user-images.githubusercontent.com/56709609/176995614-f0181644-7d12-404c-bbf2-2820efe74e67.mp4

kisvegabor, 2022-07-04:
From where do you call `UpdateWiFiInfo`?

LusterWong, 2022-07-07:
> From where do you call `UpdateWiFiInfo`?

In the key task (freertos task) callback not use hardware intterupt.

LusterWong, 2022-07-07:
> 

I have no create draw event. The mouse cursor under my finger In that video .

FASTSHIFT, 2022-07-07:
For data from other threads, you should use queues or locks to protect the data.

kisvegabor, 2022-07-07:
@LusterWong 
See https://docs.lvgl.io/master/porting/os.html

LusterWong, 2022-07-11:
> For data from other threads, you should use queues to transfer data to the UI thread for display.

Thanks for help.

closed 2022-07-11 by LusterWong