lvgl/lvgl #3441: LV_CONF_INCLUDE_SIMPLE with custom location for lv_conf.h not working

sukesh-ak, 2022-07-02:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Unable to use custom `lv_conf.h` since some defaults part of kconfig are overiding it.

### To Reproduce
Sample project is [here](https://github.com/sukesh-ak/LVGL8x-SDSPI-Template).
Custom lv_conf.h is setup [here](https://github.com/sukesh-ak/LVGL8x-SDSPI-Template/blob/17e1bf1096ded2fdffe5e156054cdc832c461012/CMakeLists.txt#L12) in CMakeList.txt (Not sure if its the right way to do it)

If I rename `Kconfig` file inside lvgl folder, everything works fine. 
I cannot use Kconfig based configuration since I have multiple board targets ESP32/ESP32-S3 and if menuconfig is used to configure, even for enabling a font, it goes back to compiling 900+ files (almost clean rebuild).

### Expected behavior
Expected behaviour is to have a simple mechanism to provide `lv_conf.h` for configuration and clearly document the usage.

Keep in mind while using `lvgl` as a component in `ESP-IDF` projects, config file is not next to the lvgl folder as mentioned in several places including code comments.
The only documentation about this is [here](https://docs.lvgl.io/latest/en/html/porting/project.html)

Not sure what below documentation means. Where should we set the include path manually?
```
lv_conf.h can be copied other places as well but then
you should add LV_CONF_INCLUDE_SIMPLE define to your compiler options 
(e.g. -DLV_CONF_INCLUDE_SIMPLE for gcc compiler) and set the include path manually.
```

There are several confusing settings related to config and mostly undocumented.
- What and how to use `LV_CONF_INCLUDE_SIMPLE`
- What is `LV_CONF_MINIMAL` ?
- What is `LV_CONF_SKIP`? Its set as `y` as default value in kconfig

Not sure what this means taken from `kconfig`
```cmake
    # Define CONFIG_LV_CONF_SKIP so we can use LVGL
    # without lv_conf.h file, the lv_conf_internal.h and
    # lv_conf_kconfig.h files are used instead.
```
There is some discussion related to this [here](https://github.com/lvgl/lvgl/issues/3078) but still no clear documentation around it.

### Screenshots or video

kisvegabor, 2022-07-04:
I'm not familiar with the custom IDF CMake commands but if you set `LV_CONF_INCLUDE_SIMPLE` with
```
idf_build_set_property(COMPILE_OPTIONS "-DLV_CONF_INCLUDE_SIMPLE=1" APPEND)
```

Then instead of:
```
idf_build_set_property(COMPILE_OPTIONS "-DLV_CONF_PATH= ${CMAKE_SOURCE_DIR}/main/lv_conf.h" APPEND)
```

Just add the include path of `lv_conf.h` (I'm not sure if it's the correct path):
```
idf_build_set_property(COMPILE_OPTIONS "-I../main" APPEND)
```

If it works, we can figure out how to make the docs cleaner :)

sukesh-ak, 2022-07-04:
**I believe the main issue is not ESP-IDF but the default setting enforced by kconfig inside lvgl folder.**

Because my workaround is to use below lines and delete/rename kconfig inside lvgl folder.
```cmake
idf_build_set_property(COMPILE_OPTIONS "-DLV_CONF_INCLUDE_SIMPLE=1" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-DLV_CONF_PATH= ${CMAKE_SOURCE_DIR}/main/lv_conf.h" APPEND)
```

### Just tested and found that either of these work but kconfig still an issue.
`idf_build_set_property(COMPILE_OPTIONS "-DLV_CONF_PATH= ${CMAKE_SOURCE_DIR}/main/lv_conf.h" APPEND)`
OR
`idf_build_set_property(COMPILE_OPTIONS "-I../main" APPEND)`

### Default/Standard ESP-IDF code tree
```
├── CMakeLists.txt [ idf_build_set_property specified ]
├── components 
│   └── lvgl (git clone)
├── main
│   ├── CMakeLists.txt
│   ├── lv_conf.h [ config file is here ]
│   └── main.cpp
└── README.md                  
```

sukesh-ak, 2022-07-04:
The issue is because `CONFIG_LV_CONF_SKIP=y` is default in kconfig and with compiler options its not getting overridden, so it skips the custom config.

sukesh-ak, 2022-07-04:
Here is how this can be addressed.

Current kconfig setting for `LV_CONF_SKIP` has no label provided so it does not show up in the menuconfig options to disable if required.
```
    # Define CONFIG_LV_CONF_SKIP so we can use LVGL
    # without lv_conf.h file, the lv_conf_internal.h and
    # lv_conf_kconfig.h files are used instead.
    config LV_CONF_SKIP
        bool
        default y
```
Change it to below adding a label
```
    # Define CONFIG_LV_CONF_SKIP so we can use LVGL
    # without lv_conf.h file, the lv_conf_internal.h and
    # lv_conf_kconfig.h files are used instead.
    config LV_CONF_SKIP
        bool "Uncheck this to use custom lv_conf.h"
        default y
```

[image]

kisvegabor, 2022-07-04:
Ah, thanks for tracking it. I've just pushed a fix.

As you mentioned the docs should be update too. Do you agree to add a paragraphs at the end of the [Configuration file](https://docs.lvgl.io/master/porting/project.html#configuration-file) section to describe this case?

closed 2022-07-04 by kisvegabor

sukesh-ak, 2022-07-04:
> Ah, thanks for tracking it. I've just pushed a fix.
> 
> As you mentioned the docs should be update too. Do you agree to add a paragraphs at the end of the [Configuration file](https://docs.lvgl.io/master/porting/project.html#configuration-file) section to describe this case?

Yes I will add it. Will take ESP32-IDF as an example?

kisvegabor, 2022-07-04:
I think it's not required to name any vendor or hardware here. Just something like "If you are using Kconfig and and a lv_conf.h file together ...."

commit f3a07a3e (2022-07-04): fix(Kconfig): allow unchecking LV_CONF_SKIP

fixes: #3441