lvgl/lvgl #3443: CI for regression in performance

kisvegabor, 2022-07-03:
### Introduce the problem
Now we have no indication if performance drops due to a new feature

### Examples and cases

For example the new layer based method to correctly blend object with opacity, or non-normal blend mode and to rotate widget rotation sure introduced performance drop in some cases. But it'd be great to know by an objective measurement what is the rate of this drop and whether the unaffected drawing tasks are really unaffected. 

### Suggested solution

I cached up with Danil, the main contributor of [SO3](https://gitlab.com/smartobject/so3/-/tree/master), which is basically on operating system with QEMU support. All the uboot, qemu and user code stuff can be compiled an initialized with a few command line instructions. The only interesting dependency is Linaro. In QEMU we can use the standard `gettimeofday()` function to measure the execution time of special tasks like:
- draw various primitives
- draw full widgets
- malloc fetures
- style related stuff
- etc

All we need is a CI, which tests the mentioned cases.

@embeddedt if I provide the list of commands could you set up a CI for this? For now, I think it'd be enough to have a single run with a `main.c` file to test all the cases we need.

embeddedt, 2022-07-04:
Sure, I can set this up. For now, is it enough to just print the execution time in the action log and not post a comment with a delta like Codecov does?

kisvegabor, 2022-07-04:
Great!

> For now, is it enough to just print the execution time in the action log and not post a comment with a delta like Codecov does?

I think it should be only something like:
```
do_something()
elapsed_time = get_the elapsed_time();
if(elapsed_time > 50) exit(1);
```

---

The commands are mainly based on the [official guide](https://smartobject.gitlab.io/so3/user_guide.html#user-guide), but there are some differences to make the frame buffer work. Actually I'm not sure we need a display in the CI as we will test only the performance of some internal algorithms.

So:
Get Linaro:
I downloaded `gcc-linaro-6.4.1-2018.05-i686_arm-linux-gnueabihf.tar.xz` from [here](https://releases.linaro.org/components/toolchain/binaries/latest-6/arm-linux-gnueabihf/).

Add the bin subdirectory to your path.

```shell
#!/bin/bash
set -o xtrace
set -e

#Pre-requisite
sudo dpkg --add-architecture i386
sudo apt-get update
sudo apt-get install libc6:i386 libncurses5:i386 libstdc++6:i386 lib32z1-dev zlib1g:i386 pkg-config libgtk2.0-dev bridge-utils unzip bc elfutils u-boot-tools device-tree-compiler fdisk libncurses-dev bison flex ninja-build

#Get SO3:
git clone https://gitlab.com/smartobject/so3.git

#Build what needs to be built:
cd so3
cd qemu
./configure --target-list=arm-softmmu --disable-attr --disable-werror --disable-docs
make -j
cd ..

cd u-boot
make vexpress_defconfig
make -j
cd ..

cd filesystem
./create_img.sh vexpress
cd ..

cd usr
./build.sh
cd ..

cd rootfs
./create_ramfs.sh vexpress
cd ..

cd so3
make vexpress_fb_defconfig
make -j
cd ..

./deploy.sh -bu

#Start
./stg
```

Start the demo
```
demofb
```

---

Important note: the above script is not working now due to this build error:
```
+ cd ..
+ cd u-boot
+ make vexpress_defconfig
/bin/sh: 1: arm-none-linux-gnueabihf-gcc: not found
./scripts/gcc-version.sh: 25: arm-none-linux-gnueabihf-gcc: not found
./scripts/gcc-version.sh: 26: arm-none-linux-gnueabihf-gcc: not found
./scripts/gcc-version.sh: 28: arm-none-linux-gnueabihf-gcc: not found
#
# configuration written to .config
#
+ make -j
/bin/sh: 1: arm-none-linux-gnueabihf-gcc: not found
make: arm-none-linux-gnueabihf-gcc: No such file or directory
/bin/sh: 1: arm-none-linux-gnueabihf-gcc: not found
dirname: missing operand
Try 'dirname --help' for more information.
/bin/sh: 1: arm-none-linux-gnueabihf-gcc: not found
scripts/kconfig/conf  --syncconfig Kconfig
./scripts/gcc-version.sh: 25: arm-none-linux-gnueabihf-gcc: not found
./scripts/gcc-version.sh: 26: arm-none-linux-gnueabihf-gcc: not found
./scripts/gcc-version.sh: 28: arm-none-linux-gnueabihf-gcc: not found
```

It tries to use `arm-none-linux-gnueabihf-gcc` instead of `arm-linux-gnueabihf-gcc`. To fix it go to `so3/u-boot/Makefile` and change `CROSS_COMPILE = arm-none-linux-gnueabihf-` to `CROSS_COMPILE = arm-linux-gnueabihf-`

Similar issue in `so3/so3/arch/Kconfig` `default "arm-none-linux-gnueabihf-" if ARCH_ARM32`

After it, all built well, but got this 
```
Now bootstraping the kernel ...
SO3: allocating a kernel heap of 33554404 bytes at address c0040000.
memory_init: relocating the device tree from 0x84a00000 to 0xc2250000 (size of 4224 bytes)
SO3 Memory information:
  - kernel size without frame table is: 35987456 (0x2252000) bytes, 34 MB / 0x2252 PFNs
  - kernel size including frame table is: 36966400 (0x2341000) bytes, 35 MB / 0x2341 PFNs
  - Number of available page frames: 0x1ddae
  - Frame table size is: 978288 bytes meaning 0xef page frames
  - Page frame number of the first available page: 0x82341
so3: rootfs in RAM detected (ramdev enabled) with size of 17825792 bytes...
BUG in kernel/timer.c at line: 353
kernel_panic: entering infinite loop... CPU: 0
```
Which is interesting because it worked well a few days ago.

I reached out to Daniel to take a look at this problem.

kisvegabor, 2022-07-05:
Thanks Daniel, but I still see the same issue with `arm-none-linux-gnueabihf-gcc` for bot u-boot and so3. 

Am I doing something wrong? `-none-` is working for you with this Linaro?
> I downloaded gcc-linaro-6.4.1-2018.05-i686_arm-linux-gnueabihf.tar.xz from [here](https://releases.linaro.org/components/toolchain/binaries/latest-6/arm-linux-gnueabihf/).

drossier, 2022-07-05:
[The doc has been updated](https://smartobject.gitlab.io/so3/user_guide.html). Using this toolchain should be OK. Please report me any trouble with it.

kisvegabor, 2022-07-05:
I've just tested it and my above script is working now with the toolchain you have linked. 

It just came to my mind that our `aarch64`, `armv6`, `armv7` tests also might be used for benchmarking. They also [uses QEMU](https://github.com/uraimo/run-on-arch-action):
> A GitHub Action that executes commands on non-x86 CPU architecture (armv6, armv7, aarch64, s390x, ppc64le) via QEMU.

@drossier @embeddedt 
What could be pros and cons of using SO3 vs our current QEMU tests?

kisvegabor, 2022-07-05:
`aarch64`, `armv6`, `armv7` are running Linux which is probably less predictable than SO3. So the jitter of the measurements can be larger in our current tests.

embeddedt, 2022-07-05:
The ARM runners are very slow because they're using QEMU's userspace translation. There will be a lot of overhead there; using a bare-metal test makes more sense. 

I have started writing a Dockerfile based on Gabor's instructions but I am concerned about QEMU and other components being rebuilt from source every time. It seems that there are a lot of moving parts just to run a simple application. What are your thoughts? We had also talked about using Arm's Corstone tool before in https://github.com/lvgl/lvgl/issues/3018.

kisvegabor, 2022-07-05:
> The ARM runners are very slow because they're using QEMU's userspace translation. There will be a lot of overhead there; using a bare-metal test makes more sense.

SO3 runs the widgets demo with ~10 FPS so the performance is not that bad. 

> We had also talked about using Arm's Corstone tool before in https://github.com/lvgl/lvgl/issues/3018.

Yes, it's an other option. @GorgonMeducer mentioned that he will create Linux project for it but AFAIK he haven't had time to deal with it yet.

embeddedt, 2022-07-21:
What direction did we decide to go with here? I had paused work on the Docker script because the containers it produces are very large (2GB, due to all the object files) and I was concerned about the time it takes to rebuild as mentioned above.

drossier, 2022-07-21:
If I got it correctly, this huge size is mainly due to QEMU right ?

embeddedt, 2022-07-21:
I believe so.

kisvegabor, 2022-07-21:
So we have 3 options:
- SO3: seems to large
- aarch64, armv6, armv7: too slow
- Arm's Corstone: not tested yet (promising as it's bare metal)

@GorgonMeducer can you estimate when will you have time to create a Linux project for Arm's Corstone?

GorgonMeducer, 2022-07-21:
@kisvegabor I am sorry, I am really busy these-days. I am working on adding stretch-transform (and this is the preparing step for perspective transform). 
My suggestion is, you can generate an elf from other platform, copy it to linux environment and test the FVP first. I think this step can help you evaluate whether the FVP fulfil your expectation.

Good news is you can use functions in <stdio.h> directly without retargeting it to peripherals such as UART, because **FVP support semihosting in command line mode**. That means you can not only use printf but also fopen, fread and fwrite easily without special software arrangement. 

The memory layout of the FVP can be found here:
https://developer.arm.com/documentation/dai0547/c?_ga=2.157798205.688811587.1624957483-616249991.1623083451

drossier, 2022-07-22:
@kisvegabor In the meanwhile, I will try to set up a docker container based build which should lead to a much smaller buid env and reduce the build time as well.

kisvegabor, 2022-07-24:
@GorgonMeducer 
Probably next week I'll install Windows and can try it out.

@drossier 
Cool, thank you!

kisvegabor, 2022-08-15:
Not stale. I still couldn't find time to install Windows...

drossier, 2022-08-30:
I'm doing progress on SO3 docker containerization. The final image should be less than 1GB. I just need more time to achieve since I'm very busy these days.

GorgonMeducer, 2022-09-12:
@kisvegabor for solving current problem, you might want to check this 
https://github.com/Open-CMSIS-Pack/cmsis-toolbox

kisvegabor, 2022-09-13:
How does it help? Can it be used to create Make/CMake project from https://github.com/lvgl/lv_port_an547_cm55_sim?

GorgonMeducer, 2022-09-13:
@kisvegabor
 Yes, it is very close. It creates a CMake project and allows you to use CMSIS-Packs, so LVGL, CMSIS-Device etc. can be deployed in command line mode.

kisvegabor, 2022-09-15:
Thanks. I've look into the installation guide but I couldn't find from where to download  [CMSIS-Pack](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/installation.md#environment-variables). I thought it's the collection of packages available online [here](https://developer.arm.com/tools-and-software/embedded/cmsis/cmsis-packs).

GorgonMeducer, 2022-09-15:
> I thought it's the collection of packages available online [here](https://developer.arm.com/tools-and-software/embedded/cmsis/cmsis-packs).

YES. And you can try the other one also:

https://www.keil.com/dd2/pack/

drossier, 2022-10-07:
Still working on int. We should have something in the coming days.

drossier, 2022-10-24:
We are on an specific ABI issue related to the cross-compiler running in Docker (the compiler is pretty old apparently). 
I expect to solve the problem within the two next weeks (still in a rush situation ;-)

kisvegabor, 2022-10-25:
Thanks for the update, Daniel. :+1:

GorgonMeducer, 2022-11-03:
@kisvegabor please check this

https://resources.github.com/devops/actions/arm-virtual-hardware/

kisvegabor, 2022-11-04:
> https://resources.github.com/devops/actions/arm-virtual-hardware/

@embeddedt could you take a look at it?

embeddedt, 2022-11-05:
Maybe I'm missing something, but it seems that it involves setting up a separate AWS instance that runs the Corstone tool we discussed earlier in the year. Then the GitHub runner becomes a client that submits CI jobs to that instance. It can work, but I wonder for something as small as LVGL if running the Corstone emulator on GitHub's runner is fast enough already.

drossier, 2022-11-20:
Still in development. We expect more details in a few days.

drossier, 2022-12-06:
Hope it will be done before end of this Year :-)

kisvegabor, 2022-12-07:
Okay :slightly_smiling_face:

daniel-rossier, 2022-12-22:
Finally, I was able to have a docker image of 770 MB that can be used in the CI. Everything compiles with LVGL of course.
The Dockerfile is available in the root directory of[ SO3 ]((https://github.com/smartobjectoriented/so3)).

I've also managed to use the mainline QEMU (qemu-system-arm) thus reducing the delay of build time as well as to use the arm-none-eabi toolchain (instead of the arm-linux-* which contains the libc related files that are not used in so3.

There is still room for improvement in the way how LVGL is compiled in the container; we could create a separate layer to optimize the CI process when LVGL has modifications (provided that the rest of code is unchanged, it will be stored in the container cache in the CI server).

The next step will be to make a first measure of time in a drawing function and to collect the result through a log file in a way or the other.
I would suggest to use the ``-ncount`` facility of QEMU which allows to have a deterministic execution independently of the execution environment.

Don't hesitate to comment. Anyone can help; I keep you posted.

kisvegabor, 2023-01-02:
Great news Daniel, thank you! 

If you can put together a very basic example for measuring time and a getting started guide, I'd be happy to add some more measurements.

kisvegabor, 2023-01-26:
Not stale

daniel-rossier, 2023-02-14:
Will be updated be end of February hopefully with the new release of SO3 to be used for the CI.

kisvegabor, 2023-02-14:
Looking forward to it :)

daniel-rossier, 2023-03-01:
Still ongoing work on SO3 code base. The new version has been released Yesterday. Adaptations along the CI will be made soon.

daniel-rossier, 2023-03-09:
I'm now ready to make some performance measurement with LVGL code. Please could you provide me an example of simple app using LVGL that can be used as a first reference to measure the duration of processing (like drawning a line or more complex object)? Thanks in advance

kisvegabor, 2023-03-13:
Great news, Daniel!

I think measuring the first execution time of `lv_timer_handler` or `lv_refr_now(NULL)` with an empty screen would be a good first test.

daniel-rossier, 2023-03-30:
Well, I made some progress on this topic. I guess I have all (main) pieces of the puzzles :-)
What I have so far:
- A docker container running SO3 and the LVGL test app automatically, using shell script and docker facilities
- A deterministic behaviour using QEMU icount option to run the same execution path along several executions
- A measure of execution time of functions; I tested with lv_timer_handler() as Gàbor suggested, but of course, any function can be tested now. By the way, I got 12335 microseconds for this function. If I add some sleep time before, I still have the same duration, that's a good sign (deterministic behaviour).

So, that's good news. But I will need some help to go forward with further sophisticated testings on the LVGL code base.
Now, to make a step further, how could we start implementing this environment in the LVGL github? Would it be possible to have a dedicated CI action? For sure, it will still require to configure the artifacts required by the CI, but it should be straightforward.

Anyway, I will have to improve the Dockerfile to get the SO3 base code (from github) and to make the required patches to run into the CI, but if you can already give me some advices, it will be cool :-)

kisvegabor, 2023-03-31:
Great news, Daniel! Thank you! 

At this point I think we can create a dedicated GitHub CI with a single dummy test.  Can you create it in a `lvgl` fork and send a PR?

daniel-rossier, 2023-03-31:
Great! I will do that.

daniel-rossier, 2023-04-07:
Hi  Gàbor (@kisvegabor), I just made a pull request for the performanche check of LVGL based on SO3.
It's not finished yet because it remains to adapt the workflow to download the LVGL code within the SO3 directory so that
we can apply the check on the last commit of this main branch. For the time being, the LVGL code is the one which is contained in SO3 repository.
However, I made this PR for people who would like to take a look on the workflow, and then to contribute to the check application in SO3. Just tell me if you'd prefer to proceed differently.

kisvegabor, 2023-04-11:
Thank you for the PR and the great work! I've commented on the PR. (#4122)

kisvegabor, 2023-04-26:
Let's continue in PR.

closed 2023-04-26 by kisvegabor