lvgl/lvgl #3452: Cant adjust CHART's to more than 10 points.  lv_chart_set_point_count negates LV_PART_ITEMS from callback

Coffeeye, 2022-07-06:
_Originally posted by @Coffeeye in https://github.com/lvgl/lvgl/issues/3449#issuecomment-1175824129_

### Describe the bug
Lets say my ```number``` is 200

- Effect presented above (mask) and callback call on line ```if(dsc->part == LV_PART_ITEMS)``` in particular seems being somehow suspended by calling ```lv_chart_set_point_count(chart, number);``` and setting points manualy inside loop with ```lv_chart_set_next_value(chart, ser1, n);``` does not trigger that type/part of 'dsc'

- Also creating an array with more than 10(irrelevant) positions not taking the amount of point stored inside ```lv_chart_set_ext_y_array(chart, ser1, value_array);```

```LV_CHART_DRAW_PART_LINE_AND_POINT``` type is never there same as ```LV_PART_ITEMS``` [as it would appear from this document](https://docs.lvgl.io/8.2/widgets/extra/chart.html#events)

### To Reproduce
FUNCTION BUILD (CALLD ON SYSTEM STARTUP ONLY)
```c
	chart = lv_chart_create(screen);
	lv_obj_remove_style_all(chart);
	lv_obj_align_to(chart, box_top, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 0);
	lv_obj_set_size(chart, DISP_HOR_RES, DISP_VER_RES - 29);
	lv_chart_set_type(chart, LV_CHART_TYPE_LINE);
	lv_chart_set_update_mode(chart, LV_CHART_UPDATE_MODE_CIRCULAR);
	lv_chart_set_div_line_count(chart, 7, 11);
	lv_obj_set_style_size(chart, 1, LV_PART_INDICATOR);
	lv_obj_set_style_line_width(chart, 1, LV_PART_ITEMS);
	lv_obj_set_style_border_width(chart, 1, LV_PART_MAIN);
	lv_obj_set_style_bg_color(chart, lv_color_black(), LV_PART_MAIN);
	lv_obj_set_style_bg_opa(chart, LV_OPA_COVER, LV_PART_MAIN);
	lv_obj_add_event_cb(chart, lv_chart_callback, LV_EVENT_DRAW_PART_BEGIN, NULL);
```

CALLBACK
```c
void lv_chart_callback(lv_event_t *event) {
	lv_obj_t *obj = lv_event_get_target(event);
	lv_obj_draw_part_dsc_t *dsc = lv_event_get_draw_part_dsc(event);

	/*Add the faded area before the lines are drawn*/
	if(dsc->part == LV_PART_ITEMS) {
		if(!dsc->p1 || !dsc->p2) {
			return;
		}

		/*Add a line mask that keeps the area below the line*/
		lv_draw_mask_line_param_t line_mask_param;
		lv_draw_mask_line_points_init(&line_mask_param, dsc->p1->x, dsc->p1->y, dsc->p2->x, dsc->p2->y, LV_DRAW_MASK_LINE_SIDE_BOTTOM);
		int16_t line_mask_id = lv_draw_mask_add(&line_mask_param, NULL);

		/*Add a fade effect: transparent bottom covering top*/
		lv_coord_t h = lv_obj_get_height(obj);
		lv_draw_mask_fade_param_t fade_mask_param;
		lv_draw_mask_fade_init(&fade_mask_param, &obj->coords, LV_OPA_COVER, obj->coords.y1 + h / 8, LV_OPA_TRANSP, obj->coords.y2);
		int16_t fade_mask_id = lv_draw_mask_add(&fade_mask_param, NULL);

		/*Draw a rectangle that will be affected by the mask*/
		lv_draw_rect_dsc_t draw_rect_dsc;
		lv_draw_rect_dsc_init(&draw_rect_dsc);
		draw_rect_dsc.bg_opa = LV_OPA_40;
		draw_rect_dsc.bg_color = dsc->line_dsc->color;

		lv_area_t a;
		a.x1 = dsc->p1->x;
		a.x2 = dsc->p2->x - 1;
		a.y1 = LV_MIN(dsc->p1->y, dsc->p2->y);
		a.y2 = obj->coords.y2;
		lv_draw_rect(dsc->draw_ctx, &draw_rect_dsc, &a);

		/*Remove the masks*/
		lv_draw_mask_free_param(&line_mask_param);
		lv_draw_mask_free_param(&fade_mask_param);
		lv_draw_mask_remove_id(line_mask_id);
		lv_draw_mask_remove_id(fade_mask_id);
	}
	/*Hook the division lines too*/
	else if(dsc->part == LV_PART_MAIN) {
		if(dsc->line_dsc == NULL || dsc->p1 == NULL || dsc->p2 == NULL) {
			return;
		}

		/*Vertical line*/
		if(dsc->p1->x == dsc->p2->x) {
			if(dsc->id == 0) {
				dsc->line_dsc->width = 1;
				dsc->line_dsc->dash_gap = 0;
				dsc->line_dsc->dash_width = 0;
				dsc->line_dsc->color = lv_palette_darken(LV_PALETTE_YELLOW, 2);
				dsc->line_dsc->opa = LV_OPA_COVER;
			} else {
				dsc->line_dsc->width = 1;
				dsc->line_dsc->dash_gap = 13;
				dsc->line_dsc->dash_width = 4;
				dsc->line_dsc->color = lv_palette_darken(LV_PALETTE_GREY, 1);
				dsc->line_dsc->opa = LV_OPA_0;
			}
		}
		/*Horizontal line*/
		else {
			if(dsc->id == 6) {
				dsc->line_dsc->width = 1;
				dsc->line_dsc->dash_gap = 0;
				dsc->line_dsc->dash_width = 0;
				dsc->line_dsc->color = lv_palette_darken(LV_PALETTE_YELLOW, 2);
				dsc->line_dsc->opa = LV_OPA_COVER;
			} else if(dsc->id == 0) {
				dsc->line_dsc->width = 1;
				dsc->line_dsc->dash_gap = 0;
				dsc->line_dsc->dash_width = 0;
				dsc->line_dsc->color = lv_palette_darken(LV_PALETTE_LIGHT_GREEN, 3);
				dsc->line_dsc->opa = LV_OPA_COVER;
			} else {
				dsc->line_dsc->width = 1;
				dsc->line_dsc->dash_gap = 5;
				dsc->line_dsc->dash_width = 5;
				dsc->line_dsc->color = lv_palette_darken(LV_PALETTE_GREY, 1);
				dsc->line_dsc->opa = LV_OPA_0;
			}
		}
	}
}
```

FUNCTION DATA UPDATE (CALLD BY USER)
```c
	lv_chart_set_x_start_point(chart, ser1, 0);

	lv_chart_set_range(chart, LV_CHART_AXIS_PRIMARY_Y, 0, 1000);
	lv_chart_set_range(chart, LV_CHART_AXIS_PRIMARY_X, 0, MAX_ARRAY_LEN);

	lv_chart_set_point_count(chart, MAX_ARRAY_LEN);
	int n = 0;
	for(int i = 0;i < MAX_ARRAY_LEN;i++) {
		n = lv_rand(0, 1000);
		lv_chart_set_next_value(chart, ser1, n);
	}
	lv_chart_refresh(chart);
	lv_scr_load_anim(screen, direction, 400, 0, false);
```

### Expected behavior
_Originally posted by @kisvegabor in 
https://github.com/lvgl/lvgl/issues/3449#issuecomment-1175824129_
> It works for me. This minimal example produces this:
> [image]

But for more than 10(default number) points.

kisvegabor, 2022-07-09:
Also works, but only barely visible:
[image]

With these settings it looks better:
```c
       /*Add a fade effect: transparent bottom covering top*/
        lv_coord_t h = lv_obj_get_height(obj);
        lv_draw_mask_fade_param_t fade_mask_param;
        lv_draw_mask_fade_init(&fade_mask_param, &obj->coords, LV_OPA_COVER, obj->coords.y1 + h / 8, LV_OPA_30, obj->coords.y2);
        int16_t fade_mask_id = lv_draw_mask_add(&fade_mask_param, NULL);

        /*Draw a rectangle that will be affected by the mask*/
        lv_draw_rect_dsc_t draw_rect_dsc;
        lv_draw_rect_dsc_init(&draw_rect_dsc);
        draw_rect_dsc.bg_opa = LV_OPA_80;
        draw_rect_dsc.bg_color = dsc->line_dsc->color;
```

[image]

Are you user the event is really not triggered? Can you confirm it with a debugger or `printf`?

What are the atual values here:
> lv_obj_set_size(chart, DISP_HOR_RES, DISP_VER_RES - 29);

I used 300, 200.

Coffeeye, 2022-07-12:
> Are you user the event is really not triggered? Can you confirm it with a debugger or `printf`?

[image]

This is the only code Iam getting in callback `LV_PART_MAIN`

kisvegabor, 2022-07-12:
With 160 px width it doesn't work for me either:
[image]

The problem is that if the chart has more point than its width it goes to "crowded mode" in whcih LVGL makes some optimization to effectively draw a huge amount of data. As in this case (e.g. if width=160, point_cnt=1,600) it can happen that 10 point are drawn on a single column. As part of the optimization there are no extra draw events, and no points are drawn on line joints. 

With the following modification I could make it work:
- In your event_cb
```c
        lv_area_t a;
        a.x1 = dsc->p1->x;
        a.x2 = dsc->p2->x;   //remove the -1
        a.y1 = LV_MIN(dsc->p1->y, dsc->p2->y);
        a.y2 = obj->coords.y2;
        lv_draw_rect(dsc->draw_ctx, &draw_rect_dsc, &a);
```

In lv_char.c
```diff
diff --git a/src/extra/widgets/chart/lv_chart.c b/src/extra/widgets/chart/lv_chart.c
index da6c18c0..f8c40e33 100644
--- a/src/extra/widgets/chart/lv_chart.c
+++ b/src/extra/widgets/chart/lv_chart.c
@@ -968,7 +968,13 @@ static void draw_series_line(lv_obj_t * obj, lv_draw_ctx_t * draw_ctx)
                             p1.y = y_min;
                             p2.y = y_max;
                             if(p1.y == p2.y) p2.y++;    /*If they are the same no line will be drawn*/
+                            part_draw_dsc.id = i - 1;
+                            part_draw_dsc.p1 = &p1;
+                            part_draw_dsc.p2 = &p2;
+                            part_draw_dsc.value = ser->y_points[p_prev];
+                            lv_event_send(obj, LV_EVENT_DRAW_PART_BEGIN, &part_draw_dsc);
                             lv_draw_line(draw_ctx, &line_dsc_default, &p1, &p2);
+                            lv_event_send(obj, LV_EVENT_DRAW_PART_END, &part_draw_dsc);
                             p2.x++;         /*Compensate the previous x--*/
                             y_min = y_cur;  /*Start the line of the next x from the current last y*/
                             y_max = y_cur;

```

Coffeeye, 2022-07-12:
Works like a charm :), thank you!

closed 2022-07-12 by Coffeeye

kisvegabor, 2022-07-13:
Great!