lvgl/lvgl #3457: Coordinates and rendered image are out of sync

FASTSHIFT, 2022-07-07:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I have a window sliding animation, and I want to load a resource after the window sliding animation ends. This operation of loading resources may take a long time. The code is as follows.
It can be seen that a relatively obvious stutter occurred in the frame before the animation is about to end. Intuitively, the stutter should occur after the animation has completely ended.
This is caused by the rendering mechanism, because after setting the coordinates, lvgl will start rendering in the next cycle.
Is there any way to bypass this problem?

### To Reproduce

```c
static void on_anim_ready(lv_anim_t* a)
{
    LV_LOG_USER("a = %p", a);

    /* Do some time-consuming operations, such as loading resources */
    usleep(500 * 1000);

    LV_LOG_USER("OK");
}

void anim_test(void)
{
    lv_obj_clear_flag(lv_scr_act(), LV_OBJ_FLAG_SCROLLABLE);

    lv_obj_t* ui_root = lv_obj_create(lv_scr_act());
    lv_obj_set_size(ui_root, LV_HOR_RES, LV_VER_RES);

    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_var(&a, ui_root);
    lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t)lv_obj_set_y);
    lv_anim_set_values(&a, LV_VER_RES, 0);
    lv_anim_set_time(&a, 1000);
    lv_anim_set_delay(&a, 3000);
    lv_anim_set_ready_cb(&a, on_anim_ready);
    lv_anim_start(&a);
}
```

### Expected behavior
Stuttering occurs after the animation ends instead of the frame before the animation ends.

### Screenshots or video

https://user-images.githubusercontent.com/26767803/177683125-d55c942b-7ddb-44c5-8f31-51db114090ba.mp4

kisvegabor, 2022-07-07:
I'm not sure is solves the problem, but calling `lv_refr_now()` in `on_anim_ready` might help.

FASTSHIFT, 2022-07-07:
> I'm not sure is solves the problem, but calling `lv_refr_now()` in `on_anim_ready` might help.

However, `lv_refr_now()` may not be allowed to be called at any time, the draw buffer used for rendering may be occupied by the LCD, and immediate rendering will cause errors.

FASTSHIFT, 2022-07-07:
Is it possible to consider implementing an asynchronous event mechanism using `monitor_cb`? Because this event occurs when the next frame rendering is complete.

kisvegabor, 2022-07-07:
> However, lv_refr_now() may not be allowed to be called at any time, the draw buffer used for rendering may be occupied by the LCD, and immediate rendering will cause errors.

It's not that immediate. :slightly_smiling_face: It still waits for `lv_disp_flush_ready()` like it would normally. The only catch is that `lv_refr_now()` also calls the animation handler. So  probably it'd be more appropriate:
```c
lv_disp_t * disp = lv_disp_get_default();
_lv_disp_refr_timer( disp->refr_timer);
```

However if multiple animations are finished on the same frame it will make multiple refreshes. 

> Is it possible to consider implementing an asynchronous event mechanism using monitor_cb? Because this event occurs when the next frame rendering is complete.

Sure, it's also possible. You can set a flag in `on_anim_ready` and poll it in `monitor_cb`.
It seems like a better solution. (In v9 I think we should rename `monitor_cb` to `render_ready_cb`).

FASTSHIFT, 2022-07-07:
I tried to implement my idea and it works fine.
I think it is possible to add an API similar to `lv_async_call()`, but it is called after rendering, what do you think

```c
static lv_ll_t render_async_ll;

typedef struct {
    lv_async_cb_t cb;
    void * user_data;
}render_async_info_t;

static void on_render_ready(lv_disp_drv_t * disp_drv, uint32_t time, uint32_t px)
{
    render_async_info_t* info;
    _LV_LL_READ(&render_async_ll, info) {
        info->cb(info->user_data);
    }
    _lv_ll_clear(&render_async_ll);
}

static void render_async_init(void)
{
    _lv_ll_init(&render_async_ll, sizeof(render_async_info_t));
}

static void render_async_call(lv_async_cb_t async_xcb, void * user_data)
{
    render_async_info_t* info = _lv_ll_ins_tail(&render_async_ll);
    info->cb = async_xcb;
    info->user_data = user_data;
}

static void on_anim_async_ready(void* user_data)
{
    /* Do some time-consuming operations, such as loading resources */
    usleep(500 * 1000);
    LV_LOG_USER("OK");
}

static void on_anim_ready(lv_anim_t* a)
{
    LV_LOG_USER("a = %p", a);
    render_async_call(on_anim_async_ready, NULL);
}

static void anim_test(void)
{
    render_async_init();

    lv_obj_clear_flag(lv_scr_act(), LV_OBJ_FLAG_SCROLLABLE);

    lv_obj_t* ui_root = lv_obj_create(lv_scr_act());
    lv_obj_set_size(ui_root, LV_HOR_RES, LV_VER_RES);
    lv_obj_set_style_bg_color(ui_root, lv_palette_main(LV_PALETTE_BLUE), 0);

    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_var(&a, ui_root);
    lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t)lv_obj_set_y);
    lv_anim_set_values(&a, LV_VER_RES, 0);
    lv_anim_set_time(&a, 1000);
    lv_anim_set_delay(&a, 2000);
    lv_anim_set_ready_cb(&a, on_anim_ready);
    lv_anim_start(&a);
}

void hal_init(void)
{
    ...
    disp_drv.monitor_cb = on_render_ready;
    ...
}
```

closed 2022-07-30 by lvgl-bot