lvgl/lvgl #3470: About the color or transparency setting of the border of the selected control

Micky513673326, 2022-07-11:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I want to modify the selected border color as shown in the picture. I have used all the code below to no avail.
`  lv_obj_set_style_line_color(btn, lv_color_black(), LV_STATE_ANY | LV_PART_ANY);
  lv_obj_set_style_bg_color(btn, lv_color_black(), LV_STATE_ANY | LV_PART_ANY);
  lv_obj_set_style_outline_color(btn, lv_color_black(), LV_STATE_ANY | LV_PART_ANY);
  lv_obj_set_style_border_color(btn, lv_color_black(), LV_STATE_ANY | LV_PART_ANY);
  lv_obj_set_style_arc_color(btn, lv_color_black(), LV_STATE_ANY | LV_PART_ANY);
  lv_obj_set_style_shadow_color(btn, lv_color_black(), LV_STATE_ANY | LV_PART_ANY);`
[image]

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2022-07-11:
Hi,

I answer here now but next time please use the Forum for "How-to" questions.

So this should work:
```c
    lv_obj_set_style_outline_color(btn1, lv_color_black(), LV_STATE_FOCUS_KEY);
```

closed 2022-07-11 by kisvegabor

Micky513673326, 2022-07-12:
Thanks for your reply, I will ask questions in the forum next time