lvgl/lvgl #3472: lv_obj_add_style/lv_obj_remove_style/... : lv_style_t* parameter is not const

geert-KLA-BE, 2022-07-11:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When I want to use const styles I cannot use the lv_obj_add_style/lv_obj_remove_style/... functions since the arguments are not const.

### To Reproduce
```c++
   template<size_t N>
    class const_style_t
    {
        private:
            lv_style_const_prop_t  m_properties[N + 1];
            lv_style_t             m_style;

        public:
            constexpr const_style_t( const lv_style_const_prop_t (&props)[N] )
                : 
                m_properties { LV_STYLE_PROP_INV },
                m_style
                {
                  .sentinel  = LV_STYLE_SENTINEL_VALUE,
                  .v_p       = { .const_props = static_cast<lv_style_const_prop_t*>(&m_properties[0]) },
                  .is_const  = 1,
                  .has_group = 0xFF
                }
            {
                for( size_t i = 0; i != N; ++i )
                {
                    m_properties[i] = props[i];
                }

                // Last element of the properties array needs to end with this value
                m_properties[ N ].prop = LV_STYLE_PROP_INV;
            };

            lv_style_t* get_c_style()
            {
                return &m_style;
            };

            operator lv_style_t* ( )
            {
                return &m_style;
            };
        }

       ...
       const const_style_t color_primary                { { LV_STYLE_CONST_TEXT_COLOR(color::primary)             } };

       ...
       lv_obj_add_style(btn, color_primary.get_c_style(), 0);   <--  style is const and function parameter lv_style_t* isn't
       lv_obj_remove_style(btn, color_primary.get_c_style(), 0);   <--  style is const and function parameter lv_style_t* isn't
```

### Expected behavior

Being able to use the lv_obj_x_style API with const lv_style.

kisvegabor, 2022-07-13:
I see no problem with making them `const`.  

A fix is pushed.

pete-pjb, 2022-07-13:
Hi @kisvegabor ,

This fix causes a warning:

```
Description	Resource	Path	Location	Type
assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]	lv_obj_style.c	/LVGL/src/core	line 100	C/C++ Problem```

Sorry really busy right now so can't look into.

Kind Regards,

Pete

kisvegabor, 2022-07-13:
And it seems working, all CI tests are passed.

commit b9a29777 (2022-07-13): fix(style): make the style parameter const in lv_obj_add/remove_style functions

fixes #3472
closed 2022-07-13 by kisvegabor