lvgl/lvgl #3480: Incorrect button registered sometimes in a btnmatrix

martaaay, 2022-07-12:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)
Sorry, I dont have this set up. Will try to get to it.

### Describe the bug
I created a simple numbers only (and backspace / enter) button matrix like this:
`    
    lv_obj_t *ta = lv_textarea_create(parent);
    lv_textarea_set_one_line(ta, true);
    lv_textarea_set_text(ta, initial_value);
    lv_textarea_set_placeholder_text(ta, place_holder);
    lv_obj_set_width(ta, 200);
    lv_obj_align(ta, LV_ALIGN_TOP_MID, 0, 10);
    lv_obj_add_event_cb(ta, textarea_event_handler, LV_EVENT_READY, ta);
    lv_obj_add_state(ta, LV_STATE_FOCUSED); /*To be sure the cursor is visible*/
        
    static const char * btnm_map[] = {"1", "2", "3", "\n",
                               "4", "5", "6", "\n",
                               "7", "8", "9", "\n",
                               LV_SYMBOL_BACKSPACE, "0", LV_SYMBOL_NEW_LINE, ""};
            
    lv_obj_t * btnm = lv_btnmatrix_create(parent);
    lv_obj_set_size(btnm, 200, 150);
    lv_obj_align(btnm, LV_ALIGN_BOTTOM_MID, 0, -10);
    lv_obj_add_event_cb(btnm, btnm_event_handler, LV_EVENT_VALUE_CHANGED, ta);
    lv_obj_clear_flag(btnm, LV_OBJ_FLAG_CLICK_FOCUSABLE); /*To keep the text area focused on button clicks*/
    lv_btnmatrix_set_map(btnm, btnm_map);
`
I found if i tap the '2' (for instance) at the top and then tap juuust below one of the bottom most keys, like the backspace, it'll
repeat the '2'. I traced it to the button matrix code in lv_btnmatrix_event (lv_btnmatrix.c:419 in 8.3):
`        
         if(indev_type == LV_INDEV_TYPE_POINTER || indev_type == LV_INDEV_TYPE_BUTTON) {
            uint16_t btn_pr;
            /*Search the pressed area*/
            lv_indev_get_point(param, &p);
            btn_pr = get_button_from_point(obj, &p);
            /*Handle the case where there is no button there*/
            if(btn_pr != LV_BTNMATRIX_BTN_NONE) {
                if(button_is_inactive(btnm->ctrl_bits[btn_pr]) == false &&
                   button_is_hidden(btnm->ctrl_bits[btn_pr]) == false) {
                    btnm->btn_id_sel = btn_pr;
                    invalidate_button_area(obj, btnm->btn_id_sel); /*Invalidate the new area*/
                }
            }
        }

        if(btnm->btn_id_sel != LV_BTNMATRIX_BTN_NONE) {
`
I believe what is happening is I'm tapping perhaps on the space around the button or maybe the shadow (?). This code segment is entered and `btn_pr == LV_BTNMATRIX_BTN_NONE`. We do not enter the scope of the `btn_pr != LV_BTNMATRIX_BTN_NONE` if statement, and soon find ourselves at the `btnm->btn_id_sel != LV_BTNMATRIX_BTN_NONE` statement below where we process the previously tapped key (which is still stored in btnm->btn_id_sel). My quick fix is to do the following:
`
        if(indev_type == LV_INDEV_TYPE_POINTER || indev_type == LV_INDEV_TYPE_BUTTON) {
            uint16_t btn_pr;
            /*Search the pressed area*/
            lv_indev_get_point(param, &p);
            btn_pr = get_button_from_point(obj, &p);
            /*Handle the case where there is no button there*/
            if(btn_pr != LV_BTNMATRIX_BTN_NONE) {
                if(button_is_inactive(btnm->ctrl_bits[btn_pr]) == false &&
                   button_is_hidden(btnm->ctrl_bits[btn_pr]) == false) {
                    btnm->btn_id_sel = btn_pr;
                    invalidate_button_area(obj, btnm->btn_id_sel); /*Invalidate the new area*/
                }
            } else {
                btnm->btn_id_sel = LV_BTNMATRIX_BTN_NONE;
            }
        }

        if(btnm->btn_id_sel != LV_BTNMATRIX_BTN_NONE) {
`
This works, but I'm wondering if this is masking an underlying problem.

### To Reproduce
(See above)

### Expected behavior
(See above)

### Screenshots or video

kisvegabor, 2022-07-13:
Hi,

It's great that you found the root of the issue.

Your fix look good to me. Could you send a Pull request?

martaaay, 2022-07-15:
Done and done:
[https://github.com/lvgl/lvgl/pull/3487](https://github.com/lvgl/lvgl/pull/3487)

kisvegabor, 2022-08-02:
As the PR is merged I close this issue.

PR #4439 (2023-08-09): Reapplied the no-button bntmatrix fix (#3480)
### Description of the feature or fix

#3480 added a fix to the button matrix when an empty area is being clicked. This fix was merged in the master branch, but not on the release/v8.3 branch, which I am currently using in a project. Therefore, I'd like to reapply the same fix to this branch as well

### Checkpoints
- [x] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- ~~Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed~~
- ~~Add [Examples] [... cut]
closed 2022-08-02 by kisvegabor