lvgl/lvgl #3485: Can no longer create nested tabviews. Suspected bug.

pete-pjb, 2022-07-14:
Hi @kisvegabor,

As of the current Master branch I can no longer created nested tabviews.

Here is some example code which I have ripped from my project, which also fails on the simulator with the latest master branch:

```c
void roomctrl_create( lv_obj_t *parent ) {

}

void config_create( lv_obj_t *parent ) {

	lv_obj_t		*tv;
	lv_obj_t		*tab;

	tv = lv_tabview_create(parent, LV_DIR_TOP, 24);
	lv_obj_set_size(tv, lv_obj_get_width(parent)-48, lv_obj_get_height(parent)-48);
	lv_obj_align(tv, LV_ALIGN_CENTER, 0, 0);  /*Align to the Centre*/
	lv_obj_set_style_border_width (tv, 1, LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_border_color (tv, lv_palette_main(LV_PALETTE_BLUE_GREY), LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_border_width (tv, 1, LV_PART_ITEMS);
	lv_obj_set_style_border_color (tv, lv_palette_main(LV_PALETTE_BLUE_GREY), LV_PART_ITEMS);

	tab = lv_tabview_add_tab(tv, "Configure System");
//	conf_sys_create(tab);
	tab = lv_tabview_add_tab(tv, "Configure Rooms");
//	conf_rooms_create(tab);
	lv_obj_invalidate(tv);

}

void sysinfo_create( lv_obj_t *parent ) {

	lv_obj_t		*tv;
	lv_obj_t		*tab;

	tv = lv_tabview_create(parent, LV_DIR_TOP, 24);
	lv_obj_set_size(tv, lv_obj_get_width(parent)-48, lv_obj_get_height(parent)-48);
	lv_obj_align(tv, LV_ALIGN_CENTER, 0, 0);  /*Align to the Centre*/
	lv_obj_set_style_border_width (tv, 1, LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_border_color (tv, lv_palette_main(LV_PALETTE_BLUE_GREY), LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_border_width (tv, 1, LV_PART_ITEMS);
	lv_obj_set_style_border_color (tv, lv_palette_main(LV_PALETTE_BLUE_GREY), LV_PART_ITEMS);

	tab = lv_tabview_add_tab(tv, "System Information 1");
//	conf_sys_create(tab);
	tab = lv_tabview_add_tab(tv, "System Information 2");
//	conf_rooms_create(tab);
	lv_obj_invalidate(tv);

}

void log_create( lv_obj_t *parent ) {

}

void my_test( void ) {

		lv_obj_t			*tab;
		lv_coord_t			scrx = lv_obj_get_width(lv_scr_act()), scry = lv_obj_get_height(lv_scr_act());

		main_tv = lv_tabview_create(lv_scr_act(), LV_DIR_TOP, 24);
		lv_obj_set_size(main_tv, scrx, scry-24);
		lv_obj_align(main_tv, LV_ALIGN_BOTTOM_LEFT, 0, 0);  /*Align to the Bottom*/
		lv_obj_set_style_border_width (main_tv, 1, LV_PART_MAIN | LV_STATE_DEFAULT);
		lv_obj_set_style_border_color (main_tv, lv_palette_main(LV_PALETTE_BLUE_GREY), LV_PART_MAIN | LV_STATE_DEFAULT);
		lv_obj_set_style_border_width (main_tv, 1, LV_PART_ITEMS);
		lv_obj_set_style_border_color (main_tv, lv_palette_main(LV_PALETTE_BLUE_GREY), LV_PART_ITEMS);

		tab = lv_tabview_add_tab(main_tv, "Room Controls");
		roomctrl_create(tab);
		tab = lv_tabview_add_tab(main_tv, "Configuration");
		config_create(tab);
		tab = lv_tabview_add_tab(main_tv, "System Information");
		sysinfo_create(tab);
		tab = lv_tabview_add_tab(main_tv, "Logs");
		log_create(tab);

		utc_current_time = lv_label_create(lv_scr_act());
		lv_obj_set_style_text_font (utc_current_time, &lv_font_roboto_12, LV_PART_MAIN | LV_STATE_DEFAULT);
		gui_load = lv_label_create(lv_scr_act());
		lv_obj_set_style_text_font (gui_load, &lv_font_roboto_12, LV_PART_MAIN | LV_STATE_DEFAULT);
		loc_current_time = lv_label_create(lv_scr_act());
		lv_obj_set_style_text_font (loc_current_time, &lv_font_roboto_12, LV_PART_MAIN | LV_STATE_DEFAULT);

	    lv_obj_invalidate(lv_scr_act());

}
```

I suspect this is something to do with the flex layout?  I currently have a fixed size screen and all screens are predefined so I am not too bothered about automatic layout, all my objects are place using ```lv_obj_set_pos();```

When the parent sizes are returned they appear to be incorrect.

This example creates the four main tabs not the secondary tabviews, maybe I need to set a flag or something that I have missed?

Any help would be much appreciated 🙂.

Kind Regards,

Pete

pete-pjb, 2022-07-14:
Hi again @kisvegabor ,

Having experimented further it seems the problem is with this code the ```get_width``` and ```get_height``` functions are returning strange numbers?:

```c
lv_obj_set_size(tv, lv_obj_get_width(parent)-48, lv_obj_get_height(parent)-48);
```

If I hard code the widths as follows all works well:
```c
lv_obj_set_size(tv,1366-48, 744-48);
```

Hope fully this is a clue?

Kind Regards,

Pete

kisvegabor, 2022-07-14:
Hi Pete,

Does it work if you call `lv_obj_update_layout(parent);` before `lv_obj_set_size(tv, lv_obj_get_width(parent)-48, lv_obj_get_height(parent)-48);`

pete-pjb, 2022-07-14:
Hi @kisvegabor ,

Yes it does indeed!

Thank you, I'll close the issue.

Kind Regards,

Pete

kisvegabor, 2022-07-14:
Ideally 

```c
lv_obj_set_size(tv, lv_obj_get_width(parent)-48, lv_obj_get_height(parent)-48);
```

 should be replaced by 
```c
lv_obj_set_style_pad_all(parent, 24, 24);
lv_obj_set_size(tv, lv_pct(100), lv_pct(100));
```

But it's really not that obvious. We should make it work without the extra `lv_obj_update_layout()` in v9 somehow.

pete-pjb, 2022-07-14:
Hi @kisvegabor ,

```c
lv_obj_set_style_pad_all(parent, 24, LV_PART_MAIN);
lv_obj_set_size(tv, lv_pct(100), lv_pct(100));
```

Also works well for me, now I know I will do things this way in future.

Thanks again for your help.

Cheers,

Pete