lvgl/lvgl #3496: GPU wait for finish callback gets called useless

nicusorcitu, 2022-07-18:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Recently I implemented the wait_for_finish() callback for NXP PXP and VGLite GPU.
I have done a lot of tests, but I do not see the performance much improved. The VGLite gets up to 10% improved with this callback. But I believe it can do more.

With the wait_for_finish() implementation the CPU and GPU shall work in parallel for an amount of time, so that CPU can advance further while the GPU is working on blit/draw.
Whenever the CPU needs the GPU completion it shall call this callback
"draw_ctx->wait_for_finish(draw_ctx)"

I understand the reason of having the GPU completion before flushing the draw buffer: 
draw_buf_flush() --> draw_ctx->wait_for_finish(draw_ctx)

But not sure why is it needed all the time from here:
lv_draw_sw_blend() --> draw_ctx->wait_for_finish(draw_ctx)

The blend callback is also implemented and handled by GPU. VGLite for example can queue the command buffers, it does not have to wait for the previous operation to complete.
I know that some parts might be blit by CPU (if the area size is smaller than LV_GPU_NXP_VG_LITE_BLIT_SIZE_LIMIT or if GPU does not support that). But having the GPU wait for finish called from EVERY blend call will actually flood it, most of the time there is nothing to wait from the GPU. Please see the logs attached.

Maybe the area we are about to blend with the CPU is not intersecting the one that GPU is currently process. I think this check needs to be added here.

### To Reproduce
Run the bechmark application and trace how often the wait_for_finish() gets called. You might look how many times the wait_for_finish() gets called between two draw_buf_flush() for each frame. And there are at least 30 frames per second.

### Expected behavior
Reduce the number of wait_for_finish() calls.

This is not really an issue, but more like an improvement. So I open this discussion to clarify it.

nicusorcitu, 2022-07-18:
[wait_for_finish_flood.txt](https://github.com/lvgl/lvgl/files/9130399/wait_for_finish_flood.txt)

nicusorcitu, 2022-07-19:
> So ideally everything can be rendered by GPU, right?

That would a very optimistic scope. But I know it will be difficult to achieve it. I am thinking maybe we can have a check to verify if the drawing area by CPU will get intersected with the one pending by GPU. If not, there is no reason to wait for GPU completion.
But indeed, maybe this check cost too much.

With the wait_for_finish() implementation I got the below results. I have run the benchmark application from lvgl.
[image]

kisvegabor, 2022-07-23:
Thanks for the benchmark. 

I think the simplest way we can take is adding an `lv_area_t` parameter to `wait_for_finish()` to tell which area will be updated now. (The parameter can be NULL if we are waiting before the `flush_cb` ). 

The GPU can keep track of the areas it is busy with and if the areas intersect the GPU can wait for finish.

What do you think?

nicusorcitu, 2022-07-24:
Hi Gabor,

I am out of office till 1 August but will get back to you then.
At a first think it seems a good idea!

Thank you for your supoort.
Nicu

________________________________
De la: Gabor Kiss-Vamosi ***@***.***>
Trimis: Saturday, July 23, 2022 8:48:15 PM
Către: lvgl/lvgl ***@***.***>
Cc: Nicușor Cîțu ***@***.***>; Author ***@***.***>
Subiect: [EXT] Re: [lvgl/lvgl] GPU wait for finish callback gets called useless (Issue #3496)

Caution: EXT Email

Thanks for the benchmark.

I think the simplest way we can take is adding an lv_area_t parameter to wait_for_finish() to tell which area will be updated now. (The parameter can be NULL if we are waiting before the flush_cb ).

The GPU can keep track of the areas it is busy with and if the areas intersect the GPU can wait for finish.

What do you think?

nicusorcitu, 2022-08-01:
Hi Gabor,

If you can provide me the patch with the new argument in `wait_for_finish()`  then I can start working on the GPU side to keep tracking the pending areas.
Which way to you think will be better: using a static array, dynamic array or a linked list? I think dynamic array will be good and to play with the `lv_mem_realloc()` in case we need to grow/shrink the size.
But I have to measure also the CPU cost for all of this extra code.

Thanks,
Nicu.

kisvegabor, 2022-08-02:
I pushed the update to the `feat/wait_for_finish-area-param` branch.

> Which way to you think will be better: using a static array, dynamic array or a linked list? 

Good question. I agree that the dynamic array would be the best choice. 
I guess there will be only a few areas rendered in parallel so checking all the saved areas shouldn't cause much overhead.

nicusorcitu, 2022-08-02:
Thanks for the branch, I will check it and I will let you know the status.

nicusorcitu, 2022-08-05:
Hi Gabor,

I used (for the beginning) a static array of 10 elements to keep tracking the areas flushed to the GPU (which might be pending).
In `wait_for_finish()` I checked the intersection of the area is about to blend with the areas from the array. I checked the code with traces to be sure everything works as expected.

Unfortunately the rectangle result is even worse, 83 FPS. All the other benchmark tests also are suffering for performance decrease.
It looks like the extra code consume more than simple waiting useless for VGLite GPU.

In the last days I add the SystemView support to MCUX so I will try doing some profiling to see what else can be improved.
But so far this "improvement" I suggested does not seem to be efficient.

Thanks, Nicu.

kisvegabor, 2022-08-07:
That's sad. 

> In the last days I add the SystemView support to MCUX so I will try doing some profiling to see what else can be improved.

Looking forward to the results.

nicusorcitu, 2022-08-08:
BTW, I saw you reworked on master branch the need_argb8565_support() so please add the "draw_ctx" argument everywhere is called.

Is a very long process in order to create PR from NXPmicro/lvgl repository, so I attached the minor patch here.
[0008-Fix-need_argb8565_support-argument.txt](https://github.com/lvgl/lvgl/files/9280016/0008-Fix-need_argb8565_support-argument.txt)

nicusorcitu, 2022-08-08:
I add for reference also the patch that checks the area intersection. 
[0010-Add-static-array-to-hold-the-pending-GPU-draw-areas-.txt](https://github.com/lvgl/lvgl/files/9281420/0010-Add-static-array-to-hold-the-pending-GPU-draw-areas-.txt)

The comparing results are below:

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/nxf68195/AppData/Local/Temp/2/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/nxf68195/AppData/Local/Temp/2/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

  | NO   wait_for_finish() | wait_for_finish() | wait_for_finish()   BUT not from blend() | wait_for_finish()   + check areas
-- | -- | -- | -- | --
Rectangle | 83 | 93 | 103 | 86
Circle | 64 | 76 | 90 | 69
Image RGB | 70 | 77 | 90 | 66
Image ARGB | 38 | 36 | 40 | 38
Arc thin | 58 | 60 | 70 | 59

</body>

</html>

nicusorcitu, 2022-08-09:
“NO wait_for_finish()” means there is no callback implementation, cureently upstreamed version in v8.2.0. E.G. The CPU flushes a command to the GPU and waits right away for its completion.
________________________________
De la: Gabor Kiss-Vamosi ***@***.***>
Trimis: Tuesday, August 9, 2022 6:30:43 PM
Către: lvgl/lvgl ***@***.***>
Cc: Nicușor Cîțu ***@***.***>; Author ***@***.***>
Subiect: [EXT] Re: [lvgl/lvgl] GPU wait for finish callback gets called useless (Issue #3496)

Caution: EXT Email

How is it possible that wait_for_finish() BUT not from blend() is faster than NO wait_for_finish()? It seems having one wait (before flush?) is faster than no wait at all.

kisvegabor, 2022-08-09:
I see. So `NO wait_for_finish()` is kind if "blocking" mode and it's the slowest.

In the benchmark with `wait_for_finish() BUT not from blend()` it could happen that the GPU and the CPU are writing the same are right? And does the GPU waits for itself before bleneding a new area? 

It seems usually there is ~2 ms difference between `wait_for_finish()` and `wait_for_finish() BUT not from blend()`. 
 
I'm not sure it's wort chasing it. I'd be more interested in the other 10 ms e..g. with `Image RGB` which is a very GPU friendly operation. Do you know what is the limiting factor here? Memory bandwidth? Or GPU frequency? Or some LVGL overhead?

nicusorcitu, 2022-08-10:
> I see. So NO wait_for_finish() is kind of "blocking" mode and it's the slowest.
>
Yes, without wait_for_finish() implementation, the CPU and GPU can not work in parallel.

> In the benchmark with wait_for_finish() BUT not from blend() could it happen that the GPU and the CPU are writing the same area, right? And does the GPU waits for itself before blending a new area?
>
Yes, it can happen. I actually have a set of logs that clearly shows there are some area intersections. E.G. "lv_draw_nxp_shall_wait: TRUE" from the logs attached.
[logs_check_area.txt](https://github.com/lvgl/lvgl/files/9296899/logs_check_area.txt)
The VGLite GPU driver can queue (`vg_lite_flush()`) more commands and the `wait_for_finish()` (`vg_lite_finish()`) shall wait for all of them to complete. It does not necessarily have to wait for itself before blending/drawing a new area, except if the area is intersecting with another one processed by itself or by the CPU.
The PXP does not have this ability, it can blend only once at a time. It has to wait for itself if a new area is about to blend.

I tried to do some more inline functions and use a static `valid_area[i]` variable instead of using lvgl functions `if(lv_area_get_size(&wait_area[i]) == 1)` or `lv_area_set(&wait_area[i], 0, 0, 0, 0);` but not much difference. I was expecting not to obtain the all 10% difference between `wait_for_finish() BUT not from blend()` and `wait_for_finish()` but at least 5% more... Unfortunately the results are as shown in the table.

> I'd be more interested in the other 10 ms e..g. with Image RGB which is a very GPU friendly operation. Do you know what is the limiting factor here? Memory bandwidth? Or GPU frequency? Or some LVGL overhead?
>
I think all of the factors are valid. The CPU is at 1Ghz while PXP is 240MHz and VGLite is 500MHz. The PXP is known that it can access slower the flash memory, but it has a bigger performance accessing OCRAM.
I did in the past some measurements with a us timer implemented in SDK and I got these results - tested with O2 optimisation.
Remember that image was stored in FLASH, in txt section. The Blit time is the execution time of `lv_draw_sw_blend_basic()` or of the GPU callback `nxp_draw_ctx->blend()`.
[image]
I am going to do some more measurements with the SystemView and doublecheck these table timestamps and also see if I can find out more.

nicusorcitu, 2022-08-10:
Off Topic:
Why is there need for a new fill screen (with white color) every time the screen is refreshed? E.G. at every 30ms?
Is this a kind of init screen buffer?

kisvegabor, 2022-08-15:
> Do you mean the difference between CPU and VGLite? How did you calculate 10ms?

With `wait_for_finish() BUT not from blend()` and `Image RGB` there is 90 FPS -> 11ms. I guess wait for finish takes 1ms but what about the other 10ms?

> Why is there need for a new fill screen (with white color) every time the screen is refreshed? E.G. at every 30ms?
Is this a kind of init screen buffer? Because it looks like is taking ~4ms for each frame.

Have you enabled `disp_drv->full_refresh`? If so, LVGL will redraw the whole screen even if only one pixel changes.

nicusorcitu, 2022-08-18:
Hi Gabor,

I have improved the systemview support and now I can see all the tasks and messages correctly.
The default benchmark application uses rgb100x100 and that one is not supported by vglite, due to the stride not aligned to 16 pixels.

I have changed the benchmark test to blit the rgb 128x128, below are the results with the `wait_for_finish()` implemented.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/nxf68195/AppData/Local/Temp/2/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/nxf68195/AppData/Local/Temp/2/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

  | VGLite | PXP | CPU
-- | -- | -- | --
RGB 128x128 | 41 | 31 | 59

</body>

</html>

I placed the systemview to see what happens on PXP and VGLite due to a frame time (one run of `_lv_disp_refr_timer()`). There is no big gap from main lvgl to see where the system is busy. 

1. The blit with PXP takes longer than CPU (-O2) but it frees up a lot of CPU. PXP can process only one command at a time, it can queue up another one in the NEXT register but that is currently not implemented. So PXP jobs are serialized.

[image]

What I can tell from this picture is the Fill Screen takes ~4ms and a single image blit takes ~1ms  (selected area) same values I got with a hight resolution timer from SDK and posted previously. There are 13 full blits plus some others that takes less time and probably are fallback to the CPU due to the small size. I would say that 13 + 4 = 17ms is PXP working. Small of this time is done in paralel with the CPU (due to `wait_for_finish()`).

The total frame runs for 37ms (this includes the overhead of the systemview calls - 262 extra function calls to trace).
From 37 ms, the lvgl task app takes 16ms E.G. the CPU is busy this time. The Idle task runs for 15.5 ms. 
I assume the 5.5ms is the systemview overhead. So without systemview, the frame time would be 31.5ms.

Having that, the CPU Usage shall be at 50% - we need to fix that in lvgl as currently it is not aware of the FreeRTOS idle task.

2. The blit with VGLite implies an extra FreeRTOS task used to queue the GPU commands.
 
[image]

Fill screen with VGLite takes ~4.5 ms. I see 16 blits processed by GPU, each taking ~650us (selected area). It is less than in the table I provided with SDK timer because the `wait_for_completion()` now allows to queue more GPU commands at a time so the CPU and GPU can work in parallel.
The total frame runs for 29ms (including the systemview overhead).
The lvgl task runs for 15,5ms, the vglite queue task time is 1,1 ms and the Idle time is 7.5ms. The extra 5-5ms is the systemview overhead - now I have more markers.

Conclusions:
1. On smaller images the PXP and VGLite does not perform better than CPU. We are also comparing with the CPU which runs at 1Ghz. But it frees up a lot of CPU. Need to fix the CPU usage.
2. There is no bottleneck that I can see and investigate the time spend on that one.
4. PXP can be improved to queue another command while is processing.
5. Investigate how VGLite can also be improved even more, maybe by caching some of the previous operations.

kisvegabor, 2022-08-19:
Very helpful measurements!

Some notes:

  | VGLite | PXP | CPU
-- | -- | -- | --
RGB 128x128 | 41 | 31 | 59

It's in `ms` or FPS?

---

It takes ~4ms for both PXP and VGLite to fill the screen.  From this we can calculate the real memory speed. 
On 1280x720 resolution it means the throughput is 
1280 x 720 / 4=225k pixels/ms. Or 450MB/sec. It's the same speed as a 100MHz MCU writing into its internal RAM 32bit at once. (100MHz x 4bytes  = 400MB/sec). On 1 GHz and 32 bit the internal RAM access would be  4GB/sec -> 10 times faster. I guess the read is quite the same.  

So assuming that 
- LCDIF reads the whole frame buffer once in every 30ms
- we fill the screen with a color
- draw some widgets, read back some pixels to calculate opacity. let's say also in the size of the whole screen

So counting only the read/write time 1280 x 720 x 3 screens x 2 bytes = 5.2MB data. At 450MB /sec rate it's 11ms. So 36% CPU usage (with 30ms period time) for calculating nothing, just a fill the screen twice.  

I had meeting with of your engineers and he mentioned that the code now runs from a slower quad SPI flash, but Octo SPI flash is also available.  How much difference can it make?

---
From your measurement with PXP the whole time is 37ms: 
- lvgl task app:  16ms  
- idle task: 15.5 ms.
- systemview overhead: 5.5ms 

Form where the idle time comes from? This the sleep while we are waiting for the next redraw period, or waiting for PXP?

What CPU usage do you see on LVGL's CPU usage indicator?

nicusorcitu, 2022-08-22:
1. This is FPS measurements. I need to find a way to improve this, the only way would be to make CPU and GPU work more time in parallel. That would imply changes on draw mechanism.
2. Do you recall the engineer name? I will do some tests in order to measure the difference.
3. The idle time comes from CPU waiting for PXP to complete. I will look to implement the NEXT command on PXP. It shall improve the performance.

Currently, with PXP enabled for example the CPU indicator shows wrongly 86% usage.
After I changed the "lvgl" app task to not delay anylonger between executing timers:
```
     for (;;)
     {
         lv_task_handler();
-        vTaskDelay(5);
+        vTaskDelay(0);
     }
```
the CPU usage goes to almost wrongly 100% usage, as I would expected.

The CPU usage indicator is another item I wanted to discuss with you, as it does not take account of what happens during a frame time and who is processing what. It does not care if there is a GPU or not. It simply calculate how much time did it took to process a frame. I have done in the past a local patch for a team that wanted to test the performance and the CPU offload, and I used an SDK high resolution timer to exclude from CPU usage the time the CPU waits for GPU. It works fine on PXP.
But on VGLite I see that also the `vg_lite_flush` command can put the CPU into idle. It looks that in some cases the `vg_lite_flush` actually works like a `vg_lite_finish`, e.g. it has to wait for some previous commands to complete before flushing another one. So not all of the idle time was counted on VGLite with my approach.

The best way to fix the CPU idle would be to add an FreeRTOS idle hook function and count the time spend on it.
PXP can run also without FreeRTOS so I think the count of `wait_for_completion` shall work just fine without FreeRTOS.

kisvegabor, 2022-08-28:
There is something I'm not sure about RTOS and GPU. Is it really worth yielding from the task while waiting for the GPU?  
I mean usually the time needed by the GPU to finish is very little, e.g. < 1 ms. So will the context switch, and chance of not getting back control in time is worth yielding?

nicusorcitu, 2022-08-30:
In some cases, e.g. the fill screen, the time to wait is 4 ms, so it worth yielding. But I agree with you for cases where does not have to wait for GPU.

The real impact while using the GPU is that each command is submitted at once and waiting for its completion (flush sometimes acts like the finish).

I am thinking to add the "flush" meaning on the GPU, e,g, instead of the repeated sequence:
"initbuf + blit + flush" to have
"initbuf1, blit1, initbuf2, blit2 ... initbufx, blitx     flush or finish"

The GPU has two command buffers, that's why after the second flush it will have to wait for the first command to complete.
Instead of flushing each command I will flush a bunch of commands. Those command might include also the small area processing, that currently does not In this case the yield will certainly be required. Will see what happens.

Currently I am working to implement the CPU usage on FreeRTOS because I am interested to see exactly the benefit time when GPU will offload the CPU.

kisvegabor, 2022-09-01:
> The real impact while using the GPU is that each command is submitted at once and waiting for its completion (flush sometimes acts like the finish).

It assumes that the GPU can handle an area on its own and doesn't need to wait for CPU before blit3.

This idea is similar to what @GorgonMeducer mentioned in #3643.

nicusorcitu, 2022-09-16:
Till now I concluded that checking if the CPU area is intersecting with the GPU area actually has a bigger cost than simply calling the `wait_for_finish` useless.

I think we can close this ticket as the multitask implementation will mitigate the need of "wait for GPU completion".

nicusorcitu, 2022-09-16:
Won't fix.

closed 2022-09-16 by nicusorcitu