lvgl/lvgl #3514: ../../lv_conf.h: No such file or directory

Sarah-C, 2022-07-23:
It's back again.

I've just run the new "factory" build for Lilygo T-QT...

            from C:\Users\sarah\Documents\Arduino\sketches\T-QT\factory\factory.ino:5:
c:\users\sarah\documents\arduino\libraries\lvgl\src\lv_conf_internal.h:41:18: fatal error: ../../lv_conf.h: No such file or directory
#include "../../lv_conf.h" /Else assume lv_conf.h is next to the lvgl folder/
^~~~~~~~~~~~~~~~~
compilation terminated.

kisvegabor, 2022-07-24:
Have you done [these](https://docs.lvgl.io/master/get-started/platforms/arduino.html#configure-lvgl) steps?

Sarah-C, 2022-07-31:
I'll get back to it this week! Thread not dead.

met989, 2022-08-16:
it was a error in line 41 in https://github.com/lvgl/lvgl/blob/master/src/lv_conf_internal.h
repleace 
#include "../../lv_conf.h"
with
#include "../lv_conf.h"
work after configure https://docs.lvgl.io/master/get-started/platforms/arduino.html#configure-lvgl

kisvegabor, 2022-08-17:
`#include "../../lv_conf.h"` seems correct as `lv_conf_internal.h` is in the `lvgl/src` folder. So `../../` goes to the parent folder of `lvgl` where `lv_conf.h` should be.

dorel50, 2023-03-17:
C:\Program Files (x86)\Arduino\libraries\T-Display-S3-main\example\factory\lvgl.h:24:10: fatal error: src/misc/lv_log.h: No such file or directory
 #include "src/misc/lv_log.h"
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
Aceeasi problema !!

suqmadiq2ama, 2023-07-03:
same problem Compilation error: src/misc/lv_log.h: No such file or directory

DenDonnar, 2023-10-27:
I've got the same issue. Tried some couple of different lvgl versions (uning Arduino IDE and some from github)
[#include "../lv_conf.h"](c:\arduino\libraries\lvgl\src/lv_conf_internal.h:41:18: fatal error: ../lv_conf.h: No such file or directory)

kisvegabor, 2023-10-29:
Have you copied `lv_conf.h` as described [here](https://docs.lvgl.io/8.3/get-started/platforms/arduino.html#configure-lvgl)?

byNMI, 2024-03-18:
so, not working

kisvegabor, 2024-03-18:
Could you share 
1. a project the we can try to compile
2.  the required steps/links to set up an environment?

closed 2024-06-03 by lvgl-bot

tsgouros, 2024-06-30:
I have exactly this issue. From the Arduino IDE, with the long and ungainly path shortened to XX.

> In file included from XX/libraries/lvgl/src/../src/lv_init.h:17,
>                  from XX/libraries/lvgl/src/../lvgl.h:24,
>                  from XX/libraries/lvgl/src/lvgl.h:16,
>                  from XX/Teensy_40_NMEA2000_depth_and_temperature_480x320_LCD.ino:6:
> XX/libraries/lvgl/src/../src/lv_conf_internal.h:59:18: fatal error: ../../lv_conf.h: No such file or directory
>    59 |         #include "../../lv_conf.h"                /*Else assume lv_conf.h is next to the lvgl folder*/
>       |                  ^~~~~~~~~~~~~~~~~
> compilation terminated.
> exit status 1

I believe that lv_conf.h is exactly where it is supposed to be, NEXT TO the lvgl folder, with the "#if 0" changed to "#if 1". If I open lv_conf_internal.h in an editor and then look for ../../lv_conf.h, my editor finds it no problem. A tar of the code I'm using: https://sgouros.com/lvgl-sample.tgz. Any suggestions welcome, thank you.

danserrano23, 2024-11-01:
Any solution now?

maneeshsit, 2026-10-23:
Getting the same error -  #include "../../../../lv_conf.h"
Is there a known fix to the issue