lvgl/lvgl #3519: Modifying the status color of the Switch does not take effect

evansherry, 2022-07-25:
The input device I am using is an encoder.

The input device I am using is an encoder.
The background color didn't work when I set the LV_PART_INDICATOR state to LV_STATE_FOCUSED, but this works fine with LV_PART_KNOB.
When I call
```
 lv_obj_add_state(sw, LV_STATE_CHECKED)
lv_obj_add_event_cb(sw, switch_event, LV_EVENT_ALL, NULL) 
```
cannot monitor LV_STATE_FOCUSED and LV_STATE_CHECKED in switch_event.

Is there any way to modify the background color of the LV_STATE_FOCUSED state LV_PART_INDICATOR?
Can I listen to callbacks for state changes?

```
        lv_obj_t *sw = lv_obj_create(parent);
        lv_obj_set_size(sw, 100, 50)

        lv_obj_set_style_bg_color(sw, color_141414, LV_PART_MAIN);
        // focused & unchecked
        lv_obj_set_style_bg_color(sw, color_333333, LV_PART_MAIN | LV_STATE_FOCUSED);

        // LV_PART_INDICATOR
        lv_obj_set_style_bg_color(sw, lv_color_make(0x14, 0x22, 0x33), LV_PART_INDICATOR);
        // focused
        lv_obj_set_style_bg_color(sw, color_61a6ff, LV_PART_INDICATOR | LV_STATE_FOCUSED);

        // LV_PART_KNOB
        lv_obj_set_style_bg_color(sw, lv_color_white(), LV_PART_KNOB | LV_STATE_CHECKED | LV_STATE_FOCUSED);
        // checked
        lv_obj_set_style_bg_color(sw, color_333333, LV_PART_KNOB | LV_STATE_CHECKED);
        // focused
        lv_obj_set_style_bg_color(sw, color_999999, LV_PART_KNOB | LV_STATE_FOCUSED);
        
        lv_obj_set_style_bg_color(sw, color_494949, LV_PART_KNOB);
```

kisvegabor, 2022-07-25:
Hi,

I believe it's just a typo but 
```c
lv_obj_t *sw = lv_obj_create(parent);
```

should be
```c
lv_obj_t *sw = lv_switch_create(parent);
```

I've tested the styling with this code:
```c
  //Styles switch
  lv_obj_t *sw = lv_switch_create(lv_scr_act());
  lv_obj_set_pos(sw, 10, 10);
  lv_obj_set_style_bg_color(sw, lv_color_hex3(0xafa), LV_PART_INDICATOR | LV_STATE_CHECKED); //light green
  lv_obj_set_style_bg_color(sw, lv_color_hex3(0x0f0), LV_PART_INDICATOR | LV_STATE_FOCUSED); //bright green

  //Default switch
  sw = lv_switch_create(lv_scr_act());
  lv_obj_set_pos(sw, 100, 10);
```

And it seems working well to me:
[image]

> cannot monitor LV_STATE_FOCUSED and LV_STATE_CHECKED in switch_event.

For `LV_STATE_FOCUSED` `LV_EVENT_FOCUSED` is sent.

For state change `LV_EVENT_VALUE_CHANGED` is sent. See this [example](https://docs.lvgl.io/master/widgets/core/switch.html#simple-switch).

evansherry, 2022-07-26:
@kisvegabor  Sorry, the above typo should be lv_switch_create.

I re-tested, there is no problem on the virtual machine, but it is not displayed on the encoder.

virtual machine
[image]

encoder
[image]

switch test code
```
    lv_obj_set_flex_flow(lv_scr_act(), LV_FLEX_FLOW_COLUMN);
    lv_obj_set_flex_align(lv_scr_act(), LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);

    lv_obj_t *sw;
    sw = create_switch();
    lv_obj_add_state(sw, LV_STATE_CHECKED);
    lv_obj_clear_state(sw, LV_STATE_FOCUSED);

    sw = create_switch();
    lv_obj_add_state(sw, LV_STATE_FOCUSED);
    lv_obj_clear_state(sw, LV_STATE_CHECKED);

    sw = create_switch();
    lv_obj_add_state(sw, LV_STATE_CHECKED | LV_STATE_FOCUSED);

    sw = create_switch();
    lv_obj_clear_state(sw, LV_STATE_CHECKED);
    lv_obj_clear_state(sw, LV_STATE_FOCUSED);

```

create_switch
```
lv_obj_t *create_switch(void)
{

    lv_obj_t *sw = lv_switch_create(lv_scr_act());
    lv_obj_set_size(sw, 100, 50);
    lv_obj_set_pos(sw, 10, 10);

    lv_obj_set_style_pad_all(sw, -5, LV_PART_KNOB);

    // main
    lv_obj_set_style_bg_color(sw, lv_color_hex(0x141414), LV_PART_MAIN);
    lv_obj_set_style_bg_color(sw, lv_color_hex(0x333333), LV_PART_MAIN | LV_STATE_FOCUSED);

    // indicator
    lv_obj_set_style_bg_color(sw, lv_color_make(0x14, 0x22, 0x33), LV_PART_INDICATOR);
    lv_obj_set_style_bg_color(sw, lv_color_hex(0x61a6ff), LV_PART_INDICATOR | LV_STATE_FOCUSED);

    // knob
    lv_obj_set_style_bg_color(sw, lv_color_white(), LV_PART_KNOB | LV_STATE_CHECKED | LV_STATE_FOCUSED);
    lv_obj_set_style_bg_color(sw, lv_color_hex(0x333333), LV_PART_KNOB | LV_STATE_CHECKED);
    lv_obj_set_style_bg_color(sw, lv_color_hex(0x999999), LV_PART_KNOB | LV_STATE_FOCUSED);
    lv_obj_set_style_bg_color(sw, lv_color_hex(0x494949), LV_PART_KNOB);

    return sw;
}

```
I don't know what's causing it, the effect is very different.

kisvegabor, 2022-07-27:
Two thing are very strange to me. 
But first let's see how the default styles of the switch are set in the theme:
```c
        /*Related to KNOB*/
        lv_obj_add_style(obj, &styles->knob, LV_PART_KNOB);
        lv_obj_add_style(obj, &styles->bg_color_white, LV_PART_KNOB);
        lv_obj_add_style(obj, &styles->switch_knob, LV_PART_KNOB);
        lv_obj_add_style(obj, &styles->disabled, LV_PART_KNOB | LV_STATE_DISABLED);
```
1. The knob of the first switch looks smaller. Here you can see there are no styles on the non-default states of the knob so I can't see why does it look different on different states in real hardware.
https://github.com/lvgl/lvgl/blob/55e95ed351e96b93c03518c2d6504bb1915d7b86/src/extra/themes/default/lv_theme_default.c#L912-L915

2.  From where does the dark blue indicator color of the first switch in the simulator?

It's interesting that both issues are with the first switch. What if you duplicate the first switch to have 5 switches in total?

It looks like this for me which seems correct (please confirm):
[image]

evansherry, 2022-08-04:
I am also confused by these two questions.
The first light color, when not in focus, is 0x142233, but it's not blue.
I don't know the cause of this problem, I can solve the requirement by customizing a switch

closed 2022-08-18 by evansherry