lvgl/lvgl #3520: using about user_data of lv_group

yin2021, 2022-07-26:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

There is a member named "user_data" in lv_group_t. However, there is not any function to set or get the user_data in lv_group.h. So, i wonder whether it is temporary or permanent that developers can not operate the user_data.

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2022-07-27:
There is no dedicated API because we use the `user_data` of `groups` (and the others too except `lv_obj_t`) primarily only for the Micropython binding.  

You can still use it like `group->user_data = &value`.

Just out of curiosity, for what purpose  do use the `user_data` of a  `group`?

yin2021, 2022-07-28:
> There is no dedicated API because we use the `user_data` of `groups` (and the others too except `lv_obj_t`) primarily only for the Micropython binding.
> 
> You can still use it like `group->user_data = &value`.
> 
> Just out of curiosity, for what purpose do use the `user_data` of a `group`?

First, thank you for your reply! The framework of our project is developed by C, and we hope to develop UI by Lua. So,  there will be  a lot of data conversion between C and Lua.  It is difficult for swig to convert some data type such as void* (type of user_data is void *) . In lv_group,  we find user_data of lv_group, but no API for the user_data. In developing, we use user_data of lv_obj many times and think it is very useful, so we think the user_data of  lv_group maybe as useful as lv_obj(if so, we need to convert the user_data of lv_group by ourself not swig. Your answer indicates we are wrong) . In fact, we have not be in a situation we must to use user_data of lv_group and we just prepare for possible future situations.

kisvegabor, 2022-08-01:
I see, as I mentioned the main use cases for user data are the bindings, so it perfectly makes sense to use them in a Lua binding. 

I see no problem with adding API functions but if  complete groups we should add API for other modules too (e.g. indev, display, animation, image decoders, etc).

closed 2022-08-23 by lvgl-bot