lvgl/lvgl #3522: "Core 1 panic'ed (LoadProhibited). Exception was unhandled" when calling roller_1() function to display roller using LVGL library

bil7317, 2022-07-26:
### Issue:

I get this error when I call "roller_1();" after WiFi Scan and want display the names in roller so it can be selected by user.Keep in mind that the esp works fine for all other codes so it means that esp is not damaged. I have already selected 3MB partition scheme and currrently only using 51% of memory. I have also tried multithreading but the result was same. I have also tried flashing ESP but I was getting the same result.

Arduino IDE

ESP 32 WROOM

LVGL 9.0

ILI9488 touch display using SPI communication

### Code:
```
#include <lvgl.h>
#include "ui.h"
#include <Ticker.h>
#include <TFT_eSPI.h>
#include "WiFi.h"

int lop=1;
int l=0;
String str[15];
int n;
String n_lin;
String test;

#define LVGL_TICK_PERIOD 60

Ticker tick;   // timer for interrupt handler 
TFT_eSPI tft = TFT_eSPI(); // TFT instance

static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf1[480 * 320 / 10];

lv_obj_t * slider_label;
 
// Display flushing 
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
  uint16_t c;

  tft.startWrite(); /* Start new TFT transaction */
  tft.setAddrWindow(area->x1, area->y1, (area->x2 - area->x1 + 1), (area->y2 - area->y1 
  + 1)); /* set the working window */
  for (int y = area->y1; y <= area->y2; y++) {
    for (int x = area->x1; x <= area->x2; x++) {
      c = color_p->full;
      tft.writeColor(c, 1);
      color_p++;
    }
  }
  tft.endWrite(); // terminate TFT transaction
  lv_disp_flush_ready(disp); // tell lvgl that flushing is done 
}

void my_touchpad_read(lv_indev_drv_t * indev_driver, lv_indev_data_t * data)
{
    uint16_t touchX, touchY;

    bool touched = tft.getTouch(&touchX, &touchY, 600);

    if(!touched)
    {
      //return false;
    }

    if(touchX>480 || touchY > 320)
    {
 
    }
    else
    {
      data->state = touched ? LV_INDEV_STATE_PR : LV_INDEV_STATE_REL; 
      data->point.x = touchX;
      data->point.y = touchY;
    }
}

void setup() {

  Serial.begin(115200); // prepare for possible serial debug 

  lv_init();
 
  tft.begin(); // TFT init 
  tft.setRotation(3);
  //test[] = "test 1\ntest 2\ntest 3\ntest 4\ntest 5";
  uint16_t calData[5] = {299, 3588, 348, 3474, 1};
  tft.setTouch(calData);

  lv_disp_draw_buf_init(&draw_buf, buf1, NULL, 480 * 320 / 10);  //Initialize the 
  display buffer.

  //Initialize the display
  static lv_disp_drv_t disp_drv;        //Descriptor of a display driver
  lv_disp_drv_init(&disp_drv);          //Basic initialization
  disp_drv.flush_cb = my_disp_flush;    //Set your driver function
  disp_drv.draw_buf = &draw_buf;        //Assign the buffer to the display
  disp_drv.hor_res = 480;   //Set the horizontal resolution of the display
  disp_drv.ver_res = 320;   //Set the vertical resolution of the display
  lv_disp_drv_register(&disp_drv);      //Finally register the driver

  static lv_indev_drv_t indev_drv;           //Descriptor of a input device driver
  lv_indev_drv_init(&indev_drv);             //Basic initialization
  indev_drv.type = LV_INDEV_TYPE_POINTER;    //Touch pad is a pointer-like device
  indev_drv.read_cb = my_touchpad_read;      //Set your driver function
  lv_indev_drv_register(&indev_drv);         //Finally register the driver

  WiFi.mode(WIFI_STA);
  WiFi.disconnect();

  ui_init();
 
  strcpy(w_name,"1. hallo\n");
  
  
  n_lin = "\n";

}

void loop() {

  lv_task_handler(); /* let the GUI do its work */
  pinMode(13,OUTPUT);
  if(wif == 1)
  {
    digitalWrite(13,LOW);
    while(lop)
    {
          Serial.println("scan start");

      // WiFi.scanNetworks will return the number of networks found
      n = WiFi.scanNetworks();
      Serial.println("scan done");
      if (n == 0) {
          Serial.println("no networks found");
      } else {
          Serial.print(n);
          Serial.println(" networks found");
          for (int i = 0; i < n; ++i) {
              str[i]= WiFi.SSID(i);
              Serial.print("\n");
              delay(10);
              
              lop = 0;
              wif = 0;
              
                                      }
              }
              Serial.println("");

              // Wait a bit before scanning again
              delay(1000);
             
              
    }
    lop =1;

     roller_1(); //  I get Error when I call this function. 

     
  }
  
  delay(5);

  if(k==1)
  {

    Serial.println(buf);
     k=0;
     memset(&buf[0], 0, sizeof(buf));
   }
  
 }

void roller_1(void)
{
    lv_obj_t * roller1 = lv_roller_create(lv_scr_act());
    lv_roller_set_options(roller1,w_name,LV_ROLLER_MODE_INFINITE);

    lv_roller_set_visible_row_count(roller1, 4);
    lv_obj_center(roller1);
    lv_obj_add_event_cb(roller1, event_handler, LV_EVENT_ALL, NULL);
}

static void event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);
    if(code == LV_EVENT_VALUE_CHANGED) {
        char buf[32];
        lv_roller_get_selected_str(obj, buf, sizeof(buf));
        LV_LOG_USER("Selected month: %s\n", buf);
    }
}
```

### Error:
16:32:15.966 -> E (40) psram: PSRAM ID read error: 0xffffffff
16:32:28.301 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
16:32:28.347 -> Core 1 register dump:
16:32:28.347 -> PC      : 0x4016d087  PS      : 0x00060430  A0      : 0x800d11e8  A1      : 0x3ffb1e20  
16:32:28.347 -> A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3ffb0001  A5      : 0x80000020  
16:32:28.347 -> A6      : 0x00000008  A7      : 0x00000001  A8      : 0x800ea12d  A9      : 0x3ffb1d90  
16:32:28.347 -> A10     : 0x3ffcac04  A11     : 0x0000002a  A12     : 0x00000001  A13     : 0x3ffc9c98  
16:32:28.347 -> A14     : 0x3ffc95b8  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
16:32:28.394 -> EXCVADDR: 0x0000001c  LBEG    : 0x400ee4f8  LEND    : 0x400ee516  LCOUNT  : 0x00000004  
16:32:28.394 -> 
16:32:28.394 -> ELF file SHA256: 0000000000000000
16:32:28.394 -> 
16:32:28.394 -> Backtrace: 0x4016d087:0x3ffb1e20 0x400d11e5:0x3ffb1e40 0x400d0ee9:0x3ffb1e60 0x400d1c7d:0x3ffb1e80 0x400d1d36:0x3ffb1ea0 0x400d22ac:0x3ffb1ee0 0x400d29cc:0x3ffb1f00 0x400eeba1:0x3ffb1f50 0x400d13ff:0x3ffb1f80 0x400fe099:0x3ffb1fb0 0x4008d65e:0x3ffb1fd0
16:32:28.394 -> 
16:32:28.394 -> Rebooting...
16:32:28.394 -> ets Jun  8 2016 00:22:57
16:32:28.394 -> 
16:32:28.394 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:32:28.394 -> configsip: 0, SPIWP:0xee
16:32:28.394 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:32:28.441 -> mode:DIO, clock div:1
16:32:28.441 -> load:0x3fff0018,len:4
16:32:28.441 -> load:0x3fff001c,len:1216
16:32:28.441 -> ho 0 tail 12 room 4
16:32:28.441 -> load:0x40078000,len:10944
16:32:28.441 -> load:0x40080400,len:6388
16:32:28.441 -> entry 0x400806b4
16:32:28.909 -> E (40) psram: PSRAM ID read error: 0xffffffff
16:34:11.750 -> E (40) psram: PSRAM ID read error: 0xffffffff

kisvegabor, 2022-07-27:
I suggest using lvgl v8.3.1 as v9 is still heavily under development. 

> 16:32:15.966 -> E (40) psram: PSRAM ID read error: 0xffffffff

I'm not an ESP expert but it seems like a memory corruption issue. Maybe writing out of bounds? Wild pointers?

If you are migrating from lvgl v7 be sure to follow [this guide](https://docs.lvgl.io/master/CHANGELOG.html#driver-changes).

embeddedt, 2022-07-27:
How is `w_name` being constructed in the code? If it's not terminated correctly the roller could keep attempting to read other memory as though it were the list of options and that could lead to a crash.

On another note, I assume this is just a quick test, but for an actual application you should avoid recreating the roller every time; this will lead to an out-of-memory error and new rollers constantly appearing on top of each other.

bil7317, 2022-08-01:
**Issue Resolved:**
In my case this problem was caused when strings or character arrays are manipulated incorrectly. I was not getting any kind of compilation error but it was causing problem during execution. I was confused because in Arduino IDE you can apply strcpy() on char arrays but not on strings and some other builtin functions related to char and Strings. After proper manipulation of strings the issue was resolved. It also helps if you NULL a char array if you are not using it anymore i.e.
 `memset(&buf, 0, sizeof(buf));`
I also got an idea from question on stakeoverflow. It can also be helpful: [https://stackoverflow.com/questions/70708980/how-to-initialize-a-queue-on-a-second-core-of-an-esp32](url)

@kisvegabor Yes it can be caused by wild pointers but I think memory is not corrupt because after changes the code is working fine during both compilation and execution. May be Arduino IDE did not added error name for this kind of error so it shows that memory is damaged. I am new to this area so kindly correct me if I am wrong.

@embeddedt Yes you are right. Thanks

Thanks for responding.

kisvegabor, 2022-08-01:
Thanks for the update. Happy to hear that it's resolved.

I'm also not an Arduino expert, I was just guessing... :slightly_smiling_face: