lvgl/lvgl #3525: lvgl/lvgl.h: No such file or directory

leetut, 2022-07-27:
I have the file lvgl.h in the lvgl folder, still get the error when compiling lvgl_arduino

leetut, 2022-07-27:
/Users/lee/Documents/Arduino/libraries/lvgl/src/demos/benchmark/assets/img_benchmark_cogwheel_rgb565a8.c:4:23: fatal error: lvgl/lvgl.h: No such file or director

kisvegabor, 2022-07-27:
It's already fixed in master but wasn't fixed in v8.3. I've just added the fix to v8.3 too and it will included in the next bugfix release.

Until that please manually change [this](https://github.com/lvgl/lvgl/commit/4133f1d3171242667182ade7f3154d17a44c4b5c).

kisvegabor, 2022-08-01:
You're welcome. :slightly_smiling_face: 
I'm closing is this issue.

closed 2022-08-01 by kisvegabor

robonxt, 2022-09-22:
Just wanted to let whoever is also struggling with this issue that the fix from @kisvegabor wasn't included in 8.3.1 yet. Hopefully it gets integrated into the next update soon!

kisvegabor, 2022-09-25:
@embeddedt Are you ok with releasing v8.3.2 next week? Are these anything we should for?

embeddedt, 2022-09-25:
I don't think there are any other outstanding problems at the moment. We can always release 8.3.3 if needed. :laughing:

kisvegabor, 2022-09-27:
I've just released v8.3.2