lvgl/lvgl #3526: lv_

haidongpan, 2022-07-28:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
lv_music_demo can not run successful, and output the same log in loop and continuously

### To Reproduce

0. the lvgl version is v8.3
1. build the demo app
2. run it on RISC-V platform
```c
#include "lvgl/lvgl.h"
#include "lvgl/demos/lv_demos.h"
#include "lv_drivers/display/fbdev.h"
#include "lv_drivers/indev/evdev.h"
#include <unistd.h>
#include <pthread.h>
#include <time.h>
#include <sys/time.h>
#include <fb_ops.h>

#define DISP_BUF_SIZE (128 * 1024)

int main(void)
{
    /*lvGL init*/
    lv_init();

    /*Linux frame buffer device init*/
    fbdev_init();

    /*A small buffer for LittlevGL to draw the screen's content*/
    static lv_color_t buf1[DISP_BUF_SIZE];
    static lv_color_t buf2[DISP_BUF_SIZE];

    /*Initialize a descriptor for the buffer*/
    static lv_disp_draw_buf_t disp_buf;
    lv_disp_draw_buf_init(&disp_buf, buf1, buf2, DISP_BUF_SIZE);

    /*Initialize and register a display driver*/
    static lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.draw_buf   = &disp_buf;
    disp_drv.flush_cb   = fbdev_flush;

    disp_drv.hor_res    = 1024;
    disp_drv.ver_res    = 600;
    lv_disp_drv_register(&disp_drv);

    evdev_init();
    static lv_indev_drv_t indev_drv_1;
    lv_indev_drv_init(&indev_drv_1); /*Basic initialization*/
    indev_drv_1.type = LV_INDEV_TYPE_POINTER;

    /*This function will be called periodically (by the library) to get the mouse position and state*/
    indev_drv_1.read_cb = evdev_read;
    lv_indev_t *mouse_indev = lv_indev_drv_register(&indev_drv_1);

    /*Set a cursor for the mouse*/
    LV_IMG_DECLARE(mouse_cursor_icon)
    lv_obj_t * cursor_obj = lv_img_create(lv_scr_act()); /*Create an image object for the cursor */
    lv_img_set_src(cursor_obj, &mouse_cursor_icon);           /*Set the image source*/
    lv_indev_set_cursor(mouse_indev, cursor_obj);             /*Connect the image  object to the driver*/

    /*Create a Demo*/
    //lv_demo_widgets();
    lv_demo_music();
    //lv_demo_stress();
    //lv_demo_benchmark();

    /*Handle LitlevGL tasks (tickless mode)*/
    while(1) {
        lv_timer_handler();
        usleep(5000);
    }

    return 0;
}

/*Set in lv_conf.h as `LV_TICK_CUSTOM_SYS_TIME_EXPR`*/
uint32_t custom_tick_get(void)
{
    static uint64_t start_ms = 0;
    if(start_ms == 0) {
        struct timeval tv_start;
        gettimeofday(&tv_start, NULL);
        start_ms = (tv_start.tv_sec * 1000000 + tv_start.tv_usec) / 1000;
    }

    struct timeval tv_now;
    gettimeofday(&tv_now, NULL);
    uint64_t now_ms;
    now_ms = (tv_now.tv_sec * 1000000 + tv_now.tv_usec) / 1000;

    uint32_t time_ms = now_ms - start_ms;
    return time_ms;
}
```

### Expected behavior

I want to know why this music demo can not run, but another demo widget

### Screenshots or video

the log and lv_conf.h as the attachment shows
[log.txt](https://github.com/lvgl/lvgl/files/9204619/log.txt)
[lv_conf.h.txt](https://github.com/lvgl/lvgl/files/9204623/lv_conf.h.txt)

haidongpan, 2022-07-28:
Addtional info that the cpu frequency is 60M, I don't know if it affects

haidongpan, 2022-07-28:
I track the log and code, and found that there is a infinite loop between lv_obj_mark_layout_as_dirty and lv_obj_update_layout, i do not known why

kctan0805, 2022-07-29:
don't know whether it is relevant, I got divide by zero exception on my ARM platform at here: 
https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw_mask.c#L1265

I workaround it by replace this line to
`        
int pdiff = p1.y - p2.y;
`    
`    
int psign = (pdiff != 0) ? (pdiff / LV_ABS(pdiff)) : -1;
`

Should I create a PR?

haidongpan, 2022-07-29:
Thanks, i have try your modify, it is useless to my case

kisvegabor, 2022-08-01:
@kctan0805 
Yes, please send a PR. even if it's unrelated, it should be fixed.

@haidongpan  
Unfortunately it's working for me with your settings. Can you use a debugger on your platform?
If so I can help to find the rot of the issue via screen share.

closed 2022-08-23 by lvgl-bot