lvgl/lvgl #3527: How load picture of wink.png  in Visual Studio Project

maotouying, 2022-07-28:
I want to load a PNG image on my PC to LVGL,So I use the example of png，But the picture can't load success.

PGNetHun, 2022-07-29:
Hello,
Try to register the FS driver, and then load the image from that virtual FS drive.

See:  
Image loading: https://docs.lvgl.io/master/widgets/img.html#image-source
> > You also need to use LVGL's file system module and register a driver with some functions for the basic file operation.

FS:  https://docs.lvgl.io/master/overview/file-system.html

So, in the end:
`lv_img_set_src(img, "A:path_to_img/wink.png")`

(`A` is the LVGL's virtual FS drive letter)

Please let us know whether it worked.
(I use MicroPython-LVGL this way both on Unix and ESP32 device, and loading image works from LVGL's virtual FS)

maotouying, 2022-07-30:
Thanks for your answer，I use the LVGL's virtual FS，it can load the picture from my PC to LVGL

closed 2022-07-30 by maotouying