lvgl/lvgl #3530: Why LV_DEF_REFR_PERIOD is fixed

GorgonMeducer, 2022-07-29:
### Introduce the problem
I notice that since  [5369d7d473446a8953f2ea37b9116da918d3c8d2](https://github.com/lvgl/lvgl/commit/0e367d0a76ad7e961f439ff9482b8c43abf67831) macro `LV_DISP_DEF_REFR_PERIOD` has been renamed to `LV_DEF_REFR_PERIOD`. It is OK.

Why does the macro `LV_DEF_REFR_PERIOD` have a fixed value 33 rather than configurable? 

### Examples and cases

### Suggested solution

kisvegabor, 2022-08-02:
I was trying to reduce the number of config options  in `lv_conf.h` and removed some which can be run-time configured anyway and LVGL only needs a default value. 

It's not documented yet and might be worth a dedicated API function but you can adjust the period time like this:
```c
  lv_timer_set_period(disp->refr_timer, 10);
```

GorgonMeducer, 2022-08-02:
OK, as long as there is a method, no problem.

closed 2022-08-02 by GorgonMeducer

brightproject, 2025-06-17:
Sorry @kisvegabor , I encountered the need to increase `FPS`.

`My platformio.ini`

> build_flags = 
>   ; Don't use lv_conf.h. Tweak params via platfom.ini.
>   -D LV_CONF_SKIP
>   -D LV_CONF_INCLUDE_SIMPLE
>   ; Enable LVGL monitoring
>   -D LV_USE_SYSMON=1
>   -D LV_USE_MEM_MONITOR=1
>   -D LV_USE_PERF_MONITOR=1
>   -D LV_DEF_REFR_PERIOD=20

The flag has no effect on the system.

Where should I put this function?

> lv_timer_set_period(disp->refr_timer, 10);

GorgonMeducer, 2025-06-18:
I think your FPS is not limited by the `LV_DEF_REFR_PERIOD` but the device's 2D capability.
Please provide more details:

- Processor and Processor Frequency
- Resolution
- Colour Depth?
- Any GPU?
- Any Acceleration?