lvgl/lvgl #3533: Can you check lv_img_buf_get_img_size maybe miscalculation for 1/2/4 BIT images due to LV_IMG_BUF_SIZE_ALPHA_1BIT 2BIT 4BIT et-al

carterd, 2022-07-30:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

lv_img_buf.h Line 37-40

    /*+ 1: to be sure no fractional row*/
    #define LV_IMG_BUF_SIZE_ALPHA_1BIT(w, h) ((((w / 8) + 1) * h))
    #define LV_IMG_BUF_SIZE_ALPHA_2BIT(w, h) ((((w / 4) + 1) * h))
    #define LV_IMG_BUF_SIZE_ALPHA_4BIT(w, h) ((((w / 2) + 1) * h))
    #define LV_IMG_BUF_SIZE_ALPHA_8BIT(w, h) ((w * h))

Just adding addition of 1 seems to consume extra memory when widths effectively on byte boundary:

I'm thinking maybe something like would be more exact a calculation of the required size

    #define LV_IMG_BUF_SIZE_ALPHA_1BIT(w, h) ((( (w + 7) / 8) * h))
    #define LV_IMG_BUF_SIZE_ALPHA_2BIT(w, h) ((( (w  + 3) / 4) * h))
    #define LV_IMG_BUF_SIZE_ALPHA_4BIT(w, h) ((( (w +1 ) / 2) * h))

Thoughts????

### To Reproduce
  // OK an image with 7x8 pixels and 8x8 pixels should take the same amount of space
  lv_img_dsc_t* a = lv_img_buf_alloc(8, 8, LV_IMG_CF_INDEXED_1BIT);
  lv_img_dsc_t* b = lv_img_buf_alloc(7, 8, LV_IMG_CF_INDEXED_1BIT);

  LV_LOG_USER("data_size 8bit = %d", a->data_size); // 8 bytes for 2 colors + 8 bytes for pixels = 16 bytes
  LV_LOG_USER("data_size 7bit = %d", b->data_size); // 8 bytes for 2 colors + 8 bytes for pixels = 16 bytes

### Expected behavior

I believe that image data sizes for LV_IMG_CF_INDEXED_1BIT /2BIT /3BIT / 4BIT are incorrectly being calculated and extra unused memory is being allocated for such images when image widths are multiples of bytes.

### Screenshots or video

BLAHH Just look at the code looks dodgy to me HAHAHA

kisvegabor, 2022-08-02:
Perfectly makes sense. Can you send a pull request?

carterd, 2022-08-06:
I.. I.. dunno ... there is some other code in say lv_example_Canvas_2.py which also looks like it has this same issue ... can't see any unit tests that'll need updating. Just confirm you want a pull request but their isn't anything I'll be doing than the changes identified in the issue-report!

kisvegabor, 2022-08-08:
Yes, we don't have unit tests for all widget yet. So all you need to change rounding in the macros. 

FYI, we decided to sponsor writing unit tests from the donations LVGL received. Take look at this [issue](https://github.com/lvgl/lvgl/issues/2337) for more details.

closed 2022-08-30 by lvgl-bot