lvgl/lvgl #3535: Missing Micropython examples

amirgon, 2022-07-30:
I'm opening this issue to point out missing Micropython examples:

- [ffmpeg](https://github.com/lvgl/lvgl/tree/master/examples/libs/ffmpeg)
- [fragment](https://github.com/lvgl/lvgl/tree/master/examples/others/fragment)
- [gridnav](https://github.com/lvgl/lvgl/tree/master/examples/others/gridnav)
- [ime](https://github.com/lvgl/lvgl/tree/master/examples/others/ime)
- [imgfont](https://github.com/lvgl/lvgl/tree/master/examples/others/imgfont)
- [msg](https://github.com/lvgl/lvgl/tree/master/examples/others/msg) (only [lv_example_msg_1.py](https://github.com/lvgl/lvgl/blob/master/examples/others/msg/lv_example_msg_1.py) implemented)

Also, [Demos](https://github.com/lvgl/lvgl/tree/master/demos) are missing.  
[lv_demo_widgets.py](https://github.com/lvgl/lvgl/blob/master/demos/widgets/lv_demo_widgets.py) was once implemented, but is now outdated and doesn't work.

When someone adds a new C example, I think it's best asking them to provide Micropython example as well.  
The Micropython examples are [automatically tested in the CI](https://github.com/lvgl/lvgl/blob/master/.github/workflows/build_micropython.yml) so adding a Micropython example can ensure that the Micropython API is correct and the C callback conventions are respected for that feature.

Regarding demos, today these are not tested automatically in the CI but we could easily fix that when they are available.

CC: @kisvegabor @embeddedt @uraich

uraich, 2022-07-31:
Hi AmirI had already started to implement the widget demos. It is not complete yet but there is very little missing. I will try to complete it in the coming days and create a pull request.UliSent from my Galaxy

uraich, 2022-08-03:
Hi Amir,
I had a quick look into the ffmpeg example.
There is a uffmpeg written by Paul Sokolovsky for pycopy (his version of MicroPython) but this uses methods in uctypes which are not available in standard MicroPython.
I don't think it makes much sense to try implementing this example in MicroPython. In any case img.set_src will not work in MicroPython

amirgon, 2022-08-03:
Hi Uli

> I don't think it makes much sense to try implementing this example in MicroPython. In any case img.set_src will not work in MicroPython

My thinking was to support this on the unix port in the same way we support Freetype and Rlottie today.  
The idea is that [the makefile detects if the library is installed and enables the feature if it does](https://github.com/lvgl/lv_micropython/blob/0e2023e502db73a225644882a4501a2bb0774e45/ports/unix/Makefile#L220-L234).   
I agree that it does not make sense to support this for other ports, but it can still be useful on the unix port.

> AttributeError: type object 'LV_EVENT' has no attribute 'MSG_RECEIVED'

I don't see this problem on my side.

[Here is `lv_example_msg1.py` on the online simulator for v9.0 (master)](https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/6012d3e31cd58872a009674aade20f86800c7a2a/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/master/examples/others/msg/lv_example_msg_1.py)

So I think it works fine at least on the JS and the unix port.

Could you try to update submodules and rebuild?
Do you see this problem on a specific port?

uraich, 2022-08-04:
I recompiled lv_micropython from scratch and lv_example_msg1.py now works fine
Looking into lv_conf.h I see that 
 LV_USE_MONKEY,  LV_USE_FRAGMENT, LV_USE_IMGFONT, LV_USE_IME_PINYIN
are disabled. I will have to enable these to test the examples.
Concerning ffmpeg, I see that libavcodec, libavutil and libswscale are used. I found .pc files for  libavcodec and libavutil but not for the others. I guess that including ffmpeg into lv_micropython will require a bit more effort. Therefore I will start with the other examples.

uraich, 2022-08-16:
Hi Amir,
I managed to get a lv_micropython built with ffmpeg included. The first example also works. I do have a strange problem with the second example though:
This is what I see in lv_mpy.c:
/*
 * lvgl LV_FFMPEG_PLAYER_CMD object definitions
 */
    

STATIC const mp_rom_map_elem_t LV_FFMPEG_PLAYER_CMD_locals_dict_table[] = {
    { MP_ROM_QSTR(MP_QSTR_START), MP_ROM_PTR(MP_ROM_INT(LV_FFMPEG_PLAYER_CMD_START)) },
    { MP_ROM_QSTR(MP_QSTR_STOP), MP_ROM_PTR(MP_ROM_INT(LV_FFMPEG_PLAYER_CMD_STOP)) },
    { MP_ROM_QSTR(MP_QSTR_PAUSE), MP_ROM_PTR(MP_ROM_INT(LV_FFMPEG_PLAYER_CMD_PAUSE)) },
    { MP_ROM_QSTR(MP_QSTR_RESUME), MP_ROM_PTR(MP_ROM_INT(LV_FFMPEG_PLAYER_CMD_RESUME)) }
};

STATIC MP_DEFINE_CONST_DICT(LV_FFMPEG_PLAYER_CMD_locals_dict, LV_FFMPEG_PLAYER_CMD_locals_dict_table);

STATIC void LV_FFMPEG_PLAYER_CMD_print(const mp_print_t *print,
    mp_obj_t self_in,
    mp_print_kind_t kind)
{
    mp_printf(print, "lvgl LV_FFMPEG_PLAYER_CMD");
}

STATIC const mp_lv_obj_type_t mp_lv_LV_FFMPEG_PLAYER_CMD_type = {
    {
        { &mp_type_type },
        .name = MP_QSTR_LV_FFMPEG_PLAYER_CMD,
        .print = LV_FFMPEG_PLAYER_CMD_print,
        
        
        .attr = call_parent_methods,
        .locals_dict = (mp_obj_dict_t*)&LV_FFMPEG_PLAYER_CMD_locals_dict,
        
        .parent = NULL
    },
#ifdef LV_OBJ_T
    NULL
#endif
};
    
When I try to access lv.FFMPEG_PLAYER_CMD.START it tells me that lv.FFMPEG_PLAYER_CMD does not exist.

uraich, 2022-08-17:
I also had a look at lv_example_msg_2 and 3.
lv.subscribe_msg() crashes with a seg fault if the third parameter is not None. So... the first line in  lv_example_msg_2:
lv.msg_subsribe(MSG_LOGIN_ATTEMPT, auth_manager, "hello")
produces a seg fault
lv_example_msg_3 gets a bit further. However there I have the problem that lv_msg_get_payload(m) returns a pointer to the value which converts to blob in Python. Is there a way to extract the value?

amirgon, 2022-08-17:
Hi Uli, 

> When I try to access lv.FFMPEG_PLAYER_CMD.START it tells me that lv.FFMPEG_PLAYER_CMD does not exist.

We recently fixed a bug with qstrs, maybe this is related. Could you try again after updating lv_micropython to latest version?

> lv.subscribe_msg() crashes with a seg fault if the third parameter is not None

Try to keep the third parameter as None. user_data is used internally in micropython, and you don't really need to use it. Instead, pass a lambda as the callback with any extra parameter you need.

uraich, 2022-08-18:
Hi Amir,
I updated to the latest version but the error is not gone. In fact it happens not just to lv.FFMPEG_PLAYER_CMD .
I wrote this stupid test:
#!/opt/bin/lv_micropython -i
import sys
import lvgl as lv
import display_driver

try:
    print("LV_RES: ",lv.RES)
except:
    print("lv.RES is not defined")

try:
    print("LV_ALIGN: ",lv.ALIGN)
except:
    print("lv.ALIGN is not defined")

try:
    print("LV_ALIGN: ",lv.DIR)
except:
    print("lv.DIR is not defined")
    
try:
    print("LV_FONT_SUBPX: ",lv.FONT_SUBPX)
except:
    print("lv.FONT_SUBPX is not defined")

try:
    print("LV_TEXT_CMD_STATE: ",lv.TEXT_CMD_STATE)
except:
    print("lv.TEXT_CMD_STATE is not defined")

try:
    print("LV_STYLE_RES: ",lv.STYLE_RES)
except:
    print("lv.STYLE_RES is not defined")

try:
    print("LV_FS_RES: ",lv.FS_RES)
except:
    print("lv.FS_RES is not defined")

try:
    print("LV_FS_MODE: ",lv.FS_MODE)
except:
    print("lv.FS_MODE is not defined")

try:
    print("LV_IMG_CF: ",lv.IMG_CF)
except:
    print("lv.IMG_CF is not defined")

try:
    print("LV_IMG_SRC: ",lv.IMG_SRC)
except:
    print("lv.IMG_SRC is not defined")

try:
    print("LV__DRAW_MASK_RES: ",lv._DRAW_MASK_RES)
except:
    print("lv._DRAW_MASK_RES is not defined")

try:
    print("LV_METER_INDICATOR_TYPE: ",lv.METER_INDICATOR_TYPE)
except:
    print("lv.METER_INDICATOR_TYPE is not defined")

try:
    print("LV_METER_DRAW_PART: ",lv.METER_DRAW_PART)
except:
    print("lv.METER_DRAW_PART is not defined")

try:
    print("LV_GRIDNAV_CTRL: ",lv.GRIDNAV_CTRL)
except:
    print("lv.GRIDNAV_CTRL is not defined")
which gives me:
[image]

uraich, 2022-08-18:
Btw:
When I replace 
player.player_set_cmd(lv.FFMPEG_PLAYER_CMD.START) by player.player_set_cmd(0)
then program works fine

uraich, 2022-08-18:
When I enable LV_USE_IMGFONT I get
[image]

uraich, 2022-08-18:
Concerning the lv.msg examples:
Sure, I can pass the user_data into the function using lambda. However, m.getpayload() also returns a blob.

amirgon, 2022-08-18:
Hi Uli,
I'm currently traveling and don't have access to a computer. I'll look into these issues when I'm back (by the end of August)

uraich, 2022-08-19:
Hi Amir,
Thanks for the notice. I will try to advance with the examples as far as I can before you will be back. Quite a few are already running.

uraich, 2022-08-20:
When enabling LV_USE_IME_PINYIN, I get this error:
[image]

amirgon, 2022-08-26:
Hi Uli,

> When I try to access lv.FFMPEG_PLAYER_CMD.START it tells me that lv.FFMPEG_PLAYER_CMD does not exist.

`lv.ffmpeg_player.PLAYER_CMD.START` works well for me.
```python
>>> lv.ffmpeg_player.PLAYER_CMD.START
0
>>> help(lv.ffmpeg_player.PLAYER_CMD)
object <class 'LV_FFMPEG_PLAYER_CMD'> is of type type
  START -- 0
  STOP -- 1
  PAUSE -- 2
  RESUME -- 3
```

> In fact it happens not just to lv.FFMPEG_PLAYER_CMD

- Use `lv.DRAW_MASK_RES` without the leading underscore.
- Use `lv.img.CF` instead of `lv.IMG_CF`, same with `lv.img.SRC`
- Use `lv.meter.INDICATOR_TYPE` instead of `lv.METER_INDICATOR_TYPE`, same with `lv.meter.DRAW_PART`

You can use `help(...)` (or tab completions) to get an idea which are the members of each class.

> monkey, I don't understand what this is supposed to do.

I don't think we need monkey on Micropython.

> I also found a few small compilation problems in Gabor's code

I've updated lv_micropython to include the workaround in https://github.com/lvgl/lvgl/pull/3629

> When I enable LV_USE_IMGFONT I get

IMGFONT wasn't following the callback conventions.  

https://github.com/lvgl/lvgl/blob/efe47b17e4332bbcdc73db9788717aebc59c5ae5/src/others/imgfont/lv_imgfont.h#L42

`path_cb` is a callback but no `user_data` is passed.

Fixed here: https://github.com/lvgl/lvgl/pull/3630

> When enabling LV_USE_IME_PINYIN, I get this error:

Fixed here:
- https://github.com/lvgl/lv_binding_micropython/commit/51b666f3868d52bd3d64e6b28b9243f643af73df
- https://github.com/lvgl/lvgl/pull/3631

Please let me know if I covered everything, or of any new issues.

uraich, 2022-08-27:
Hi Amir,
Looks like I am a very lousy typist. Sorry for that.
lv.ffmpeg_player.PLAYER_CMD.START is ok. I was missing the "ffmpeg_player" . Both ffmpeg examples work fine now.
However, I used my own lv_ffmpeg.c and my own lv_conf.h and Makefile to build the ffmpeg version of lv_micropython. 
I cloned a fresh version of lv_micropython.
When trying to compile the unix port without any changes I get:
[image]
so... I disabled USE_FFMPEG and recompiled   OK!
I enabled USE_PNG: 
[image]
so, I disabled it again and enabled USE_FRAGMENT:  ok
I enabled USE_IMG_FONT: 
[image]
I disabled USE_IMGFONT and enabled USE_IME_PINYIN:
[image]

What am I doing wrong?

embeddedt, 2022-08-27:
I haven't merged all of @amirgon's aforementioned PRs yet; that's why you're still getting these errors.

uraich, 2022-08-27:
Ah... that explains!

uraich, 2022-08-27:
I made the modifications by hand and I can now enable USE_IME_PINYIN
The problem on USE_IMG_FONT remains. even with the modification as do the others.

amirgon, 2022-08-27:
Hi Uli,

> When trying to compile the unix port without any changes I get:
> [image]
> so... I disabled USE_FFMPEG and recompiled OK!

Strange. It works fine for me.  
Could you try to clean and rebuild?  
Maybe you are using some different version of ffmpeg? Or installed it differently?  

> I enabled USE_PNG:
> [image]

LV_PNG still requires some work.   
I'll let you know when it's ready.
(related: https://github.com/lvgl/lvgl/pull/3632)

> I enabled USE_IMG_FONT: 
> I disabled USE_IMGFONT and enabled USE_IME_PINYIN
> What am I doing wrong?

I enabled both IMGFONT and PINYIN by default on lv_micropython now.   (https://github.com/lvgl/lv_micropython/commit/500126aac950994e813b6b6df1850e987fd18058)
Please try latest version.

uraich, 2022-08-27:
Hi Amir,
I checked the compilation problem with FFMPEG. I am running on an Ubuntu 20.04 latest updates- On my system I need to link libavutil in addition to libavformat, libavcodec and libwscale. Then things work!
I also enabled IMGFONT and PINYIN  and that also does not give problems any longer.
Thanks!

amirgon, 2022-08-27:
> On my system I need to link libavutil in addition to libavformat, libavcodec and libwscale. Then things work!

It's still not clear to me because both `libavcodec` and `libavformat` depend on `libavutil`, so [pkgconfig should have added it automatically to the linking command line](https://github.com/lvgl/lv_micropython/blob/500126aac950994e813b6b6df1850e987fd18058/ports/unix/Makefile#L238).  
If it doesn't work on your side I can try adding it explicitly to `FFMPEG_LIBS` variable, please let me know if you think I should.

Regarding PNG, after https://github.com/lvgl/lvgl/pull/3632 is merged I'll need to remove lodepng from the Micropython bindings and fix the examples both on the Micropython bindings repo and on LVGL.  
This is done in these branches:
- https://github.com/amirgon/lvgl/tree/feat/lv_png_on_mp
- https://github.com/lvgl/lv_binding_micropython/tree/feat/lv_png_on_mp
- https://github.com/lvgl/lv_micropython/tree/feat/lv_png_on_mp

I suggest you wait with PNG until all these are settled and merged.

embeddedt, 2022-08-28:
@amirgon On my system (Arch) this is the output I see when running `pkg-config`; is it the same for you?

```console
$ pkg-config --silence-errors --libs libavcodec libavformat libswscale
-lavcodec -lavformat -lswscale
```

uraich, 2022-08-28:
@embeddedt : I see the same thing. In libavcodec.ps I see:
Requires.private: libswresample >= 3.9.100, libavutil >= 56.70.100
but it does not add -libavutil.
So I have to add libavutil to pkg-config:
pkg-config --silence-errors --libs libavcodec libavformat libswscale libavutil
which gives me:
-lavcodec -lavformat -lswscale -lavutil

uraich, 2022-08-28:
@embeddedt : I just went through all the examples and I found:
checkbox_2, keyboard_2 and menu_5 are missing from the widget examples.
All examples from others
The ffmpeg examples from lib.
I think this is the complete list. If you know of any more, please let me know. I will go through all of them and try to implement them.

amirgon, 2022-08-28:
> @amirgon On my system (Arch) this is the output I see when running `pkg-config`; is it the same for you?
> 
> ```
> $ pkg-config --silence-errors --libs libavcodec libavformat libswscale
> -lavcodec -lavformat -lswscale
> ```

I get this: 
```bash
$ pkg-config --silence-errors --libs libavcodec libavformat libswscale
-L/usr/local/lib -lavformat -lm -latomic -lz -lavcodec -pthread -lm -latomic -lz -lswscale -lm -latomic -lavutil -pthread -lm -latomic
```

Maybe that's because I followed the [download link](https://www.ffmpeg.org/download.html) in [LVGL docs](https://docs.lvgl.io/master/libs/ffmpeg.html) and built ffmpeg from sources instead of installing them by the package manager .

Anyway, I've [added libavutil to the Makefile](https://github.com/lvgl/lv_micropython/commit/8a2fecc7208cf99367fb0197125ef81e3548d700), hopefully this is ok now.

uraich, 2022-08-28:
Trying to add checkbox_2, I get an unexpected error that lv.RADIUS is not defined. lv.RADIUS.CIRCLE must now be named lv.RADIUS_CIRCLE. This, and similar changes break also other examples that worked fine before. I will have to go through all he examples and check them.

amirgon, 2022-08-28:
> This, and similar changes break also other examples that worked fine before.

The examples already on LVGL were fixed when I introduced this change, so I think you only need to fix new examples.  
The CI workflow runs all examples automatically so when something breaks we usually see that right away.

uraich, 2022-08-30:
I have finished the missing examples except the fragment ones. 
When I enable LV_USE_FRAGMENT in lv_conf.h I see plenty of definitions in lv.mpy.c but when I try to create a fragment manager with
manager = lv.fragment_manager()
it says AttributeError: 'module' object has no attribute 'fragment_manager'
I also don't know how I could handle the C structs defined in the examples.

amirgon, 2022-08-30:
Hi Uli,

Regarding lv_fragment, unfortunately it does not follow the agreed conventions.  
For example, most callbacks don't pass over `user_data` field and there may be additional problems.  

In it's current form, lv_fragment is incompatible with the Micropython bindings, so we can leave it out for now.

@kisvegabor When adding new features, I strongly suggest asking the author to follow the conventions we agreed on.  
It would be a good idea to also ask the author for a Micropython example, along with the C example.  
lv_fragment is an unfortunate example of what happens when not doing so.

kisvegabor, 2022-08-31:
> When adding new features, I strongly suggest asking the author to follow the conventions we agreed on.
It would be a good idea to also ask the author for a Micropython example, along with the C example.
lv_fragment is an unfortunate example of what happens when not doing so.

It sounds great in theory but seems hard to manage in the practice. This way we need the contributors to set up an MP environment, learn the MP binding and also implement the examples. It's doable would significantly reduce the number of PRs that are actually merged but would be useful for a lot of users. I see you point but don't know how to manage it effectively. 

The first thing could be to have a short and easy to follow list about MP conventions. We can communicate it as coding standard. Less is better so we should have as few rules as possible. I remember that it was already summarized in a list somewhere but was quite long.

Once the conventions are respected we can add the examples later by someone who is good at the MP. We can maintain a list of missing Micropython examples and communicate that contribution with them would be welcome. 

What do you think?

uraich, 2022-09-01:
Same problem on lv_imgfont_create. Is there a means to correct this?

amirgon, 2022-09-01:
> This way we need the contributors to set up an MP environment, learn the MP binding and also implement the examples.

@kisvegabor I understand the difficulty here, but I sometimes feel so frustrated seeing new features coming in, which are completely broken on Micropython.   
Looking at the merged PR there is no mention about Micropython compatibility and this new feature goes completely under the radar until someone tries to use it in Micropython and opens an issue that a feature is broken (like [this one](https://github.com/lvgl/lv_binding_micropython/issues/223) and [this fix](https://github.com/lvgl/lvgl/pull/3488) for example).   

I don't follow each LVGL PR and will not have the bandwidth to clean up and fix each case like this.  
In the long run this means more LVGL features will be broken on Micropython.

> It's doable would significantly reduce the number of PRs that are actually merged but would be useful for a lot of users. I see you point but don't know how to manage it effectively.

Well, on the most basic level, this requires very little effort:  
When a new PR comes in, just have a look at the public H file that contains the API:
- Does it follow the basic naming rules?
- Does it contain callbacks? If it does, was `user_data` added in the same fashion it appears on other LVGL APIs?

In general, if an API follows examples of existing LVGL API, chances are high that it will work fine.

But these questions are simply not asked and the Micropython bindings does not seem to be part of the review process.  
If you are uncertain about something, you can also mention me on the PR so I could have a look.  

> The first thing could be to have a short and easy to follow list about MP conventions. We can communicate it as coding standard. Less is better so we should have as few rules as possible. I remember that it was already summarized in a list somewhere but was quite long.

Good idea.  
I've opened a PR to track this: https://github.com/lvgl/lv_binding_micropython/issues/234

> Once the conventions are respected we can add the examples later by someone who is good at the MP. We can maintain a list of missing Micropython examples and communicate that contribution with them would be welcome.

Best is to have the same person write the examples on both C and Python.  
I understand that this could put off the PR author, so I'm not suggesting to enforce it. Only to ask.  
Maybe some authors would be willing to do it, and if not at least it will help raising the awareness to the Micropython API, coding convention etc.   

And if the author didn't add a Micropython example, we should at least maintain a list of such missing examples and keep track of them.

amirgon, 2022-09-01:
> Same problem on lv_imgfont_create. Is there a means to correct this?

@uraich I don't this this is the same problem.  
Just call `lv.imgfont_create` / `lv.imgfont_destroy`.  
Pass `None` on the user_data field, the Micropython bindings should handle it automatically.

kisvegabor, 2022-09-02:
@amirgon 
So in summary:
- create a simple of rules to follow and check them during review
- ask for MP example
- if the contributor can't/won't add the example add it to a TODO list

uraich, 2022-09-02:
ok, thanks.
I can create an imgfont but when I set the text style on the label to imgfont:
label1.set_style_text_font(imgfont, lv.PART.MAIN)
the program seg-faults

amirgon, 2022-09-03:
> So in summary:

@kisvegabor Yes, and I would add one more thing:  
When you are looking at a PR, pay attention to callbacks (passing over C function pointers).  
See if `user_data` is passed correctly.  
Most Micropython bindings problems are related to that, eventually.

amirgon, 2022-09-03:
> I didn't manage yet to find out why. Should I upload those programs somewhere, such that you could check it?

Yes, I would like to try these examples on my side and see the segfault.  
The simplest thing is to create a PR. You can mark it as "Draft" PR (GitHub has that feature) so it wouldn't be accidentally merged prematurely.  
Once there's a PR, it's very easy to pull it and test it.

uraich, 2022-09-03:
On the lv_demo_widget side it was an error on my side. The C code uses a function _lv_area_intersect, which is not accessible in MP. I therefore wrote a simple MP function replacing it and made a stupid error there. So... this problem is solved. I believe that the current version of the widget demo in the PR should be complete and correct.
I am still fighting with the chart4 example. I can also create a PR with all the example programs well knowing that char4 still gives some problem.

amirgon, 2022-09-03:
> I can also create a PR with all the example programs well knowing that char4 still gives some problem.

That would be great!

amirgon, 2022-09-20:
Related PRs:
- https://github.com/lvgl/lvgl/pull/3672
- https://github.com/lvgl/lvgl/pull/3671
- https://github.com/lvgl/lvgl/pull/3641
- https://github.com/lvgl/lvgl/pull/3677

uraich, 2022-09-20:
#3677 is a draft PR and can be closed. The problems of the imgfont example have been solved and the MP version integrated into #3672

amirgon, 2022-09-28:
A new LVGL feature was added without Micropython examples: https://github.com/lvgl/lvgl/pull/3680

kisvegabor, 2022-09-28:
Aaaah, shame on me. I missed it on first possible occasion... :sob: 

I'll add the example but I have some questions:
- I need to update `lv_conf.h` to enable the new feature. Can I push it back to `master`?
- Same question for the `lvgl` submodule.
- What is the best way to store a huge hex array and pass it as parameter? Is there an example that I can uses as reference?

uraich, 2022-09-28:
When I try to compile the tiny_ttf example in the simulator, this is what I see:
[image]

kisvegabor, 2022-09-28:
The math library (`-lm`) also needs to be linked.

amirgon, 2022-09-28:
> * I need to update `lv_conf.h` to enable the new feature. Can I push it back to `master`?
> * Same question for the `lvgl` submodule.

You need to update it locally, but you don't have to push it to lv_bindings right now. I can do it later when I update LVGL which I do from time to time.   
If you do want to push it, you'll need to first push the lv_conf.h update to lv_bindings, then update the submodule in lv_micropython, then push the submodule update.

> * What is the best way to store a huge hex array and pass it as parameter? Is there an example that I can uses as reference?

I think that it might make more sense reading it from a file. Is this supported?  
If you have to keep it in memory, I still suggest creating a file and reading it into an array in Python.

kisvegabor, 2022-09-28:
> You need to update it locally, but you don't have to push it to lv_bindings right now. I can do it later when I update LVGL which I do from time to time.

Won't it make crash GitHub's CI? That is if I push an example to `lvgl`, the MP test tries to call an undefined function until you update `lv_bindings`.

> I think that it might make more sense reading it from a file. Is this supported?
If you have to keep it in memory, I still suggest creating a file and reading it into an array in Python.

Reading from file is supported too, I'll try this way.

amirgon, 2022-09-28:
> Won't it make crash GitHub's CI? That is if I push an example to `lvgl`, the MP test tries to call an undefined function until you update `lv_bindings`.

You are right.  
It's probably best to update lv_bindings and lv_micropython before adding the example.  
Let me know if you need my help there.

uraich, 2022-09-28:
When trying to build lv_micropython (unix port) with LV_USE_TINY_TTF enabled I get this error:
[image]

embeddedt, 2022-09-28:
@amirgon Skipping this one was intentional on my end. I didn’t ask the author of that PR to provide a MicroPython example as they had already been having quite a few issues getting the C tests to build and run locally for them, so I figured MicroPython would frustrate them more.

uraich, 2022-09-28:
I gave it a quick try, converting ubuntu_font.c to ubuntu_font.py and converting the font into a bytearray.
This the result or lv_example_tiny_ttf_1.py
[image]

uraich, 2022-09-28:
I also tried to implement example_2 where you read the ttf file but for some reason 
lv.tiny_ttf_create_file is not generated in the lv_bindings.

amirgon, 2022-09-28:
I've updated lv_micropython and enabled `LV_USE_TINY_TTF` and `LV_TINY_TTF_FILE_SUPPORT`.  
@uraich You can open a PR on LVGL if you want.

amirgon, 2022-09-28:
@embeddedt looks like `LV_USE_TINY_TTF` breaks stm32: https://github.com/lvgl/lv_micropython/actions/runs/3145618901/jobs/5113126400

I can see [special SQRT handling](https://github.com/lvgl/lv_micropython/blob/b57d52411ad5a30e4320ecb7fe5217c9fb688006/ports/stm32/Makefile#L187-L193) on the Makefile, I'm not sure why it doesn't work here.

Any idea?

embeddedt, 2022-09-28:
For some reason the entire MicroPython-provided math library seems not to be linked on stm32.

uraich, 2022-09-28:
CI test of pull request #3707 fails with this error:
[image]

kisvegabor, 2022-09-28:
Thank you for adding the examples, Uli. 

I wonder if the assertation is also related to the math lib somehow. It's strange the we have errors only in MP.

amirgon, 2022-09-29:
> For some reason the entire MicroPython-provided math library seems not to be linked on stm32.

Since I'm not sure how to fix that, I [disabled TINY_TTF on stm32](https://github.com/lvgl/lv_micropython/pull/61) for now.  
If this gets fixed we can re-enable it.

> CI test of pull request #3707 fails with this error:

This is interesting.  
It looks related to the [fs_driver](https://github.com/lvgl/lv_binding_micropython/blob/master/lib/fs_driver.py) somehow. Example 1 which doesn't use the file system, passes.  
I'll need to look into it further.

amirgon, 2022-09-30:
I was looking into the crash in `lv_example_tiny_ttf_2.py` - here is what I found out so far:
- I'm not sure this is related to Micropython. This could be a bug in TINY_TTF.
- This is definitely unrelated to the Micropython FS driver, since it reproduces with LVGL Posix FS driver.
- Memory corruption occurs only when FS driver callbacks work slower. When they are fast - it doesn't.

Here is how I got to these conclusions:
- I replaced Micropython's `fs_driver` with LVGL Posix FS driver (`LV_USE_FS_POSIX`). **Problem disappeared**
- I created a new delegating FS driver(*), that calls LVGL Posix FS driver with some delay. **Problem reappeared**  
 ==> so it's unrelated to Micropython's `fs_driver`
- Setting a shorter label ("H" or "Hello" instead of "Hello world\nI'm a font created with Tiny TTF"). **Problem disappears**
- Changing the FS driver cache size affects the problem: With larger cache it works with longer strings.

<details>
  <summary><sub>(*) Here is the delegating driver implementation</sub></summary>

```python
def delegate(cb, cb_name, *args):
    # print(f'{cb.__class__.__name__} {cb_name} {args}')
    # Some delay can be added here
    return cb(*args) 

orig_drv = lv.fs_get_drv(ord('Z')) # Z is the Posix LVGL driver (#define LV_FS_POSIX_LETTER 'Z')
dl_fs_drv = lv.fs_drv_t()
dl_fs_drv.init()
dl_fs_drv.letter = ord('D') # D is the delegating driver
dl_fs_drv.open_cb = lambda *args: delegate(orig_drv.open_cb, 'fs_open_cb', *args)
dl_fs_drv.read_cb = lambda *args: delegate(orig_drv.read_cb, 'fs_read_cb', *args)
dl_fs_drv.seek_cb = lambda *args: delegate(orig_drv.seek_cb, 'fs_seek_cb', *args)
dl_fs_drv.close_cb = lambda *args: delegate(orig_drv.close_cb, 'fs_close_cb', *args)
dl_fs_drv.cache_size = 0 # Larger cache makes the problem happen only with longer labels
dl_fs_drv.register()
```

</details>

So my suspicion is that slower FS read/seek calls make TINY_TTF corrupt the RAM, but I'm not sure exactly how.  

I would appreciate if someone could reproduce this (in C) on their side to make sure it's not a local problem on my side.  
Just add some delay to the read/seek callbacks of the LVGL FS driver and run `lv_example_tiny_ttf_2.c`.

[11 more replies]

closed 2022-11-10 by kisvegabor