lvgl/lvgl #3542: invalid conversion from 'void*' to 'lv_obj_t* {aka _lv_obj_t*}' [-fpermissive]

bil7317, 2022-08-02:
### Perform all steps below and tick them with [x]
- [ x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I want to use keyboard to add text in a text area and save this text in a string. I copied this code directly from lvgl 8.2 example but it is giving me this error.

I am using Arduino IDE
ESP 32 
Display : ILI9488

### To Reproduce
```c
static void ta_event_cb(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t *  ta = lv_event_get_target(e);
    lv_obj_t *  kb = lv_event_get_user_data(e);
    if(code == LV_EVENT_FOCUSED) {
        lv_keyboard_set_textarea(kb, ta);
        lv_obj_clear_flag(kb, LV_OBJ_FLAG_HIDDEN);
    }

    if(code == LV_EVENT_DEFOCUSED) {
        lv_keyboard_set_textarea(kb, NULL);
        lv_obj_add_flag(kb, LV_OBJ_FLAG_HIDDEN);
    }
}

void lv_example_keyboard_1(void)
{
    /*Create a keyboard to use it with an of the text areas*/
    lv_obj_t *kb = lv_keyboard_create(lv_scr_act());

    /*Create a text area. The keyboard will write here*/
    lv_obj_t * ta;
    ta = lv_textarea_create(lv_scr_act());
    lv_obj_align(ta, LV_ALIGN_TOP_LEFT, 10, 10);
    lv_obj_add_event_cb(ta, ta_event_cb, LV_EVENT_ALL, kb);
    lv_textarea_set_placeholder_text(ta, "Hello");
    lv_obj_set_size(ta, 140, 80);

    ta = lv_textarea_create(lv_scr_act());
    lv_obj_align(ta, LV_ALIGN_TOP_RIGHT, -10, 10);
    lv_obj_add_event_cb(ta, ta_event_cb, LV_EVENT_ALL, kb);
    lv_obj_set_size(ta, 140, 80);

    lv_keyboard_set_textarea(kb, ta);
}

```

### Expected behavior
When I compile code it gives me this error:
`invalid conversion from 'void*' to 'lv_obj_t* {aka _lv_obj_t*}' [-fpermissive]`

It should compile the code as I did not make any changes and should be able to create keyboard.

### Screenshots or video

embeddedt, 2022-08-02:
Arduino generally forces you to compile all your code as C++, so you need to add a cast on the erroring line. We write the examples assuming a C environment.

davidoccam, 2022-08-18:
lv_obj_t *  kb = lv_event_get_user_data(e); // <---------------- I get error in this line

i changed to 
   lv_obj_t * kb = (lv_obj_t*) lv_event_get_user_data(e); //<------------------------no error now!

davidoccam, 2022-08-18:
lv_obj_t *  kb = lv_event_get_user_data(e); // <---------------- I get error in this line

lv_obj_t * kb = (lv_obj_t*) lv_event_get_user_data(e); // <----------------no error in this line

closed 2022-09-11 by lvgl-bot