lvgl/lvgl #3548: Slow fading with v8.3.1 vs v8.1.0 - esp-box (esp32)

chmorgan, 2022-08-05:
### Perform all steps below and tick them with [x]
- [X ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ]X Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Running on esp-box hardware (esp32).

Full screen bitmap fading out. v8.1.0 has a smooth fade over the time interval specified. v8.3.1 has no fade.

I'm trying to use v8.3.1 as I'd like to sequence some UI elements based on time and wanted to use the new animation deleted callback to detect the end of the fade out animation so I can show the next UI elements.

### To Reproduce
```c
Nothing just yet.
```
-->

### Expected behavior

### Screenshots or video

chmorgan, 2022-08-06:
Yeah, this is almost certainly an issue on my side, closing this out.

kisvegabor, 2022-08-07:
Be sure `LV_COLOR_SCREEN_TRANSP` is enabled in lv_conf.h. See more info [here](https://docs.lvgl.io/master/overview/style.html#opacity-blend-modes-and-transformations).

chmorgan, 2022-08-08:
@kisvegabor I checked on this and it looks like LV_COLOR_SCREEN_TRANSP requires 32bpp but I'm on a 16bpp system.

Is this a new requirement for fading (32bpp + COLOR_SCREEN_TRANSP) that showed up since 8.1.0? Trying to understand the origin and then what the fix might be, implementation improvement, using 32bpp even though my display is 16bpp etc.

chmorgan, 2022-08-09:
@kisvegabor and I tried 32bpp here just because and it didn't work, the graphics are corrupted.

kisvegabor, 2022-08-09:
> I checked on this and it looks like LV_COLOR_SCREEN_TRANSP requires 32bpp but I'm on a 16bpp system.

In v8.3 it works with any color depth.

chmorgan, 2022-08-09:
Is this coming back in a later 8.x release?

I’ll try 8.3 when I get home tonight. Thank you for that recommendation!

Chris

embeddedt, 2022-08-09:
I believe the intended solution for all future versions is to enable `LV_COLOR_SCREEN_TRANSP` since it now works with any color depth.

kisvegabor, 2022-08-09:
As @embeddedt says. In master (preparing v9) I've already removed `LV_COLOR_SCREEN_TRANSP` as it should be enabled in all cases.

chmorgan, 2022-08-09:
Tried master (76dc45f54682059e3070006cf8813a4234220530) but hitting a number of compile errors, some of which were easy fixes, missing ';' etc, others caused me to stop.  I got stuck at these and wrote this msg:

```
/Users/cmorgan/projects/esp-box/components/lvgl/lvgl/src/draw/sw/lv_draw_sw.h:40:51: error: unknown type name 'lv_draw_sw_blend_dsc_t'
     void (*blend)(lv_draw_ctx_t * draw_ctx, const lv_draw_sw_blend_dsc_t * dsc);
                                                   ^~~~~~~~~~~~~~~~~~~~~~
/Users/cmorgan/projects/esp-box/components/lvgl/lvgl/src/draw/sw/lv_draw_sw_blend.c: In function 'map_argb':
/Users/cmorgan/projects/esp-box/components/lvgl/lvgl/src/draw/sw/lv_draw_sw_blend.c:638:24: error: 'color_blend_true_color_additive' undeclared (first use in this function)
             blend_fp = color_blend_true_color_additive;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/cmorgan/projects/esp-box/components/lvgl/lvgl/src/draw/sw/lv_draw_sw_blend.c:638:24: note: each undeclared identifier is reported only once for each function it appears in
/Users/cmorgan/projects/esp-box/components/lvgl/lvgl/src/draw/sw/lv_draw_sw_blend.c:641:24: error: 'color_blend_true_color_subtractive' undeclared (first use in this function)
             blend_fp = color_blend_true_color_subtractive;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/cmorgan/projects/esp-box/components/lvgl/lvgl/src/draw/sw/lv_draw_sw_blend.c:644:24: error: 'color_blend_true_color_multiply' undeclared (first use in this function)
             blend_fp = color_blend_true_color_multiply;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/cmorgan/projects/esp-box/components/lvgl/lvgl/src/draw/sw/lv_draw_sw_blend.c: In function 'map_blended':
/Users/cmorgan/projects/esp-box/components/lvgl/lvgl/src/draw/sw/lv_draw_sw_blend.c:774:24: error: 'color_blend_true_color_additive' undeclared (first use in this function)
             blend_fp = color_blend_true_color_additive;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/cmorgan/projects/esp-box/components/lvgl/lvgl/src/draw/sw/lv_draw_sw_blend.c:777:24: error: 'color_blend_true_color_subtractive' undeclared (first use in this function)
             blend_fp = color_blend_true_color_subtractive;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/cmorgan/projects/esp-box/components/lvgl/lvgl/src/draw/sw/lv_draw_sw_blend.c:780:24: error: 'color_blend_true_color_multiply' undeclared (first use in this function)
             blend_fp = color_blend_true_color_multiply;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

Tried v8.3.1 and it still has LV_COLOR_SCREEN_TRANSP and won't let me select that option unless I'm 32bpp, its a check in Kconfig (and I'm updating this projects Kconfig.projbuild, which is an esp-idf specific file used by their Kconfig based configuration system).

Checked releases/v8.3 and it also has the check for 32bpp:

```
        config LV_COLOR_SCREEN_TRANSP
            bool "Enable more complex drawing routines to manage screens transparency."
            depends on LV_COLOR_DEPTH_32
            help
                Can be used if the UI is above another layer, e.g. an OSD menu or video player.
                Requires `LV_COLOR_DEPTH = 32` colors and the screen's `bg_opa` should be set to
                non LV_OPA_COVER value
```

embeddedt, 2022-08-09:
It seems that the Kconfig needs an update.

chmorgan, 2022-08-09:
It's also present in v8.3.1 tag:

```
        config LV_COLOR_SCREEN_TRANSP
            bool "Enable more complex drawing routines to manage screens transparency."
            depends on LV_COLOR_DEPTH_32
            help
                Can be used if the UI is above another layer, e.g. an OSD menu or video player.
                Requires `LV_COLOR_DEPTH = 32` colors and the screen's `bg_opa` should be set to
                non LV_OPA_COVER value
```

Should I open a PR against releases/8.3 to remove the conditional check, after testing here that it works? I'm glad to help out but also don't want to duplicate work. I'd be glad to help out.

embeddedt, 2022-08-09:
Yes, please!

chmorgan, 2022-08-09:
@embeddedt https://github.com/lvgl/lvgl/pull/3556, question on the doc update. Tested here and I'm getting callbacks (the reason I wanted v8.3) and fade out now. Thank you much.

kisvegabor, 2022-08-10:
Thank for the PR. I merged it, so I close the issue.

closed 2022-08-10 by kisvegabor