lvgl/lvgl #3559: Faulty spinbox decrement algorithm

MYaqoobEmbedded, 2022-08-10:
https://github.com/lvgl/lvgl/blob/e19410f8f8a256609da72cff549598e0df6fa4cf/src/extra/widgets/spinbox/lv_spinbox.c#L302

The spinbox decrement algorithm special case fails with the following scenario:
- Set Range (-11 to 12)
- Change cursor position of spinbox to 10 (step = 10)
- When decrementing from 02 by 10, we get -12, which exceeds the low range (we should get -11, the algorithm clearly fails)

I am not sure what to suggest as a fix right now, I am working on a fix for my own code, I will let you know when I fix it.

C47D, 2022-08-15:
I'll try to set a test case for this behavior. Is the expected behavior the correct one @kisvegabor @embeddedt?

kisvegabor, 2022-08-15:
It seems the problem is that 02-10=-8 but we expect it to be -12. It happens only at zero crossing, so at first look bringing [this line](https://github.com/lvgl/lvgl/blob/e19410f8f8a256609da72cff549598e0df6fa4cf/src/extra/widgets/spinbox/lv_spinbox.c#L302) to line 299 can solve the issue.This way we immediately assume that 02 is -02 and can decrement properly. Probably some checks are still required at the beginning to no invert the value if subtraction would cause out of limits

C47D, 2022-08-15:
> It seems the problem is that 02-10=-8 but we expect it to be -12.

I think the expected value is -11 instead of -12

MYaqoobEmbedded, 2022-08-16:
Hi,
Sorry, meant to say when decrementing from -2 by 10 once, I get -12, which is wrong because by min limit is -11.

MYaqoobEmbedded, 2022-08-16:
[image]

MYaqoobEmbedded, 2022-08-16:
Moving this
`    /*Special mode when zero crossing*/
    if((spinbox->value - spinbox->step) < 0 && spinbox->value > 0) spinbox->value = -spinbox->value;`
to line 299 in spinbox source file fixes this condition, as [kisvegabor](https://github.com/kisvegabor) suggested earlier

kisvegabor, 2022-08-23:
I've pushed a fix to https://github.com/lvgl/lvgl/tree/fix/spinbox-range-violation

Please take a look

embeddedt, 2022-08-23:
@kisvegabor Maybe I missed something but that branch appears to just be a copy of `master` at the moment. Did the push fail?

kisvegabor, 2022-08-23:
Ah, true. Updated.

MYaqoobEmbedded, 2022-08-23:
Great, I will test it on my project

kisvegabor, 2022-09-07:
It's already fixed and merged so I close this issue. Let me know if you have any remarks.