lvgl/lvgl #3560: Label touch point is not working

rmshub, 2022-08-11:
### Describe the bug
Created a play button label. Click is working only when pressed towards the edge. But, PC simulator is working properly. If text font is small montserrat 16, the click is not working at all.

### To Reproduce
```c
status =  lv_obj_create(home_screen);
lv_obj_set_pos(status, 5, 105);
lv_obj_set_size(status, 470, 55);
lv_obj_clear_flag(status, LV_OBJ_FLAG_SCROLLABLE);
lv_obj_add_flag(status, LV_OBJ_FLAG_CLICKABLE);

start = lv_label_create(status);
lv_obj_set_pos(start, 330, -3);
lv_label_set_text(start, LV_SYMBOL_PLAY);
lv_obj_add_flag(start, LV_OBJ_FLAG_CLICKABLE | LV_OBJ_FLAG_CLICK_FOCUSABLE);
lv_obj_set_style_text_color(start, lv_color_make(0, 102, 153), 0);
lv_obj_set_style_text_opa(start, 255, 0);
lv_obj_set_style_text_font(start, &lv_font_montserrat_22, 0);
lv_obj_add_event_cb(start, loom_process_cb, LV_EVENT_ALL, NULL);

static void loom_process_cb(lv_event_t *e)
{
    lv_event_code_t event = lv_event_get_code(e);
    lv_obj_t *obj = lv_event_get_target(e);
    if (event == LV_EVENT_CLICKED)
    {
        if (state)
        {
            lv_label_set_text(start, LV_SYMBOL_PAUSE);
        }
        else
        {
            lv_label_set_text(start, LV_SYMBOL_PLAY);
        }
        state ^= true;
    }
}
```

### Expected behavior
Click should work if pressed on any position on text label

### Screenshots or video

rmshub, 2022-08-12:
If touch coordinates x: 346, y: 138, label touch is not working.
If touch coordinates x: 367, y: 141 label touch is working.

kisvegabor, 2022-08-15:
I added this to the end of your code:

```c
  lv_obj_update_layout(status);
  printf("x1: %d, x2: %d\n", start->coords.x1, start->coords.x2);
  printf("y1: %d, y2: %d\n", start->coords.y1, start->coords.y2);
```

which prints:
```
x1: 357, x2: 375
y1: 124, y2: 147
```

So `x: 346` is really out of the label, but `x: 367` is on the label.

rmshub, 2022-08-19:
Thanks for response. It seems like issue with touch panel

closed 2022-08-19 by rmshub

BestStark, 2023-07-26:
> Thanks for response. It seems like issue with touch panel

Hi,

I'm using the same environment as you, but I came across LCD: 480 x 320 ILI9488
Touch: FT6236 does not work, only lvgl UI can be displayed, but there is no response when I click on the screen. I have modified the TFT touch function and used the FT6236 driver, I do not know the reason, could you help me?

thanks..

rmshub, 2023-07-30:
@BestStark Is your touch issue resolved ?