lvgl/lvgl #3561: How to customize the animation parameters of label scrolling?

qinshijing, 2022-08-11:
### Introduce the problem

When label sets long mode to LV_LABEL_LONG_SCROLL or LV_LABEL_LONG_SCROLL_CIRCULAR, I want to modify its animation parameters (for example: repeat_count/playback_delay/repeat_delay/ready_cb, etc.), how to do it?
### Examples and cases

### Suggested solution

At present, the speed of the animation is obtained using lv_obj_get_style_anim_speed(), and the animation is also obtained using lv_obj_get_style_anim(), and then a few specific parameters are replaced. However, it always feels that these two methods are not perfect, and all customization of animation has not been realized. I haven't thought of a better way yet.

kisvegabor, 2022-08-15:
Hi,

You can take a look at this example: https://docs.lvgl.io/master/examples.html#customize-circular-scrolling-animation

---

We use GitHub Issues for development-related discussions.

Please ask your question in the [Forum](https://forum.lvgl.io) to make it visible for a larger community. You can simply log in with your GitHub account.

closed 2022-08-15 by kisvegabor

qinshijing, 2022-08-15:
hi, i run the example, but can't set repeat_count and ready_cb, right?

qinshijing, 2022-08-15:
[image]
Only act_time and repeat_delay are provided here, so should we add other parameters of anim?

kisvegabor, 2022-08-16:
Ah, true. I see no problem with using more parameters after `if(anim_template)`. Which properties do you find useful?

qinshijing, 2022-08-16:
hi. I send a pull requests. Do you think this change is reasonable?

kisvegabor, 2022-08-17:
Yes, I think it's reasonable. Once the animation can be customized, lets make it really customizable.