lvgl/lvgl #3562: I can't clean the obj and renew the new one(v7)

TingRui608470158, 2022-08-11:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2022-08-15:
It looks strange because if `hidden_obj == true` you immediately delete the object you just created.

I'm not sure I fully understand what your goal is. If you just wan't to move the rectangle, just create it once, save it in a global variable an call `lv_obj_align(obj, NULL, LV_ALIGN_IN_TOP_LEFT, 100+(x_axis), 50);`

If you want to hide it periodically use `lv_obj_add/clear_flag(obj, LV_OBJ_FLAG_FIDDEN)`

TingRui608470158, 2022-08-19:
Sorry, please ignore ` hidden_obj == true` part .
I want to show a dynamic rect，but `lv_obj_clean` , `lv_obj_del_async` , `lv_obj_del ` all not working.

kisvegabor, 2022-08-19:
What do you mean by dynamic?

TingRui608470158, 2022-08-19:
a moving object

kisvegabor, 2022-08-19:
You can simply save `obj` as a global variable and call `lv_obj_set_x(obj, x)` without recreating the object.

TingRui608470158, 2022-08-19:
```
static lv_obj_t * obj_display;
static void task_objectDet_cb(lv_task_t* task)
{ 
        lv_style_init(&style);
	lv_style_set_bg_opa(&style, LV_STATE_DEFAULT, LV_OPA_0);
	lv_style_set_bg_color(&style, LV_STATE_DEFAULT, LV_COLOR_SILVER);
	lv_style_set_bg_grad_color(&style, LV_STATE_DEFAULT, LV_COLOR_BLUE);
	lv_style_set_bg_grad_dir(&style, LV_STATE_DEFAULT, LV_GRAD_DIR_VER);
	obj_display = lv_obj_create(lv_scr_act(), NULL);
	lv_obj_add_style(obj_display, LV_OBJ_PART_MAIN, &style);
	lv_obj_set_size(obj_display,100,100);
	x_axis++;
	lv_obj_align(obj_display, NULL, LV_ALIGN_IN_TOP_LEFT, 100+(x_axis), 50);
}
```

It still not work

kisvegabor, 2022-08-19:
Call this only once
```c
        lv_style_init(&style);
	lv_style_set_bg_opa(&style, LV_STATE_DEFAULT, LV_OPA_0);
	lv_style_set_bg_color(&style, LV_STATE_DEFAULT, LV_COLOR_SILVER);
	lv_style_set_bg_grad_color(&style, LV_STATE_DEFAULT, LV_COLOR_BLUE);
	lv_style_set_bg_grad_dir(&style, LV_STATE_DEFAULT, LV_GRAD_DIR_VER);
	obj_display = lv_obj_create(lv_scr_act(), NULL);
	lv_obj_add_style(obj_display, LV_OBJ_PART_MAIN, &style);
	lv_obj_set_size(obj_display,100,100);
```

And the task:
```c
static lv_obj_t * obj_display;
static void task_objectDet_cb(lv_task_t* task)
{ 
	x_axis++;
	lv_obj_align(obj_display, NULL, LV_ALIGN_IN_TOP_LEFT, 100+(x_axis), 50);
}
```

TingRui608470158, 2022-08-22:
It work !!!
Thanks you so much~~

closed 2022-08-22 by TingRui608470158