lvgl/lvgl #3571: No examples in the docs

kisvegabor, 2022-08-15:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

The example section of the widgets is empty:
[image]

Interestingly the examples here are working: https://docs.lvgl.io/master/examples.html#customize-circular-scrolling-animation

It might be related that v8.3 is missing from the version list:
[image]

@embeddedt, could you take a look?

embeddedt, 2022-08-15:
Fixed. The include paths for the examples' RST files just needed to be updated.

As an aside, the link you gave me crashes Chromium 104 with an "Aw, snap!" error. It does work in Firefox, so I suspect it's a Chromium bug that will be fixed eventually.

closed 2022-08-15 by embeddedt

kisvegabor, 2022-08-15:
Ah, thank you.

> As an aside, the link you gave me crashes Chromium 104 with an "Aw, snap!" error. It does work in Firefox, so I suspect it's a Chromium bug that will be fixed eventually.

It works for me in Brave, Chrome and Firefox too. Probably just a temporary bug.

commit b08bc4c7 (2022-08-15): docs(examples) fix include paths after 2461ac0758eae69a94ee60a9be50d3de882ced09

Fixes #3571