lvgl/lvgl #3572: The actual running time of the animation is less than the set duration

qinshijing, 2022-08-15:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When there are two animations, the running duration of the second animation does not match the set duration.

### To Reproduce

static lv_obj_t* create_page(lv_obj_t* parent, const char* text)
{
  lv_obj_t* page = lv_obj_create(parent);
  lv_obj_set_size(page, LV_HOR_RES, LV_VER_RES);
  lv_obj_t* label = lv_label_create(page);
  lv_obj_set_style_text_font(label, &lv_font_montserrat_48, 0);
  lv_label_set_text(label, text);
  lv_obj_center(label);
  return page;
}

static void anim_ready_cb(lv_anim_t* anim)
{
  printf("[anim] anim ready, tick %d\n", lv_tick_get());
}

static void start_anim(lv_obj_t* obj)
{
  lv_anim_t a;
  lv_anim_init(&a);
  lv_anim_set_var(&a, obj);
  lv_anim_set_values(&a, LV_HOR_RES, 0);
  lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t)lv_obj_set_x);
  lv_anim_set_ready_cb(&a, anim_ready_cb);
  lv_anim_set_time(&a, 200);
  lv_anim_start(&a);
  printf("[anim] anim start, tick %d\n", lv_tick_get());
}

static void btn_event_cb(lv_event_t* e)
{
  lv_obj_t* page = create_page(lv_scr_act(), "hello world");
  /** do something */
  usleep(100 * 1000);
  start_anim(page);
}

static void page_switch_anim_example(void)
{
  lv_obj_t* btn = lv_btn_create(lv_layer_sys());
  lv_obj_add_event_cb(btn, btn_event_cb, LV_EVENT_CLICKED, NULL);
}
### Expected behavior

Hope every animation runs correctly. But this seems unlikely for single-threaded lvgl. But I think the runtime of the last animation should be basically correct, what do you think?

### Screenshots or video

The 200ms animation only ran for 80ms
[image]

kisvegabor, 2022-08-18:
It was a tricky one but I've found it. :slightly_smiling_face: 

The problem is that `last_timer_run` for the animation is saved in 2 cases:
1. when there are no running animations (`anim_timer` is paused in this case) and new animation is started. See [here](https://github.com/lvgl/lvgl/blob/956c47f6a764fd7c1ff1be2d4d8da38bc8308902/src/misc/lv_anim.c#L86).
2. when `anim_timer` is finished [here](https://github.com/lvgl/lvgl/blob/956c47f6a764fd7c1ff1be2d4d8da38bc8308902/src/misc/lv_anim.c#L411).

So when there is only one animation, `last_timer_run` is saved in `start_anim` after `usleep` so there is no problem.

However, the button has a transition animation on press/release by default, so there are other running animations too. Therefore `last_timer_run` was saved in the last call of `anim_timer`, but meanwhile some time has been elapsed due to the `usleep`, so LVGL tries to catch up with the animations. If you set `usleep(3000)` and `lv_anim_set_time(&a, 6000);` you will see that the animation starts from the center. 

Normally this behavior would be correct, but the problem is that it's not known whether the animation was created before or after `usleep` so we don't know how `last_timer_run` should be considered. 

The only solution I can imagine is adding `last_run` to `lv_anim_t` and track it on per animation basis. Do you have any other ideas?

qinshijing, 2022-08-20:
There should be no better way than this. I agree with you.

commit d0191a6e (2022-08-28): fix(anim): fix the problem that the animation time is shortened  #3572 (#3626)

fixes #3572
closed 2022-08-28 by kisvegabor