lvgl/lvgl #3579: Calling the display driver causes an error and qemu to terminate in EFI

DenisChel74, 2022-08-16:
### Perform all steps below and tick them with [x]
- [ x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2022-08-19:
Seems like a LVGL can't allocate memory. 

What if you set `#define LV_MEM_CUSTOM 1` in `lv_conf.h`?

DenisChel74, 2022-08-21:
@kisvegabor No, the result is the same

kisvegabor, 2022-08-23:
What if you call `void * p = lv_mem_alloc(32)` right after `lv_init()`.

Please also enable `TRACE` log level and `LV_LOG_PRINTF` in `lv_conf.h`.

closed 2022-09-14 by lvgl-bot