lvgl/lvgl #3585: Code format hook

kisvegabor, 2022-08-17:
Something happened with pre-commit format hook because somehow the excluded files are note excluded.

I've tried to update `lv_conf_internal.h` by running `lv_conf_internal_gen.py` but on commit `lv_conf_internal.h` is formatted although it's excluded in `code-format.cfg`.

It worked so far and I have no idea what happened. I couldn't find anything related in git history.

@C47D could you take a look?

C47D, 2022-08-17:
Sure, I will take a look at it later today.

C47D, 2022-08-18:
I did a manual change in lv_conf_internal.h and ran astyle manually as pre-commit runs it

```commandline
astyle --options=scripts/code-format.cfg --ignore-exclude-errors src/lv_conf_internal.h
```

And it also formats it, so I think the issue is in the code-format.cfg file itself. I'll try to figure out why.

C47D, 2022-08-18:
for the whitespace issue reported in #3587 i think we can include the following patch

```diff
diff --git a/.pre-commit-config.yaml b/.pre-commit-config.yaml
index 7c9b2bfd8..5a8353893 100644
--- a/.pre-commit-config.yaml
+++ b/.pre-commit-config.yaml
@@ -4,7 +4,10 @@ repos:
 -   repo: https://github.com/pre-commit/pre-commit-hooks
     rev: v3.2.0
     hooks:
-    -   id: trailing-whitespace
+    - id: trailing-whitespace
+      stages: [ commit ]
+      exclude: 'src/lv_conf_internal.h'
 
 -   repo: local
```

The issue is because the trailing-whitespace hook runs on all the files, with the patch it will run on the pre-commit hook and will exclude the `src/lv_conf_internal.h` file.

You can test this patch by applying it, staging `.pre-commit-config.yaml`, then adding whitespace to the `src/lv_conf_internal.h` file and run the command
```commandline
pre-commit run trailing-whitespace
```
You will see `src/lv_conf_internal.h` file being ignored by pre-commit.

embeddedt, 2022-08-18:
If I understand correctly there are two issues here:

1. pre-commit and `lv_conf_internal_gen.py` both attempt to format `lv_conf_internal.h` differently.
2. The trailing-whitespace hook also attempts to run on `lv_conf_internal.h` when it shouldn't.

I suspect the first issue was caused by #3493. It removed `lv_conf_internal.h` from the exclude list in the pre-commit config. I'm not sure why it's still worked without errors for the last month, though.

The patch proposed by @C47D would fix only the second issue.

Let me know if I've missed something.

C47D, 2022-08-18:
Sounds about right. We could add back `lv_conf_internal.h` to the pre-commit exclude file, but we wouldn't be fixing the astyle bad behavior, just preventing it.

There's no much information online about astyle, I found an stack overflow thread where they had to dig into the source code to find out how astyle performs.

---

Should we add more exclude files to the trailing whitespace hook?

embeddedt, 2022-08-18:
> We could add back lv_conf_internal.h to the pre-commit excluded files, but we wouldn't be fixing the astyle missbehavior, just preventing it from happening.

Since `lv_conf_internal.h` is autogenerated I thought we don't expect astyle to format it correctly, and the formatting should be applied to `lv_conf_template.h`.

C47D, 2022-08-18:
Should we add all auto-generated files to the hook exclude files? From the little experiment I did astyle is failing to exclude a file that is in the --exclude list.

kisvegabor, 2022-08-19:
I don't think it's an astyle issue. If I run `code-format.py` from the `scripts` folder it correctly ignores the manually modified `lv_conf_internal.h` file.

embeddedt, 2022-08-19:
> If I run code-format.py from the scripts folder it correctly ignores the manually modified lv_conf_internal.h file.

pre-commit uses a separate exclude config from `code-format.py`; I think that is why.

kisvegabor, 2022-08-19:
`.pre-commit-config.yaml` uses the astyle config file:
https://github.com/lvgl/lvgl/blob/878e3d33343e112e205e3a1dbcb30fd610f0b99e/.pre-commit-config.yaml#L15

So I'm really confused.

embeddedt, 2022-08-19:
It used to have a separate excludes list as well prior to https://github.com/lvgl/lvgl/pull/3493. See here: https://github.com/xiaoxiang781216/lvgl/blob/a83cae012d0fa15b9ef5c7cbdfb5a185a8fbd864/.pre-commit-config.yaml

kisvegabor, 2022-08-19:
But `code-format.cfg` is picked up by pre commit for sure. I added `asd` at the end of the cfg file and it failed on commit with invalid option error. 

So if `code-format.py` is working well than`code-format.cfg` is good. and if pre commint picks up  `code-format.cfg` it should work too. 

The only thing I can imagine is that is pre commint  runs in `lvgl` directory, but `code-format.py` runs from the `scripts` directory. And the excludes are relative to the `scripts` folder.

However I tried a lot of combinations for the path of  `lv_conf_internal.h`  like
```
--exclude=../src/lv_conf_internal.h
--exclude=src/lv_conf_internal.h
--exclude=/home/kisvegabor/projects/lvgl/eclipse-workspace/lv_port_pc_eclipse_master/lvgl/src/lv_conf_internal.h
--exclude=lv_conf_internal.h
```

but it was still formatted.

C47D, 2022-08-19:
That's weird, If I add a malformatted code to lv_conf_internal.h and then run astyle with the configuration file it always get's formatted.

kisvegabor, 2022-08-23:
I can confirm that 
```
astyle --options=code-format.cfg --verbose ../src/lv_conf_internal.h
```

formats `lv_conf_internal.h`, while 
```
astyle --options=code-format.cfg --verbose --recursive "../src/*.c,*.h" 
```
ignores it correctly

Maybe if you directly add files (not e.g. `src/*`) to `astlye` it's stronger than the exclude list?

Anyway, it seems we have no other choice than restoring the local exclude list of the  pre-commit hook.

embeddedt, 2022-08-25:
> Maybe if you directly add files (not e.g. src/*) to astlye it's stronger than the exclude list?

It must be. Even passing two files to the same invocation doesn't work; it still formats `lv_conf_internal.h`.

> Anyway, it seems we have no other choice than restoring the local exclude list of the pre-commit hook.

I agree. It's unfortunate that we need to maintain two lists but they shouldn't need to be that long.

closed 2022-08-25 by kisvegabor

kisvegabor, 2022-08-25:
For the sake of simplicity I added only `src/libs` and `src/lv_conf_internal.h` for now.  So basically reverted dc1e4c87b in `.pre-commit-config.yaml`.

commit 48ee005d (2022-08-25): fix(format): fix lv_conf_internal.h formatting by pre-commit hook

fixes #3585