lvgl/lvgl #3590: lv_menu: Back button custom icon/text

geert-KLA-BE, 2022-08-18:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Currently the menu back button cannot be customized. I use a different icon set then de default so the default menu back button icon used does not work. The current menu API does not let me change this or get the object of the back button to allow changing this.

```
            lv_obj_t * sidebar_header_back_icon = lv_img_create(menu->sidebar_header_back_btn);
            lv_img_set_src(sidebar_header_back_icon, LV_SYMBOL_LEFT);
```

kisvegabor, 2022-08-23:
I think the most flexible would be to have API functions to get the menu and sidebar back buttons.

E.g. 
```c
lv_menu_get_main_back_btn(menu);
lv_menu_get_sidebar_back_btn(menu);
```

If you agree, please send a Pull request.

geert-KLA-BE, 2022-08-25:
Seems these already exist:
```
lv_obj_t * lv_menu_get_main_header_back_btn(lv_obj_t * obj);
lv_obj_t * lv_menu_get_sidebar_header_back_btn(lv_obj_t * obj);
```
The icon is set as a child of those. So it is possible by getting the first child of the button and update the the img source that way. Maybe a API for fetching that object could be useful then.

kisvegabor, 2022-08-28:
> Seems these already exist:

Oh great! I should have check that. 

> Maybe a API for fetching that object could be useful then.

I'd keep the API as lean as possible, and make users use the existing building blocks.

closed 2022-09-20 by lvgl-bot