lvgl/lvgl #3591: av_init_package is deprecated

uraich, 2022-08-18:
### Perform all steps below and tick them with [x]
- [ x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When compiling lvgl with ffmpeg included, I get 
error: ‘av_init_packet’ is deprecated [-Werror=deprecated-declarations]

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2022-08-19:
I found that other's solved it like [this](https://github.com/kaltura/nginx-vod-module/pull/1284/files#diff-b50cc1858646a1e6d4f6e885b99c283acc39433baaadb2d3355b50e4ec573fafL202-R203).

Once you have everything up and running to test it, could you try it out?

uraich, 2022-08-19:
Hi Gabor,
Yes, this works! However it requires a few more modifications to the code because 
av_packet_alloc returns a pointer to AVPacket while av_init_packet returns the packet itself.
In order to include ffmpeg into the simulator I had to modify the CMakeList.txt of lvgl as well to included the missing libraries.
Currently I am working on the missing micropython examples as requested by Amir. I will prepare a pull request as soon as I have progressed with this.

kisvegabor, 2022-08-19:
Great, thank you!

uraich, 2022-08-20:
When enabling LV_USE_IME_PINYIN I get an error comparing values with different signedness.
I modified int flag and int count to uint32  in function pinyin_k9_get_legal_py of lv_ime_pinyin.c to get rid of the error.

kisvegabor, 2022-08-23:
`LV_USE_IME_PINYIN` wasn't enabled in the CI tests so these warnings wasn't spotted. 

I've added it to the CI now, and also fixed the warnings.

uraich, 2022-09-07:
This has been done, I therefore close the issue.

closed 2022-09-07 by uraich