lvgl/lvgl #3592: lv_menu_page: Title not changeable

geert-KLA-BE, 2022-08-18:
### Perform all steps below and tick them with [x]
- [X ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X ] Update lvgl to the latest version
- [X ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

It is only possible to set a menu page title upon creation of the page. It should be possible to modify it.
Also it is currently dynamically creating the text string. It should also be possible to use a static string.
Also the creation of the page the title parameter should be const since it is not changed.

```
lv_menu_page_create(lv_obj_t * parent, char * title)
```

## Expected behavior

- Title should be possible to be set after creation
- Title should be possible to set to a static string
- Title parameter should be const

geert-KLA-BE, 2022-08-26:
What is the typical way in lvgl used to make sure we do not free a static set title?
Add a seperate title parameter or a bool flag to indicate it is static or dynamic?

kisvegabor, 2022-08-28:
E.g. with `lv_label` we have a flag like [this](https://github.com/lvgl/lvgl/blob/master/src/widgets/label/lv_label.h#L75).

PR #3624 (2022-08-29): feat(menu): Add setting of a page title
### Description of the feature or fix

Fixes #3592

### Checkpoints
- [ ] Follow the [styling guide](https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md)
- [ ] Run `code-format.py` from the `scripts` folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the documentation

Cannot run the unit test and formatting code on windows.
closed 2022-08-29 by kisvegabor