lvgl/lvgl #3597: CMSIS-Pack broken links

koval-alexander, 2022-08-19:
### Introduce the problem

### Examples and cases

### Suggested solution

kisvegabor, 2022-08-23:
cc @GorgonMeducer

GorgonMeducer, 2022-08-23:
Will have a look

GorgonMeducer, 2022-08-23:
@kisvegabor @koval-alexander  Double Checked, I see no problem from my side.

Pack installer works well.
[image]

GorgonMeducer, 2022-08-23:
> It seems that links pointing to master branch instead of 8.3 and so on.

That's intended behaviour, no problem at all.

GorgonMeducer, 2022-08-23:
@kisvegabor please verify what I have said.

GorgonMeducer, 2022-08-23:
@koval-alexander there are some suggestions for your MDK environment:

- try the latest MDK (current version is 5.37)
- Open Pack-installer and select menu Packs->Check for Updates

[image]

koval-alexander, 2022-08-24:
Hmm, it looks like the problem is really on my side, checked on another computer, everything works fine. Clearing the MDK cache seems to solve the problem.
Links on [arm site](https://developer.arm.com/tools-and-software/embedded/cmsis/cmsis-packs) are still broken, but on keil.com it's fine. It is strange that the links are not updated at the same time.
Thanks for your help, everything is working now.

closed 2022-08-24 by koval-alexander