lvgl/lvgl #3602: Set wallpaper failed

ay123net, 2022-08-20:
Hi,

I used lv_disp_set_bg_image() to set the background as wallpaper, but it still display the default color. I can't find a related demo to set the wallpaper.

Here is my code, is it right?
`
    ...
    /*Finally register the driver*/
    lv_disp_drv_register(&disp_drv);

    LV_IMG_DECLARE(img_lv_demo_music_wave_top);
    lv_disp_set_bg_image(NULL, &img_lv_demo_music_wave_top);
`
LVGL version: 8.3.1

kisvegabor, 2022-08-24:
Hi,

See some details [here](https://docs.lvgl.io/master/overview/display.html#background).
> Every display has a background color, background image and background opacity properties. They become visible when the current screen is transparent or not positioned to cover the whole display.

ay123net, 2022-08-26:
> Hi,
> 
> See some details [here](https://docs.lvgl.io/master/overview/display.html#background).
> 
> > Every display has a background color, background image and background opacity properties. They become visible when the current screen is transparent or not positioned to cover the whole display.

Hi,

I had follow this doc, but it still not work, whatever set the backgound color or image.

ay123net, 2022-08-27:
I think the background may be override by the default theme style, because the BG color is set by
`color_scr = theme.flags & MODE_DARK ? DARK_COLOR_SCR : LIGHT_COLOR_SCR;`

closed 2022-08-28 by ay123net