lvgl/lvgl #3603: Error: unaligned opcodes detected in executable segment with flag -Os

victpork, 2022-08-21:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version <- Running with version 8.3.1
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html) <- N/A, compile issue happens in Arduino IDE only

### Describe the bug
Compile error with message `Error: unaligned opcodes detected in executable segment with flag` when compile with `-Os` flag, everything goes fine with other optimize flags(e.g. `-Ofast`)
### To Reproduce
Just running sample code bundled with lvgl distribution in Arduino IDE.
Board is SparkFun Pro Micro RP2040, DEV env is Arduino
```c
#include <lvgl.h>
#include <TFT_eSPI.h>

/*Change to your screen resolution*/
static const uint16_t screenWidth  = 480;
static const uint16_t screenHeight = 320;

static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf[ screenWidth * 10 ];

TFT_eSPI tft = TFT_eSPI(screenWidth, screenHeight); /* TFT instance */

/* Display flushing */
void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )
{
    uint32_t w = ( area->x2 - area->x1 + 1 );
    uint32_t h = ( area->y2 - area->y1 + 1 );

    tft.startWrite();
    tft.setAddrWindow( area->x1, area->y1, w, h );
    tft.pushColors( ( uint16_t * )&color_p->full, w * h, true );
    tft.endWrite();

    lv_disp_flush_ready( disp );
}

/*Read the touchpad*/
void my_touchpad_read( lv_indev_drv_t * indev_driver, lv_indev_data_t * data )
{
    uint16_t touchX, touchY;

    bool touched = tft.getTouch( &touchX, &touchY, 600 );

    if( !touched )
    {
        data->state = LV_INDEV_STATE_REL;
    }
    else
    {
        data->state = LV_INDEV_STATE_PR;

        /*Set the coordinates*/
        data->point.x = touchX;
        data->point.y = touchY;

        Serial.print( "Data x " );
        Serial.println( touchX );

        Serial.print( "Data y " );
        Serial.println( touchY );
    }
}

void setup()
{
    Serial.begin( 115200 ); /* prepare for possible serial debug */

    lv_init();

    tft.begin();          /* TFT init */
    tft.setRotation( 3 ); /* Landscape orientation, flipped */

    /*Set the touchscreen calibration data,
     the actual data for your display can be acquired using
     the Generic -> Touch_calibrate example from the TFT_eSPI library*/
    uint16_t calData[5] = { 275, 3620, 264, 3532, 1 };
    tft.setTouch( calData );

    lv_disp_draw_buf_init( &draw_buf, buf, NULL, screenWidth * 10 );

    /*Initialize the display*/
    static lv_disp_drv_t disp_drv;
    lv_disp_drv_init( &disp_drv );
    /*Change the following line to your display resolution*/
    disp_drv.hor_res = screenWidth;
    disp_drv.ver_res = screenHeight;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register( &disp_drv );

    /*Initialize the (dummy) input device driver*/
    static lv_indev_drv_t indev_drv;
    lv_indev_drv_init( &indev_drv );
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = my_touchpad_read;
    lv_indev_drv_register( &indev_drv );

    /* Begin code here */
    
    static lv_style_t style;
    lv_style_init(&style);
    lv_style_set_text_font(&style, &lv_font_montserrat_48);
    lv_style_set_width(&style, 250);

    lv_obj_t * label1 = lv_label_create(lv_scr_act());
    lv_label_set_text(label1, LV_SYMBOL_OK);
    lv_obj_add_style(label1, &style, 0); 
    lv_obj_center(label1);
}

void loop()
{
    lv_timer_handler(); /* let the GUI do its work */
    delay( 5 );
}
```
Compiler has the following output:
```
C:\Users\user\AppData\Local\Temp\ccamc2cB.s: Assembler messages:
C:\Users\user\AppData\Local\Temp\ccamc2cB.s: Error: unaligned opcodes detected in executable segment

exit status 1
```
### Expected behavior
Compile without any error/warning
When compile with `-Ofast`:
```
Sketch uses 433476 bytes (2%) of program storage space. Maximum is 16773120 bytes.
Global variables use 68260 bytes (26%) of dynamic memory, leaving 193884 bytes for local variables. Maximum is 262144 bytes.
```

### Screenshots or video
N/A

embeddedt, 2022-08-21:
There is no handwritten assembly in LVGL so this sounds like a compiler bug. It has been reported before: https://github.com/lvgl/lvgl/issues/3425

embeddedt, 2022-09-05:
I'll close this as I don't think we can do much on our end to fix it, unless we avoided switch cases with over 16 statements.

closed 2022-09-05 by embeddedt