lvgl/lvgl #3613: Encoder long-press on one-checked-button-matrix makes selection without visually selecting button

pm-mic, 2022-08-24:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

kisvegabor, 2022-09-01:
Hi,

Sorry for the late reply.

The problem is that when the PRESS happens we can don't know if it will be a long press or not so the press event  needs to be processed.

However you can make the buttons to not react on press bot in release. It worked for me:
```c
lv_btnmatrix_set_btn_ctrl_all(btnm1, LV_BTNMATRIX_CTRL_CLICK_TRIG | LV_BTNMATRIX_CTRL_NO_REPEAT);
```

pm-mic, 2022-09-13:
Hi Gabor, 

Thank you very much for this solution. It works as expected, this provides exactly the desired behavior. 

In my opinion this is a workaround for a bug because the default behavior has undesired side-effects.
But as this is probably an edge case since it seems I'm the only one so far who has run into this, I would suggest to at least mention this in the documentation.

Regardless, I'm very thankful for this solution.

closed 2022-09-13 by pm-mic

kisvegabor, 2022-09-15:
Happy to hear it works. I've updated the docs.